Koenkk/zigbee2mqtt #26917: Third Reality Soil Sensor (3RSM0147Z) wrong device_class

DanfieldNet, 2025-03-30:
### What happened?

The Third Reality Soil Sensor (3RSM0147Z) definition exposes a device_class of Humidity which is typically used as the humidity of air.  Rather, the sensor should be exposed as device_class Moisture.  

### What did you expect to happen?

Personally, I am trying to use this new sensor with the HA Plant Monitor integration, which is looking for device_class Moisture.  An issue opened with the author indicated this was a problem with the device definition. 

### Zigbee2MQTT version

2.1.3-1

### Adapter firmware version

20240716

### Adapter

SMLIGHT SLZB-06p7

### Setup

HAOS Add-on

MMi1603, 2025-04-03:
I would also appreciate the change to the soil_moisture value just as in several devices from Tuya. Regards, MMi

3reality-support, 2025-06-05:
Expected to be fixed in v2.5.0

MSL-DA, 2025-06-05:
> Expected to be fixed in v2.5.0

Please change the update interval to +/- 10% or once every hour (or something smarter). It's absolutely insane how much battery these devices consume.

3reality-support, 2025-06-06:
You can set the relevant report interval here

3reality-support, 2025-06-06:
Also, regarding the battery consumption you mentioned, we would like to know approximately how long it takes for a new battery to be used up, or how long it takes for the battery to reach 50% usage

> [@3reality-support](https://github.com/3reality-support)
> 
> > Expected to be fixed in v2.5.0
> 
> Please change the update interval to +/- 10% or once every hour (or something smarter). It's absolutely insane how much battery these devices consume.

DanfieldNet, 2025-06-06:
Here’s a portion of my battery report.  Note in the case of these sensors, the Last Reported date is the date they were first started.  The Basil is the newest and dropped off very quick, but all others were about 5 weeks ago.

MSL-DA, 2025-06-13:
@3reality-support 

You've done nothing more than take a humidity sensor and output its value? If I want a reasonably accurate value in soil, I have to calibrate it against another device. Is that correct?

peternijssen, 2025-07-02:
According to the PR, it requires a yet to be released firmware update.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9532

MSL-DA, 2025-07-02:
> > But to be honest, this sensor is worthless in z2m/zha
> 
> Why do you write that? It seems OK to me.

Why are you writing here then?

Have you ever checked this sensor against an analog sensor?

peternijssen, 2025-07-02:
I bought it based upon this review and the little (not lab) test that has been done with a different sensor. https://smarthomescene.com/reviews/thirdreality-smart-soil-moisture-sensor-review/

However, discussing whether or not it is a good device does not belong in this thread.

MSL-DA, 2025-07-02:
> I bought it based upon this review and the little (not lab) test that has been done with a different sensor. https://smarthomescene.com/reviews/thirdreality-smart-soil-moisture-sensor-review/
> 
> However, discussing whether or not it is a good device does not belong in this thread.

Fantastic... then I hope you experience true moisture 😅

RedNo7, 2025-07-02:
> Why are you writing here then?

OK, that feels a little aggressive.  What I mean is that it seems stable on Z2M and (subject to moisture working) I have no reason to believe it is worthless.

> Have you ever checked this sensor against an analog sensor?

Not yet - I only received it a couple of days ago.

jimmyeao, 2025-07-09:
is there a trick to get these into Z2M?
```
[2025-07-09 17:03:07] error: 	z2m: Failed to configure 'BobMonitor', attempt 6 (Error: Device 0xffffb40e06020092 has no input cluster msSoilMoisture
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:61:15)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:214:29)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:27)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:46:25)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
```

bochelork, 2025-07-09:
@3reality-support any updates? Is it faulty? Or should i return it?

MSL-DA, 2025-07-11:
> It will be fixed in version 45 and will be opened if there are no issues during testing

Great support 🤭

bochelork, 2025-07-14:
> > > It will be fixed in version 45 and will be opened if there are no issues during testing
> > 
> > 
> > When will this version be released? I bought 8 sensors to automate the irrigation and soil_moisture is always NULL. If these sensors don't work properly I will return them within the next 14 days.
> 
> You can temporarily use air humidity, which is actually the humidity of the soil detected by the sensor

Ah, I already wondered how it measured air humidity while the device is a closed system. Thanks for the update.

3reality-support, 2025-07-14:
Because when developing the product, we mistakenly used humidity instead of air humidity. Later, we discovered that there was also soil humidity, and this product can only measure soil humidity, not air humidity

3reality-support, 2025-07-14:
> > > > > It will be fixed in version 45 and will be opened if there are no issues during testing
> > > > 
> > > > 
> > > > When will this version be released? I bought 8 sensors to automate the irrigation and soil_moisture is always NULL. If these sensors don't work properly I will return them within the next 14 days.
> > > 
> > > 
> > > You can temporarily use air humidity, which is actually the humidity of the soil detected by the sensor
> > 
> > 
> > [@3reality-support](https://github.com/3reality-support) I am not sure i got this correctly. So the hardware exposes 3 sensors: temperature, humidity, soil moisture. Are you saying that humidity and soil moisture are actually the same? Then why are exposed as two different sensors?
> 
> If the value of soil_moisture is equal to the value of humidity than this sensor is really unusable. I am comparing the values to a very good one **Trübner SMT50** and the values of humidity of the **3RSM0147Z** has nothing to do with the current soil moisture.
> 
> Battery drain of 3RSM0147Z is very fast. I tried to reduce the intervall of sending messages to 3600 seconds via Z2M but this does not seem to work. Battery got from 100% to 85% within 3 days.

Can you provide the values of air humidity for 3RSM0147Z and soil humidity for Tr ü bner SMT50, and set the error log for conifg

[image]

FHEMbeta, 2025-07-14:
@3reality-support Trübner SMT50 is outputting a linear voltage from 0-3V for humidity and does not work over Zigbee2MQTT but with an ESPHome integration.

At the moment SMT50 is outputting 39% soil moisture (correct value) whereas the 3RSM0147Z says 94% humidity (which is much too much).

Schorse3333, 2025-07-14:
> [@3reality-support](https://github.com/3reality-support) Trübner SMT50 is outputting a linear voltage from 0-3V for humidity and does not work over Zigbee2MQTT but with an ESPHome integration.
>
> At [...]

Did you calibrate the sensor, as it is outlined in some of the other forums?

`If there is any suspicion that the humidity is abnormal, you can calibrate it manually. The method is as follows: Press the reset button quickly three times. The LED will light up for 3 seconds. Then, place it in the air for at least 15 seconds, and then put it entirely into water for at least 15 seconds. Be sure to use tap water. Do not use purified water or distilled water. After that, power off and restart, and it can be officially used.`

This worked for me as a charm. My sensor is now reporting almost the same values as the SMT50.

FHEMbeta, 2025-07-14:
@Schorse3333 I tried to recalibrate and now the values of the sensors are around double the value of the SMT50 (measures from 0-50%, not 0-100%). Are your values nearly equal or double the value of the SMT50, too?

How did you fix energy consumption? My two sensors eat up the battery very quickly reporting much too often the values. Setting up a higher min Intervall in Zigbee2MQTT does have no effects. They always send every 5 minutes.

FHEMbeta, 2025-07-15:
@Schorse3333 My Zigbee network has multiple repeater next to the soil moisture sensors and is very stable - that's not the problem.

I have two 3RSM0147Z and maybe they are both faulty. I believe it's a lottery if those sensors work and they have major quality problems. Both sensors are now more than the double value of the SMT50 (one nearly 100%). I repeated calibration but it seems not to work.

FHEMbeta, 2025-07-22:
@3reality-support Any news about the new firmware?

FHEMbeta, 2025-07-22:
> > [@3reality-support](https://github.com/3reality-support) Any news about the new firmware?
> 
> Why waste time on these sensors? Their measurements are completely wrong, and a firmware update won't change that.

Because Ecowitt WH51 needs some extra hardware. Other sensors (SMT50) are much more expensive and need further hardware to work. Tried Befle SoMoSe, but they have no good values, too.

Do you know a good working alternative which is about 20,- Euro per sensor?

FHEMbeta, 2025-07-22:
@MSL-DA The problem is not the extra 60.- bucks. The problems are: receiver is single point of failure with maybe not sufficient range, just 8 sensors (is too less for me) and one more 24/7 device consuming power, nobody wants to see and so on...

So this sensor working on an existing Zigbee network seems to be a good idea - but yes, it does not work at the moment.

peternijssen, 2025-07-23:
Currently retrieving an OTA update here. Havent checked it yet though.

RedNo7, 2025-07-24:
also just updated.

v1.00.46 now populates `soil_moisture` (same reading as `humidity` of course).

[image]

MSL-DA, 2025-07-30:
> > also just updated.
> > v1.00.46 now populates `soil_moisture` (same reading as `humidity` of course).
> > <img alt="Image" width="241" height="227" src="https://private-user-images.githubusercontent.com/130979051/470289809-aeb98450-47e8-4d3a-ba2d-0a85f30964b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM3ODExMTYsIm5iZiI6MTc1Mzc4MDgxNiwicGF0aCI6Ii8xMzA5NzkwNTEvNDcwMjg5ODA5LWFlYjk4NDUwLTQ3ZTgtNGQzYS1iYTJkLTBhODVmMzA5NjRiNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcyOVQwOTIwMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YTI1NDUwNGZmZGU4MTU0ZDk1M2I2YmJlNzM0ZWQ4YTZmZTY1NzYwNzNhMDlhZjVkYzY0M2YzYmUxODA1YmVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.8qjjE8jdnKhufbnUC7dPWXFVJZxZxXpkGcjYl2jYec0">
> 
> If you want to display only soil moisture, you can see here https://github.com/thirdreality/homeassistant/blob/main/Z2M/README(about%20soil).md

What a fucking joke! May I remind you about support for z2m and zha 

[image]

marcgeerlings7, 2025-08-02:
Still got this issue, I'm on 2.6.0 z2m and with the external soil.js.

`[2025-08-02 13:54:14] z2m: Failed to configure 'Strelitzia', attempt 1 
(Error: Device 0x<device-id> has no input cluster msSoilMoisture
    at getEndpointsWithCluster (.../zigbee-herdsman-converters/src/lib/modernExtend.ts:61:15)
    at configure (.../zigbee-herdsman-converters/src/lib/modernExtend.ts:214:29)
    at Object.configure (.../zigbee-herdsman-converters/src/index.ts:363:27)
    at Configure.configure (.../lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (.../lib/eventBus.ts:252:17))`

3reality-support, 2025-08-04:
> Still got this issue, I'm on 2.6.0 z2m and with the external soil.js.
>
> `[2025-08-02 13:54:14] z2m: Failed to configure 'Strelitzia', attempt 1 (Error: Device 0x<device-id> has no input cluster [...]

Can you describe the phenomenon of the problem in detail? After reading the error message, it seems that there is no soil moisture value. Can be reset and re matched to try again

marcgeerlings7, 2025-08-12:
@3reality-support I’ve looked into it again, and when I do a request/get, I get the error.
[2025-08-12 22:54:26] z2m: Publish 'get' 'soil_moisture' to 'Scindapsus' failed: 'Error: ZCL command.
Could this have to do with the sensor only sending data and you can’t request anything via get? The configure error is gone btw. I didn't change a thing.