Koenkk/zigbee2mqtt #26919: Payload is not a number

ptr727, 2025-03-31:
### What happened?

I noticed lots of warnings like these in my HA logs:
```
2025-03-30 17:42:22.574 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"angle":79,"angle_x":8,"angle_x_absolute":82,"angle_y":1,"angle_y_absolute":89,"angle_z":-82,"battery":83,"device":{"applicationVersion":8,"friendlyName":"Aqara-Vibrate-F00199","ieeeAddr":"0x00158d000af00199","manufacturerID":4151,"manufacturerName":"LUMI","model":"DJT11LM","networkAddress":47569,"powerSource":"Battery","type":"EndDevice"},"device_temperature":28,"last_seen":"2025-03-31T00:42:22.573Z","linkquality":69,"power_outage_count":11,"sensitivity":"low","strength":9,"vibration":false,"voltage":2975,"x_axis":109,"y_axis":15,"z_axis":-762}' is not a Number
2025-03-30 17:46:24.031 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"angle":79,"angle_x":8,"angle_x_absolute":82,"angle_y":1,"angle_y_absolute":89,"angle_z":-82,"battery":83,"device":{"applicationVersion":8,"friendlyName":"Aqara-Vibrate-F00199","ieeeAddr":"0x00158d000af00199","manufacturerID":4151,"manufacturerName":"LUMI","model":"DJT11LM","networkAddress":47569,"powerSource":"Battery","type":"EndDevice"},"device_temperature":28,"last_seen":"2025-03-31T00:46:24.030Z","linkquality":65,"power_outage_count":11,"sensitivity":"low","strength":2,"vibration":false,"voltage":2975,"x_axis":109,"y_axis":15,"z_axis":-762}' is not a Number
```

Z2M docker: `[2.1.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.3) commit: [ba337bd329aeb4ca17735c0cf09b31293c8cff06](https://github.com/Koenkk/zigbee2mqtt/commit/ba337bd329aeb4ca17735c0cf09b31293c8cff06)`

HA docker: `2025.3.4`

### What did you expect to happen?

No warnings reported.

### How to reproduce it (minimal and precise)

Look at logs

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

docker

### Debug log

Warning appears in HA log, not in Z2M log?

clockbrain, 2025-04-06:
I expect the problem is the "sensitivity":"low" part of the message body.

From what I can deduce Z2M release 2.1.0 changed sensitivity for the DJT11LM from text values of low/medium/high to numeric values 1 to 21.
See release notes https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.0

There is a suggestion in a related thread that the dev console might let you change the device to report in numeric mode but perhaps just try to reconfigure or remove and add the sensor again.

ebertek, 2025-04-25:
Duplicate of #26164

ptr727, 2025-05-03:
Changing sensitivity changed format and made error go away, inconvenient and backwards compat. would be nicer, but closing.

closed 2025-05-03 by ptr727