Koenkk/zigbee2mqtt #26923: Phantom signals in Home Assistant after short disconnect of XZG-01 coordinator

0BanZai0, 2025-03-31:
### What happened?

In a span of weeks sometimes I'm getting "Phantom signals" from all connected Z2M devices after the connection to the coordinator drops for a short amount of time.

The UCG-01 coordinator is connected via POE LAN in my unify network to a Beelink EQ12 Mini-PC running HA OS in a Proxmox VM.

After reconnecting almost instantly, it seems that devices like hue bulbs just send their current state. 
But the real issue is with Green Power Devices like my PTM 215 switches. Home Assistant suddenly recieves the last known signal from the switch again. Resulting for example in blinds opening.

### What did you expect to happen?

Reconnect of the coordinator without sending any signals.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20241001

### Adapter

HamGeek UZG-01 POE Adapter 

### Setup

Home Assitant OS runnig Z2M in a Proxmox environment on a Beelink EQ12 Mini PC

### Debug log

The Z2M log only shows this:

```
[2025-03-31 16:55:20] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-03-31 16:55:20] error: 	z2m: Adapter disconnected, stopping
```

In Home Assistant is no error in the logfile. But the logbook is full of these events:

[image]

0BanZai0, 2025-05-03:
Update: Still present in 2.3.0. I've updated it some seconds ago and now some of my blinds are moving my themselves by signals that were not caused by button presses:

[image]

0BanZai0, 2025-05-21:
@Koenkk sorry for pinging you, it's quite an important matter in my case. It does not seem to affect many others though. 
Do you maybe have a hint for me how I could debug this behaviour? THX!

Koenkk, 2025-05-21:
Could you provide the debug log when this happens?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

0BanZai0, 2025-05-29:
<h3>Final update — solved without touching Zigbee2MQTT code ✨</h3>
<p>After a lot of digging (thx ChatGPT) the culprit was <strong>my PTM-215Z (Friends-of-Hue) blueprint</strong>, not Z-Stack or Z2M.</p>
<p><strong>Root cause</strong></p>
<ul>
<li>
<p>When the coordinator briefly reconnects, Home Assistant restores<br>
<code inline="">event.&lt;switch&gt;_action</code> from <code inline="">unavailable</code> to the <em>last</em> cached value<br>
(<code inline="">release_1_and_3</code> in my case).</p>
</li>
<li>
<p>The original blueprint triggered on <strong>any</strong> state change of that entity, so the restore itself fired the automation and the blinds moved.</p>
</li>
</ul>
<hr>
<h4>What I actually changed</h4>

Layer | Exact change | Reason
-- | -- | --
Blueprint (PTM215Z_simple_w-o_helper.yaml) | Replaced the trigger with a from / to whitelist:  yaml - platform: state   entity_id: !input 'switch'   attribute: event_type   from:   # only fire if we *came* from a valid event     - press_1     - press_2     - press_3     - press_4     - press_1_and_3     - press_2_and_4     - release_1     - release_2     - release_3     - release_4     - release_1_and_3     - release_2_and_4   to:     - press_1 … release_2_and_4 | Ignores the initial restore (unavailable → timestamp) but still reacts to every real button press/release.
Zigbee2MQTT config | Kept the safety net we discussed earlier:  yaml device_options:   filtered_cache:     - '^action$' | Ensures future action fields never get cached again. Harmless for all other devices.
Automations | Reloaded blueprints, re-loaded automations, restarted Z2M. | Makes the new trigger active.

<p>No other broker or firmware tweaks were needed.</p>
<hr>
<h4>Result</h4>
<ul>
<li>
<p>Restart Z2M → <strong>no blinds move</strong></p>
</li>
<li>
<p>Short LAN drop / HA reboot → <strong>no blinds move</strong></p>
</li>
<li>
<p>Real button actions → work instantly, as before.</p>
</li>
</ul>
<p>Thanks to everyone in the thread—this finally nails it.<br>
Feel free to close the issue; if someone can reproduce with the modified trigger plus <code inline="">filtered_cache</code>, let’s reopen.</p>
<p>Cheers,<br>
— 0BanZai0</p></body></html>
</body>
</html>