Koenkk/zigbee2mqtt #26936: AddOn not startinhg anymore aftern update to 2.2.0

Sesshoumaru-sama, 2025-04-01:
### What happened?

I tried to update my add-on from v1.42.0 to 2.2.0 just now - but start of new add-on failed with the following log:
(after restore of backup everything runs again on 1.42.0 .. fortunatly).

[21:26:51] INFO: Preparing to start...
[21:26:51] INFO: Socat not enabled
[21:26:51] ERROR: Got unexpected response from the API: Service not enabled
[21:26:51] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
TypeError: Cannot create property 'enabled' on boolean 'true'
    at iterate (/app/lib/util/settings.ts:394:59)
    at iterate (/app/lib/util/settings.ts:411:21)
    at iterate (/app/lib/util/settings.ts:411:21)
    at applyEnvironmentVariables (/app/lib/util/settings.ts:417:5)
    at Object.write (/app/lib/util/settings.ts:184:5)
    at onboard (/app/lib/util/onboarding.ts:538:18)
    at start (/app/index.js:137:31)
[21:26:54] INFO: Preparing to start...
[21:26:54] INFO: Socat not enabled
[21:26:54] ERROR: Got unexpected response from the API: Service not enabled
[21:26:54] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
TypeError: Cannot create property 'enabled' on boolean 'true'
    at iterate (/app/lib/util/settings.ts:394:59)
    at iterate (/app/lib/util/settings.ts:411:21)
    at iterate (/app/lib/util/settings.ts:411:21)
    at applyEnvironmentVariables (/app/lib/util/settings.ts:417:5)
    at Object.write (/app/lib/util/settings.ts:184:5)
    at onboard (/app/lib/util/onboarding.ts:538:18)
    at start (/app/index.js:137:31)

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on on VM

Nerivec, 2025-04-01:
Looks like you might have either `homeassistant: true` or `frontend: true` in your `configuration.yaml` that is conflicting with the add-on configuration page. Something apparently prevented the auto settings migration from working properly, you might have to change it manually.

Here are the related pages with up-to-date settings for reference:
https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html#home-assistant-integration
https://www.zigbee2mqtt.io/guide/configuration/frontend.html#frontend
Example:
```yaml
homeassistant: true
````
becomes in 2.x.x:
```yaml
homeassistant:
  enabled: true
````

(same syntax for `frontend`)

If that doesn't work, please post the content of your `configuration.yaml` and the content of the add-on configuration page.

Sesshoumaru-sama, 2025-04-02:
There is nothing like this in the configuration.yaml of HA but I found this in the addons configuration.yaml. Is this meant?

/homeassistant/zigbee2mqtt/configuration.yaml

`homeassistant: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ***************************************************************
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_d2f0b2b459d9eb11beb8178e6fe9f4d9-if00-port0
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: info
device_options:
  legacy: false`

Nerivec, 2025-04-02:
Try with (replace password of course):
/homeassistant/zigbee2mqtt/configuration.yaml
```yaml
version: 4
homeassistant:
  enabled: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ***************************************************************
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_d2f0b2b459d9eb11beb8178e6fe9f4d9-if00-port0
frontend:
  enabled: true
  port: 8099
advanced:
  log_level: info
```
This should manually bring your configuration to currently expected format.

opb, 2025-04-02:
> homeassistant:
>   enabled: true

@Nerivec thanks. Not OP but I had this issue too, and this config option fixed it for me. I was going off this thread - https://github.com/Koenkk/zigbee2mqtt/discussions/24198 - and the breaking change here doesn't seem to be mentioned. in the post. At least I dd ctrl+f for "enabled" and couldn't see on the page that this particular change had been covered.

thanks again

Nerivec, 2025-04-02:
The setting should auto-migrate but a change in 2.2.0 made the migration from 1.x.x straight to 2.2.0 a problem with some settings. It has been fixed in dev branch.

ismaelit, 2025-04-04:
I've edited the `configuration.yaml` file and changed `homeassistant: enabled: true` as instructed, which resolved the Zigbee2MQTT loading issue. Now, all my devices successfully report their online status and LQI when I manually press the physical button. However, when I try sending a command via Zigbee2MQTT, I receive errors and cannot control any device. Any ideas?

```
[2025-04-03 23:39:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Luz da Sacada', payload '{"countdown":0,"linkquality":224,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}'
[2025-04-03 23:39:45] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Luz da Sacada', payload '{"countdown":0,"linkquality":220,"power_on_behavior":"previous","state":"ON","switch_type":"toggle"}'
[2025-04-03 23:40:04] error: 	z2m: Publish 'set' 'state' to 'Luz da Sacada' failed: 'Error: ZCL command 0x04cd15fffe06839b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '18430'.)'
```

Sesshoumaru-sama, 2025-04-05:
Thanks, I managed to upgrade to 2.2.1.
Adding "adapter: zstack" did the trick to have it up and running.
Thanks 👍 

> Try with (replace password of course): /homeassistant/zigbee2mqtt/configuration.yaml
> 
> version: 4
> homeassistant:
>   enabled: true
> mqtt:
>   server: mqtt://core-mosquitto:1883
>   user: addons
>   password: ***************************************************************
> serial:
>   adapter: zstack
>   port: >-
>     /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_d2f0b2b459d9eb11beb8178e6fe9f4d9-if00-port0
> frontend:
>   enabled: true
>   port: 8099
> advanced:
>   log_level: info
> This should manually bring your configuration to currently expected format.
> 
> Note: I added `adapter: zstack`, since it should be the right one from the port you have configured (Dongle-P), but in case it is the Dongle-E variant, then you need to change that to `adapter: ember` instead. Note: the global HA `configuration.yaml` is irrelevant to Z2M, only the `configuration.yaml` in the zigbee2mqtt data folder is relevant.

closed 2025-04-05 by Sesshoumaru-sama