Koenkk/zigbee2mqtt #26937: Error when starting 2.2.0

PatMan6889, 2025-04-01:
### What happened?

I updated to 2.2.0 and get this error now when I try to start:
```
root@Zigbee2MQTT:/opt/zigbee2mqtt# cd /opt/zigbee2mqtt && pnpm start

    zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
    node index.js

Starting Zigbee2MQTT without watchdog.
TypeError: Module "file:///opt/zigbee2mqtt/package.json" needs an import assertion of type "json"
    at new NodeError (node:internal/errors:405:5)
    at validateAssertions (node:internal/modules/esm/assert:95:15)
    at defaultLoad (node:internal/modules/esm/load:86:3)
    at nextLoad (node:internal/modules/esm/loader:163:28)
    at ESMLoader.load (node:internal/modules/esm/loader:603:26)
    at ESMLoader.moduleProvider (node:internal/modules/esm/loader:457:22)
    at new ModuleJob (node:internal/modules/esm/module_job:64:26)
    at ESMLoader.#createModuleJob (node:internal/modules/esm/loader:480:17)
    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:434:34)
 ELIFECYCLE  Command failed with exit code 1.
```

### What did you expect to happen?

To just run fine like previous versions

### How to reproduce it (minimal and precise)

Upgrade from 2.1.0

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20230507

### Adapter

Sonoff Stick

### Setup

LXC on Proxmox

Koenkk, 2025-04-01:
Are you using any external converters or extensions?

PatMan6889, 2025-04-01:
no. just a pretty basic installation. I read something about configuration.yaml. Do I need to edit it?

njordan77, 2025-04-01:
oh jesus, was so happy for years with v1 branch and now with v2 every update is a nitemare. 
global environement variables, configuration.yaml not able to checkout,.......

Did start from scratch with 2.2.0 as 2.13 migration did not work at all....but now again im stuck at startup with the same config that worked on 2.1.3

Any idea what could be the issue...... Raspi5
```

pi@SmartPi:/opt/zigbee2mqtt $ sudo pnpm start

> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-01 22:29:03] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-01 22:29:03] info:     z2m: Starting Zigbee2MQTT version 2.2.0 (commit #c5c07e7d)
[2025-04-01 22:29:03] info:     z2m: Starting zigbee-herdsman (3.4.11)
```

Nerivec, 2025-04-01:
@PatMan6889 can you check the node version you are running? `node --version`
If v18, can you try updating to v20?

---

@njordan77 can you try forcing a rebuild of the serialport library?
```bash
rm -rf `find ./node_modules/.pnpm/ -wholename "*/@serialport/bindings-cpp/prebuilds" -type d`
pnpm rebuild @serialport/bindings-cpp
```

njordan77, 2025-04-01:
@Nerivec 

tried, made things worse....

```
pi@SmartPi:/opt/zigbee2mqtt $ sudo pnpm rebuild @serialport/bindings-cp
pi@SmartPi:/opt/zigbee2mqtt $ sudo pnpm start

> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.

/opt/zigbee2mqtt/node_modules/.pnpm/node-gyp-build@4.8.4/node_modules/node-gyp-build/node-gyp-build.js:60
  throw new Error('No native build was found for ' + target + '\n    loaded from: ' + dir + '\n')
        ^
Error: No native build was found for platform=linux arch=arm64 runtime=node abi=115 uv=1 armv=8 libc=glibc node=20.19.0
    loaded from: /opt/zigbee2mqtt/node_modules/.pnpm/@serialport+bindings-cpp@13.0.1/node_modules/@serialport/bindings-cpp

    at Function.load.resolve.load.path (/opt/zigbee2mqtt/node_modules/.pnpm/node-gyp-build@4.8.4/node_modules/node-gyp-build/node-gyp-build.js:60:9)
    at load (/opt/zigbee2mqtt/node_modules/.pnpm/node-gyp-build@4.8.4/node_modules/node-gyp-build/node-gyp-build.js:22:30)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/@serialport+bindings-cpp@13.0.1/node_modules/@serialport/bindings-cpp/dist/serialport-bindings.js:10:48)
    at Module._compile (node:internal/modules/cjs/loader:1529:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
    at Module.load (node:internal/modules/cjs/loader:1275:32)
    at Function.Module._load (node:internal/modules/cjs/loader:1096:12)
    at Module.require (node:internal/modules/cjs/loader:1298:19)
    at require (node:internal/modules/helpers:182:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/@serialport+bindings-cpp@13.0.1/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:5:31)
 ELIFECYCLE  Command failed with exit code 1.
```

Nerivec, 2025-04-01:
@njordan77 there's a typo in your rebuild command: `@serialport/bindings-cp` => `@serialport/bindings-cpp`
PS: You can revert by `rm -rf ./node_modules` and `pnpm i --frozen-lockfile` (it will re-install the default libraries).

njordan77, 2025-04-01:
@Nerivec 

sorry for the typo, did not see myself. but still did not change anything after building serials.....

```
pi@SmartPi:/opt/zigbee2mqtt $ pnpm rebuild @serialport/bindings-cpp
node_modules/.pnpm/@serialport+bindings-cpp@13.0.1/node_modules/@serialport/bindings-cpp: Running install script, done in 13.7s
pi@SmartPi:/opt/zigbee2mqtt $
pi@SmartPi:/opt/zigbee2mqtt $
pi@SmartPi:/opt/zigbee2mqtt $
pi@SmartPi:/opt/zigbee2mqtt $
pi@SmartPi:/opt/zigbee2mqtt $ sudo pnpm start

> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-02 00:26:19] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-02 00:26:19] info:     z2m: Starting Zigbee2MQTT version 2.2.0 (commit #c5c07e7d)
[2025-04-02 00:26:19] info:     z2m: Starting zigbee-herdsman (3.4.11)
```

Nerivec, 2025-04-01:
Can you run with `log_level: debug` so we can maybe see where it's stopping more precisely?

deflao, 2025-04-02:
@Nerivec could you show me how to update to v20 or v22?

morph027, 2025-04-02:
Same here. `log_level: debug` does not yield any more information.

```
> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-02 06:55:02] info:     z2m: Logging to console
[2025-04-02 06:55:02] debug:    z2m: Loaded state from file /var/lib/zigbee2mqtt/data/state.json
TypeError: Module "file:///opt/zigbee2mqtt/package.json" needs an import assertion of type "json"
    at new NodeError (node:internal/errors:405:5)
    at validateAssertions (node:internal/modules/esm/assert:94:15)
    at defaultLoad (node:internal/modules/esm/load:93:3)
    at DefaultModuleLoader.load (node:internal/modules/esm/loader:263:26)
    at DefaultModuleLoader.moduleProvider (node:internal/modules/esm/loader:179:22)
    at new ModuleJob (node:internal/modules/esm/module_job:63:26)
    at DefaultModuleLoader.#createModuleJob (node:internal/modules/esm/loader:203:17)
    at DefaultModuleLoader.getJobFromResolveResult (node:internal/modules/esm/loader:156:34)
    at DefaultModuleLoader.getModuleJob (node:internal/modules/esm/loader:141:17)
    at DefaultModuleLoader.import (node:internal/modules/esm/loader:227:28)
```

DJKatastrof, 2025-04-02:
I get the same error message as above

Yougoshatenshi, 2025-04-02:
> [@PatMan6889](https://github.com/PatMan6889) can you check the node version you are running? `node --version` If v18, can you try updating to v20 or v22?

updating from v18 to v22 helped for `TypeError: Module "file:///opt/zigbee2mqtt/package.json" needs an import assertion of type "json"`

To update: (taken from https://github.com/nodesource/distributions?tab=readme-ov-file#using-debian-as-root-nodejs-22 )
`curl -fsSL https://deb.nodesource.com/setup_22.x -o nodesource_setup.sh`
`sudo -E bash nodesource_setup.sh`
`sudo apt-get install -y nodejs`

morph027, 2025-04-02:
Indeed, upgrading node from 20 to 22 helped.

njordan77, 2025-04-02:
sorry to ask, but i m confused. is the node update a different issue or this could fix the stuck situation at startup?

Nerivec, 2025-04-02:
@njordan77 your issue is different from the rest of this thread. Can you set log level to `debug` and post the resulting logs from startup?

njordan77, 2025-04-02:
@Nerivec did change from info to debug. not much more insights - still stuck at this point.

```
> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-02 11:57:17] debug:    z2m: Removing old log directory '/opt/zigbee2mqtt/data/log/2025-04-01.23-06-31'
[2025-04-02 11:57:17] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-02 11:57:17] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2025-04-02 11:57:17] info:     z2m: Starting Zigbee2MQTT version 2.2.0 (commit #c5c07e7d)
[2025-04-02 11:57:17] info:     z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-02 11:57:17] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6772,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/opt/zigbee2mqtt/data/database.db\",\"databaseBackupPath\":\"/opt/zigbee2mqtt/data/database.db.backup\",\"backupPath\":\"/opt/zigbee2mqtt/data/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":false,\"path\":\"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'

```

Nerivec, 2025-04-02:
@njordan77 and after the v22 update, did you try rebuilding serialport again? Let's try a full cleanup, to be sure:
Cleanup local libraries cache
```bash
rm -rf $(pnpm store path)
```
Cleanup Z2M libraries
```bash
rm -rf node_modules
```
Install fresh Z2M libraries
```bash
pnpm i --frozen-lockfile
```
Remove existing serialport prebuilds
```bash
rm -rf `find ./node_modules/.pnpm/ -wholename "*/@serialport/bindings-cpp/prebuilds" -type d`
```
Rebuild serialport directly for your machine (keep the all logs from this one, in case it doesn't work):
```bash
pnpm rebuild @serialport/bindings-cpp
```
Rebuild Z2M
```bash
pnpm run prepack
```

njordan77, 2025-04-02:
@Nerivec did all as written above (thanks for the guidelines), but it did not change anything.

Nerivec, 2025-04-02:
@njordan77 let's continue in the other thread, the issue is similar (this one isn't): https://github.com/Koenkk/zigbee2mqtt/issues/26556#issuecomment-2770733422

PatMan6889, 2025-04-02:
> updating from v18 to v22 helped for `TypeError: Module "file:///opt/zigbee2mqtt/package.json" needs an import assertion of type "json"`
>
> To update: (taken from [...]

Unpacking nodejs (22.14.0-1nodesource1) over (18.17.1-deb-1nodesource1) ...

Came from 18.17.1, after upgrading its running. Everything seems ok so far. Thank you!

Koenkk, 2025-04-02:
This doesn't seem to happen with all 18.x version, e.g. `18.20.5` works (but 18.17.1) doesn't indeed. Given that NodeJS 18 is EOL anyway users having this issue are advised to upgrade to a newer Node version (preferably 22)

deflao, 2025-04-02:
Confirm that update to 22.x works. The only thing now is all devices appears offline, even though it still works

Nerivec, 2025-04-02:
@deflao the frontend offline display bug should be fixed in dev branch (https://github.com/nurikk/zigbee2mqtt-frontend/releases/tag/0.9.6).