Koenkk/zigbee2mqtt #26939: Error after updating to 2.2.0

webjib, 2025-04-01:
### What happened?

I've updated from 2.1.3, and the error occurred immediately in mosquito:
"Client mqttjs_**** disconnected due to malformed packet."

Works again when reverting back to Z2M 2.1.3.

### What did you expect to happen?

Normal publish on mqtt

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle P

### Setup

Add-on on Home Assistant

peterf9, 2025-04-01:
How to revert to Z2M 2.1.3. Haven't make backup. Please help

danroc, 2025-04-01:
I'm seeing the same error, but I'm using a ConBeeII (0x26780700). I can confirm that it works on 2.1.3 but not on 2.2.0.

Nerivec, 2025-04-01:
What's your version of mosquitto, and what's your `mqtt` section in configuration.yaml (apart from user/password)?
If you have the `debug` logs of the failed Z2M start, that would be good.

danroc, 2025-04-01:
I'm running mosquitto 2.0.21. Here is my `mqtt` section (without `user` and `password`):

```
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
```

danroc, 2025-04-02:
Here are the logs in `debug`: [log.log](https://github.com/user-attachments/files/19559904/log.log)

danroc, 2025-04-02:
Here are the logs in `debug`: [log.log](https://github.com/user-attachments/files/19559918/log.log)

sywesk, 2025-04-02:
Reverting works, seems like a mqttjs regression

quentinreytinas, 2025-04-02:
I'm experiencing also a persistent issue with my Zigbee2MQTT connection to Mosquitto MQTT broker in my Docker setup. The problem is as follows:

- Zigbee2MQTT attempts to connect to Mosquitto but is immediately disconnected
- Mosquitto logs show: "Client mqttjs_XXXXXXXX disconnected due to malformed packet"
- The error code shown is: "MQTT disconnect: reason 129 (undefined)"
- I've confirmed network connectivity between containers (ping works fine)
- I've tried reverting to older versions of both Zigbee2MQTT and Mosquitto, but the issue persists
- My setup runs on Docker containers for Home Assistant, Zigbee2MQTT, and Mosquitto. Nodered seems to have no issue with the broker mosquitto.

I've already tried:
- Updating the Mosquitto configuration
- Checking network connectivity between containers
- Adjusting MQTT settings in Zigbee2MQTT config
- Rolling back to previous working versions

Does anyone have experience with this specific error 129 and know how to resolve it? Any suggestions or debugging steps would be greatly appreciated

PEGARifrouf, 2025-04-02:
Hello,

Same issue here:
Upgrade Z2M from v2.1.3 -> v2.20

In the use of HA
Core: 2025.3.4
Supervisor: 2025.03.4
Operating System: 15.1

Zstack: CC2652P / Z-Stack_3.x.0_coordinator_20240710

Z2M is working, packets are refused by Mosquitto broker v6.5.0 with the following logs:
`
2025-04-02 10:13:34: New connection from 172.30.33.1:39492 on port 1883.
2025-04-02 10:13:34: New client connected from 172.30.33.1:39492 as mqttjs_5270ad4a (p2, c1, k60, u'mqtt_user').
2025-04-02 10:13:34: Client mqttjs_5270ad4a disconnected due to malformed packet.
2025-04-02 10:13:35: New connection from 172.30.33.1:39498 on port 1883.
2025-04-02 10:13:35: New client connected from 172.30.33.1:39498 as mqttjs_5270ad4a (p2, c1, k60, u'mqtt_user').
`

Did try
- Reboot Z2M -> same issue
- Reboot MQTT -> same issue

Note:
I have another Z2M v2.1.3 sending to the same MQTT,  messages are still sent and recieved correctly
I did not backup before upgrading Z2M addon, so I did not revert back

alefab, 2025-04-02:
Same error "z2m: Not connected to MQTT server!"
Using mosquitto version 2.0.20

sywesk, 2025-04-02:
@Nerivec I can confirm having a Lixee device in my network

paul9171, 2025-04-02:
I have the same problem, and a Lixee device too.

webjib, 2025-04-02:
Same, I’ve got a Lixee Linky device

PEGARifrouf, 2025-04-02:
I confirm that I have a Lixee device on the network.

alefab, 2025-04-02:
Lixee too for me

sywesk, 2025-04-02:
Ok so @Nerivec was right, the issue is tied to having lixee devices. I'll try to get a debug log asap

alefab, 2025-04-02:
@sywesk here is a tcpdump from the mqtt server
[mqtt.dmp](https://github.com/user-attachments/files/19567283/mqtt.dmp)

sywesk, 2025-04-02:
Just did my tshark, I think I'm getting what's happening. If I recall correctly from my MQTT days, "+" is a special character that means anything between two slashes, and is used when subscribing.

[image]

If you look closely, it's trying to publish to an illegal topic name: `homeassistant/sensor/0x00158d0005ce780b/NJOURF+1/config`

[export_mqtt_issue.pcapng.zip](https://github.com/user-attachments/files/19567350/export_mqtt_issue.pcapng.zip)

sywesk, 2025-04-02:
@alefab the `unique_id` found in the payload is not MQTT specific, but rather "home assistant" specific, so this shouldn't cause a disconnect from the mosquitto broker, but rather a bug with hass. You could send binary if you wanted the payload, it would work fine 👌

jojo01fr, 2025-04-02:
> FYI, I find in the logs the message 2025-04-02 15:18:56] warning: zhc:lixee: Not able to detect the current tarif. Not filtering any expose... This may cause the NJOURF+1 topic to be not excluded…

Hello I have exactly de same warning and got disconnected from mosquitto. Rolled back to 2.1.3 for the moment.

Diaoul, 2025-04-02:
Have the same issue with emqx and the cause is indeed the invalid topic name.

Koenkk, 2025-04-02:
I think I found the issue and fixed it, let me know if it works!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Diaoul, 2025-04-02:
Zibgee2mqtt should not event try to publish invalid topic names 🤔
Thanks for the swift action @Koenkk, I'll try that soon 🙏

danroc, 2025-04-02:
Thanks @Koenkk, I can confirm that it's working again on `latest-dev`!

CausevicAdin, 2025-04-03:
How do you switch to dev branch inside home assistant ?

paul9171, 2025-04-03:
Thanks @Koenkk, the problem is fixed in the dev branch. It would be a good idea to add a safety net to block the emission of invalid topic names just before sending… This little bug in Lixee caused the loss of all the Zigbee devices!

sywesk, 2025-04-03:
@Koenkk Why does the switch from `Object.hasOwn() to Object.hasOwnProperty.call()` fixes the issue? I'm not a JS expert, and I've read that hasOwn is meant to replace hasOwnProperty, so there must be something I don't understand here 🙏

gazm2k5, 2025-04-03:
@Koenkk Thank you, this is fixed for me by using the latest-dev branch.

I think it would be advisable to hotfix this in before the 1st of May for those using the main release. The bug completely broke all my devices in home assistant. Would be very annoying to have to wait a whole month for the fix.

CausevicAdin, 2025-04-03:
> [@Koenkk](https://github.com/Koenkk) Thank you, this is fixed for me by using the latest-dev branch.
>
> I think it would be advisable to hotfix this in before the 1st of May for those using the main [...]

Agree

Nerivec, 2025-04-03:
A hotfix release is already planned.
For HA, you can use the `edge` version of the addon, this is the equivalent of `latest-dev` branch.
https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

jojo01fr, 2025-04-03:
> How do you switch to dev branch inside home assistant ?

I suppose you need to go to Settings - Add-ons then "ADD-ON STORE" and it is there, called "Zigbee2MQTT Edge". Be aware to read the description, namely "Steps" if you dont want to lose your configuration. I prefer to stay at 2.1.3 for a month than risk huge problems.

Koenkk, 2025-04-03:
Great! I will implement a proper fix (to not publish those invalid topic and log an error instead later).

Will publish a hotfix release in the coming days.

closed 2025-04-03 by Koenkk

ffries, 2025-04-04:
I can confirm the same bug. Using Docker, how can I get the fix? Thank you.

quentinreytinas, 2025-04-04:
Je suppose que vous parlez français. Normalement vous mettez koenkk/zigbee2mqtt:latest pour lancer l'image zigbee2mqtt. Tout simplement ajoutez -dev à la fin : koenkk/zigbee2mqtt:latest-dev et vous serez sur la branche développement de zigbee2mqtt et le problème sera résolu. Quand une nouvelle image de koenkk/zigbee2mqtt:latest sera publiée, enlevez le -dev à la fin et vous repasserez à l'image stable. Évidement refaites un "stop", puis "rm" puis un "pull" pour télécharger l'image puis un "run" pour relancer le container ou alors un "docker compose up -d" si vous avez un fichier docker-compose.yaml

-------- Message d'origine --------

PEGARifrouf, 2025-04-05:
Hello.
Latest release this morning 2.2.1-1 fixes the issue.
Thanks @Koenkk !