Koenkk/zigbee2mqtt #26940: [New device support]: _TZE284_chbyv06x

danielgantzer, 2025-04-01:
### Link

https://vi.aliexpress.com/item/1005007128052490.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=61047789681&aff_fcid=57c08130ad694e4a85a54787e2a6e575-1743541440739-07449-_EyqfRgy&tt=MG&aff_fsk=_EyqfRgy&aff_platform=default&sk=_EyqfRgy&aff_trace_key=57c08130ad694e4a85a54787e2a6e575-1743541440739-07449-_EyqfRgy&shareId=61047789681&businessType=ProductDetail&platform=AE&terminal_id=6d55b6660fcd4db0aaa74f21aab7c782&afSmartRedirect=y

### Database entry

{"id":72,"type":"Router","ieeeAddr":"0xa4c138c1ba6e5d7f","nwkAddr":5797,"manufId":4417,"manufName":"_TZE284_chbyv06x","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\r@x/i\u000e@x/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":78}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1743420336629}

### Zigbee2MQTT version

 2.2.0 commit: c5c07e7d

### Comments

{"id":72,"type":"Router","ieeeAddr":"0xa4c138c1ba6e5d7f","nwkAddr":5797,"manufId":4417,"manufName":"_TZE284_chbyv06x","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\r@x/i\u000e@x/i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":78}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1743420336629} - from alliexpress

### External definition

```shell
-
```

### What does/doesn't work with the external definition?

i try but.... not working

zekino, 2025-04-02:
The same issue here. I reported this a month ago with a similar ticket (see [https://github.com/Koenkk/zigbee2mqtt/issues/26671](https://github.com/koenkk/zigbee2mqtt/issues/url)). I have already updated to yesterday's version, but the only change after successfully pairing this device is its status being "unsupported" and the Link Quality being visible. Nothing else.

danielgantzer, 2025-04-02:
I think it might be a problem with the connection method... I use ezsp... what do you use?

zekino, 2025-04-02:
I use this setup (Dockerized):
[image]

Koenkk, 2025-04-02:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

danielgantzer, 2025-04-03:
I check and working with dev branch - 

[image]