Koenkk/zigbee2mqtt #26942: All zigbee devices appear offline after upgrading to 2.2.0, but seem to work fine !

iparout, 2025-04-01:
### What happened?

After updating to Zigbeee2MQTT 2.2.0, all zigbee devices appear offline in HA ! They do seem to work fine though, and their automations also work, so I don't know if I should be worried or not and restore the previous version !

### Zigbee2MQTT version

2.2.0-1

### Adapter firmware version

Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Addon in Home Assistant

roydestroy, 2025-04-01:
I second this

REBELinBLUE, 2025-04-01:
I was just looking at this as well, I don't remember that column being there before. According to https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#device-availability the device availability feature is disabled by default, I wondered if the default had changed to true so I edited the config to add

```
availability:
    enabled: false
```

and restarted the add-on but it made no difference so I then tried adding `availability: false` under each device and restarted the add-on and now it is back to the old behaviour

So guessing something has broken with the default value

stuffandthat, 2025-04-01:
> I was just looking into this as I don't remember that column being there before. According to https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#device-availability the device availability feature is disabled by default, I wondered if the default had changed to true so I edited the config file (using the VSCode add-on) to add
> 
> ```
> availability:
>   enabled: false
> ```
> 
> and restarted the add-on but it made no difference so I then tried adding `availability: false` under each device and restarted the add-on and now it is back to the old behaviour
> 
> So guessing something has broken with setting the default value

The availability column has always been there for me, I've never changed any settings either.

stuffandthat, 2025-04-01:
> I was just looking into this as I don't remember that column being there before. According to https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#device-availability the device [...]

I thought I had that column always, but I just restored to the previous version and it's disappeared! Will wait for another update before going back to 2.2.0

kellery82, 2025-04-01:
My devices appeared Offline and were not responding. Weirdly enough, the logs in the add-on looked ok and I could see communication between my Zigbee devices. I've reverted back to the previous version for now, but will try again sometime today when i have more time

Nerivec, 2025-04-01:
Being investigated in https://github.com/Koenkk/zigbee2mqtt/discussions/26935#discussioncomment-12692831
Seems purely cosmetic (devices are fine).

cr0wm4n, 2025-04-01:
i have this issue too, everything shows offline but is actually working fine

kellery82, 2025-04-01:
> My devices appeared Offline and were not responding. Weirdly enough, the logs in the add-on looked ok and I could see communication between my Zigbee devices. I've reverted back to the previous version for now, but will try again sometime today when i have more time

false alarm, just showing offline and still working fine

rasturies, 2025-04-01:
Same here, all offline but working

RyTsYdUp, 2025-04-02:
Same issue here. I had to re-enable again. The upgrade turned it off.

Partizanin, 2025-04-02:
The same 

[image]

comdr-chocchip, 2025-04-02:
Happened to me too. But devices work

robsalasco, 2025-04-02:
+1 it's just the UI everything is working

Ropuh, 2025-04-02:
The same here - marked offline but working.

SmartSebbo, 2025-04-02:
Same Beauvoir in my system all devices flashes offline but work. I have a Backup.zip how can i restore that - i am an newbie.
Thank you for helping

GigiPompieru, 2025-04-02:
While looking into this since I have the same issue I saw something strange. In the settings page Availability is Active (I do not remember activating it):

[image]

But in the configuration there is no line for ```availability:```. In the past the column was not visible at all, now it displays Offline for all the devices.

I get that the issue is being resolved, but does anyone know why the difference between GUI and config file?

a1ex-ak, 2025-04-02:
[image]

HQ-smart, 2025-04-02:
Hi everyone. 

I had the same issue. You need to activate "availability".

[image]

SmartSebbo, 2025-04-02:
> Hi everyone.
> 
> I had the same issue. You need to activate "availability".
> 
> [image]

IT works - thank you!

hughc-hub, 2025-04-02:
Same here.. Devices show OFFLINE in the screen, but they were working during a couple of hours.. Now they are not.. none of them
I rollback to previous version..

Mark-Muc, 2025-04-02:
i do have the same issue!!
Mark-Muc

HQ-smart, 2025-04-02:
USB hub buttons were "left, centre, right", and became l1, l2, l3 

:(

stuffandthat, 2025-04-02:
> Same issue here. All devices went to offline status but were working fine. I had to re-enable the availability option in settings again. The upgrade seems to have turned it off.

How did you re-enable in the settings?

rizzification, 2025-04-02:
> Hi everyone.
> 
> I had the same issue. You need to activate "availability".
> 
> [image]

Can confirm this worked! thanks very much.

markulatus, 2025-04-02:
> Hi everyone.
> 
> I had the same issue. You need to activate "availability".
> 
> [image]

I think that's the part of the "RAW" config that handles this setting --> 

        },
        "availability": {
            "active": {
                "timeout": 10
            },
            "enabled": true,
            "passive": {
                "timeout": 1500
            }

major328, 2025-04-02:
Same issue here

puttemanp, 2025-04-02:
Enabling availability will fix this cosmetic issue. However, if you have a large Zigbee mesh it will considerably slow down your network as every device will be pinged according to the interval you set in the availability settings.
It will even slow down more if you have a few devices that don't reply to the ping request the first time and have to be retried (battery opperated devices tend to do this).
As it is cosmetic, it should go back to the previous way of visualising the UI: only display the column 'availability' if Availability is enabled in settings. 
I have it disabled for performance reasons and I hate red flashing warnings on the UI.

mattcartlidge, 2025-04-02:
Same issue. I have reverted back as @puttemanp mentions this slows down my network to a noticeable level and my OCD prevents me from having flashing red text on the list!

uninc80, 2025-04-02:
Same issue here, because is only a cosmetic issue I keep it and wait for a new fix

JohanKrab, 2025-04-02:
Same here with the Popp 701721 thermostates, restored backup. Wait for fix.

MoWeBo, 2025-04-02:
Same here, waiting for a fix.

fastsonic, 2025-04-02:
Also from my side .
Most Devices are dissapear. Some still Present in the GUI 10 from 70 Devices. They are all working so i still waiting for an Update, Fix, Or should we go back ?

Partizanin, 2025-04-02:
> Hi everyone.
> 
> I had the same issue. You need to activate "availability".
> 
> [image]

It helps, thank you !

fastsonic, 2025-04-02:
> > Hi everyone.
> > I had the same issue. You need to activate "availability".
> > [image]
> 
> It helps, thank you !

From my side it only helps with the Devices that are still under devicess shown and offline. But i see only 8 Devices from 70.

EcoNino, 2025-04-02:
Same here, but pay attention! it appears all is working ok. but with a Sonoff sensor the temperature showing in display is different than the one is showing in HA so it appears not updating ok. Reversed to 2.1.3-1 it backs updating ok.

fastsonic, 2025-04-02:
Now like 2 Hours gone, The Devices back again. Now it seems to be ok.

GigiPompieru, 2025-04-02:
> Enabling availability will fix this cosmetic issue. However, if you have a large Zigbee mesh it will considerably slow down your network as every device will be pinged according to the interval you [...]

yea, disabled it after a few hours.
the wort part is that if a device does not reply to the ping they get marked as unavailable in HA instead of exposing the last state.

I can live with offline in the Z2M interface rather then seeing unavailable entities in my dashboards.

rambosheep, 2025-04-02:
also its killed the batteries in my SNZB-02D most have dropped from like 80% to 9-11% since the firmware update so either it wasnt correctly reporting battery life or the update really has drained the battery

fastsonic, 2025-04-02:
for alle where the Devices are no longer present under devices..
Reset Browsercache ! 
Reload is not enough. All devices are back.

dsimons2710, 2025-04-02:
I have the same problem, but I can't really find a solution to prevent the offline message from appearing. Does anyone have a solution?

gohakn, 2025-04-02:
Same for me, but with the two hints from @fastsonic ( activate availability + clean your Browser cache and reload HA) worked fine.

Just had to restart Zigbee2MQTT one Time.

[image]

bdenski18, 2025-04-02:
I ended up rolling back the update. I see some are suggesting "Enabling the Availability", but I also see some of you saying this isn't really a fix, and can actually reduce performance. I'll check back when there is a confirmed resolution.

GigiPompieru, 2025-04-02:
> I ended up rolling back the update. I see some are suggesting "Enabling the Availability", but I also see some of you saying this isn't really a fix, and can actually reduce performance. I'll check back when there is a confirmed resolution.

the fact they appear offline has no impact, it's just visual. if you spend so much time in the web page that it bothers you, ok. other than that, there is no real reason to rollback.

PV-Joe, 2025-04-02:
Same here. Did a rollback to previous version.

Stony111, 2025-04-02:
Same behaviour in my system too. It seems actually it is only a design behaviour - all devices works fine. So I wait for the next Bugfixrelease. If I follow the issues here on git, it seems there are some troubles in the current release.

puttemanp, 2025-04-02:
> the fact they appear offline has no impact, it's just visual. if you spend so much time in the web page that it bothers you, ok. other than that, there is no real reason to rollback.

It's more than just visual, a lot of users have automations to send out a notification when a device is offline, or has been offline for more than X time (either just because it's possible, of for a genuine reason. Example, if a garage door opener goes offline, you might want to know it and solve the problem before wanting to leave with your car or park it and being unable to open the door)
This has now become impossible.

uninc80, 2025-04-02:
At the end I decided to go back and restore previous versions from backup. Let's wait for a newer better version

Koenkk, 2025-04-02:
Fixed in https://github.com/nurikk/zigbee2mqtt-frontend/commit/7e72b00c55012e6d44d3e1856d34a667b9479928

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

A hotfix release will be available in the coming days.

**Note that this is just a visual issue, the devices are not actually offline!**

closed 2025-04-02 by Koenkk