Koenkk/zigbee2mqtt #26943: secret.yaml overwritten

Haraade, 2025-04-01:
### What happened?

latest version performs overwriting of secret.yaml 
Add-on Home Assistant.

configuration.yaml
mqtt:
    server: '!secret.yaml server'
    user: '!secret.yaml user'
    password: '!secret.yaml password'

secret.yaml
server: 'mqtt://localhost:1883'
user: mqtt_user
password: mqtt_password

After upgrade to latest version (2.2.0)
secret.yaml
server: '!secret.yaml server'
user: '!secret.yaml user'
password: '!secret.yaml password'

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

8.0.2

### Adapter

Ember

### Setup

Home Assistant

Koenkk, 2025-04-02:
- Do you have the MQTT addon config installed?
- Could you provide a screenshot of the addon config? (including `Show unused optional configuration options`)

Haraade, 2025-04-02:
Installed add-on Mosquitto broker 6.5.0 and with default settings.

[image]
[image]

Nerivec, 2025-04-03:
I cannot replicate this, I've tried several ways to configure as indicated in first post:
- using carbon-copy of the first post
- using environment variables on top of yaml
- using environment variables in place of yaml

On top of what Koenkk asked, can you also post your `configuration.yaml` file (_you can hide security-related fields_)?

ash286, 2025-04-04:
> What do you have in the settings of the Zigbee2MQTT addon?
> 
> > Could you provide a screenshot of the addon config? (including Show unused optional configuration options)
> 
> CC: [@ash286](https://github.com/ash286)

@Koenkk Please see the `configuration.yaml` file below. Previously, I was using `EZSP ` driver and now `ember`. This issue replicates with `EZSP` as well as `ember`. Due to this issue I have now reverted back to the previous version - please see the screenshot below.

@Nerivec Please refer to the issue I opened #26950 

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://homeassistant:1883
  user: "!/config/secrets.yaml zigbee-user"
  password: "!/config/secrets.yaml zigbee-password"
serial:
  port: /dev/ttyACM0
  adapter: ember
advanced:
  channel: 25
  network_key: "!/config/secrets.yaml zigbee_key"
zigbee_herdsman_debug: false
```

[image]

Koenkk, 2025-04-04:
Fixed in https://github.com/Koenkk/zigbee2mqtt/pull/26988, I will release a hotfix release now.

closed 2025-04-04 by Koenkk

ash286, 2025-04-04:
> Fixed in [#26988](https://github.com/Koenkk/zigbee2mqtt/pull/26988), I will release a hotfix release now.

That's wonderful. Thank you for fixing it :)

PR #26988 (2025-04-04): fix: Fix settings being overwriting when env var is set to a ref
Fixes #26943