Koenkk/zigbee2mqtt #26945: Vimar RemoteControl_v1.0 goes offline after some time

saveriol, 2025-04-02:
### What happened?

VIMAR released zigbee firmware for their [RemoteControl_v1.0](https://www.zigbee2mqtt.io/devices/RemoteControl_v1.0.html#vimar-remotecontrol-v1-0) [(link to Vimar Website)](https://www.vimar.com/en/int/catalog/product/index/code/30801) which correctly connects to Z2M, correctly configures, but goes offline after a random time in hours.
Then it stays offline until you repeatedly press the buttons for one minute, during which:
- the buttons flashes blue but Z2M doesn't receive any message
- then in the end the light stays off when you press the buttons (like the device is stuck)
- then it returns online.

I think there is some incompatibility between Z2M and VIMAR devices, eg #20045 #26676 

I see @sciaccom-vimar added support on SmartThingsEdgeDrivers repository and I assume he works for Vimar: Mariano can you please get onboard of this awesome open-source project to enhance Vimar compatibility?

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

7.4.3 [GA]

### Adapter

Home Assistant Yellow

### Setup

Home Assistant Yellow

LeoeLeoeL, 2025-04-02:
I hope somebody will take care (sooner or later) of Vimar devices because it seems they are overlooked.
E.g. https://github.com/Koenkk/zigbee2mqtt/issues/26676

saveriol, 2025-04-03:
I also think there's not so much transparency by Vimar regarding firmwares. They claim firmware is fully compatible with Zigbee 3.0, but is that true given these results? And they're not releasing any changelog about their firmware updates.

LeoeLeoeL, 2025-04-03:
I spoke to Vimar by phone many years ago. They test their devices on Amazon Echo zigbee devices only!!
Anyway, I think the coordinators based on CC2652P (Sonoff Dongle P) are guilty because now with Sonoff Dongle E I have much less losses of devices. Sometimes network is perfect and rarely not; With the previous one I had many many problems due to several disconnections of devices and, worse, Z2M crashed.

saveriol, 2025-04-03:
@LeoeLeoeL did you solve also #26676 with Dongle E?

LeoeLeoeL, 2025-04-03:
No, same behaviour.
I think the converter isn't ok anymore.
Firmware, even if changed something, is not guilty because the device continues to work under deconz.

saveriol, 2025-04-05:
This issue is still present using a different coordinator (Sonoff ZBDongle-E) on Z2M 2.2.1.

closed 2025-06-12 by github-actions[bot]

saveriol, 2025-09-08:
Yes they worked, but went offline after a while as per this issue.
They have to be used as a trigger for automations, you won't "see" anything in the logs.

Anyway I stopped trying to make them work with zigbee2mqtt, since even enabling debug logs I didn't see anything.
I paired them with ZHA and they work flawlessly since then. 
Evidently Z2M is not compatible with VIMAR products.

Thehole0, 2025-12-07:
Hi all guys!

I'm also in the same situation as you guys, i recently renovated my home and basically I have 60 (more or less) Vimar devices, 30 of them are a mix of dimmers, roller shutters and on/off switches and they works just fine, the remaining one are all remote control switches.
At the moment i tried only with the skyconnect (v1) adapter and i'm still experiencing issues only with the remote control switches, I bought a new coordinator, the smlight slzb-06p7 and did some testing, but at the moment is behaving worse than the skyconnect one, because of range (still investigating on that).
I will try to move the switches from z2m to zha to see if the will work fine this way.
But in the meantime if you have any idea or you want to do some additional tests just let me know! I would really like to use z2m only!
I will also try to open a casa with Vimar too