Koenkk/zigbee2mqtt #26946: ZB2MQTT 2.2.0 cannot start Could not init sd_notify: Could not locate the bindings file

Nardol, 2025-04-02:
### What happened?

Since I updated to 2.2.0, the Zigbee2mqtt systemd service cannot run anymore. It worked just fine before.

### What did you expect to happen?

I expected that it could run as before

### How to reproduce it (minimal and precise)

Update to Zigbee2MQTT 2.2.0 on an installtion using the Linux installation method

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20240710

### Adapter

Sonof dongle not ezsp

### Setup

Zigbee2MQTT non-containerized on an Odroid N2+ under Raspbian, Node 22

### Debug log

```txt
[02/04/2025 10:38:23] error: 	z2m: Could not init sd_notify: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/out/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/out/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/default/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/compiled/22.14.0/linux/arm64/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/release/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/debug/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/default/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/lib/binding/node-v127-linux-arm64/unix_dgram.node
```

Nardol, 2025-04-02:
Debug log related to this error is a follow:

```txt
[02/04/2025 11:55:16] error: 	z2m: Could not init sd_notify: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/out/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/out/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/default/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/compiled/22.14.0/linux/arm64/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/release/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/debug/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/default/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/lib/binding/node-v127-linux-arm64/unix_dgram.node
[02/04/2025 11:55:16] debug: 	z2m: Error: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/out/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/Debug/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/out/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/Release/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/default/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/compiled/22.14.0/linux/arm64/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/release/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/debug/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/addon-build/default/install-root/unix_dgram.node
 → /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/lib/binding/node-v127-linux-arm64/unix_dgram.node
    at bindings (/opt/zigbee2mqtt/node_modules/.pnpm/bindings@1.5.0/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/lib/unix_dgram.js:4:34)
    at Module._compile (node:internal/modules/cjs/loader:1554:14)
    at Object..js (node:internal/modules/cjs/loader:1706:10)
    at Module.load (node:internal/modules/cjs/loader:1289:32)
    at Function._load (node:internal/modules/cjs/loader:1108:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:220:24)
    at cjsLoader (node:internal/modules/esm/translators:262:5)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:196:7)
```

Lines before and after are not related.

Nardol, 2025-04-02:
I can start Zigbee2MQTT by running `node index.js" by hand so until a solution can be found I started using a screen which is not a solution at all.

Nardol, 2025-04-02:
What a fantastic version!
With all these failures, now my network is fully down!

```txt[02/04/2025 12:21:19] error: 	z2m: Error while starting zigbee-herdsman
[02/04/2025 12:21:19] error: 	z2m: Failed to start zigbee-herdsman
[02/04/2025 12:21:19] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[02/04/2025 12:21:19] error: 	z2m: Exiting...
[02/04/2025 12:21:19] error: 	z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Object.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZnpAdapterManager.beginCommissioning (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:372:31)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:104:13)
    at start (/opt/zigbee2mqtt/index.js:149:5))
    at ZnpAdapterManager.beginCommissioning (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:374:23)
    at ZnpAdapterManager.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:104:13)
    at start (/opt/zigbee2mqtt/index.js:149:5)
```

Because of a new feature only there to remove a dependency I myght have to paire everything again, what a fantastic idea!
It is not in my habit to post these kind of comment here but this version is simply a nightmare for me, first time it happens but what first time!

Nardol, 2025-04-02:
Since I (as it looks like) have restored my network by setting another channel, starting zb2mqtt, stopping, change back to the original channel then startarted again, I tried to rollback to 2.1.3.
Before the update, it worked, now, it restart in loop even with 1.1.3.

My systemd unit is the following:

```ini
Description=zigbee2mqtt
After=network.target mosquitto.service

[Service]
Environment=ZIGBEE2MQTT_DATA=/home/homeassistant/config/zigbee2mqtt
Environment=NODE_ENV=production
Environment=Z2M_WATCHDOG=default
Type=notify
ExecStart=/usr/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
#StandardOutput=null
#StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=homeassistant
#CPUSchedulingPolicy=rr
#CPUSchedulingPriority=80

[Install]
WantedBy=multi-user.target
```

So it worked before 2.2.0, I was able to use Zigbee2MQTT as expected but since the update, even with a rollback it does not work anymore. Very strange but more anoying.
It looks it is really SystemD which is not notified and which restarts Zigbee2MQTT automatically.

Koenkk, 2025-04-02:
Suggestion by @Nerivec  in https://github.com/Koenkk/zigbee2mqtt/discussions/26935#discussioncomment-12705250

>Can you try this?
>```bash
>pnpm i --frozen-lockfile
>```
>```bash
>pnpm rebuild unix-dgram
>```

Nardol, 2025-04-02:
I have updated from my previous installation from scratch but cannot reproduce my issue, Zigbee2MQTT starts again. A strange issue from the begining to the end, I'm realy sorry for all the noise I made around this issue.

closed 2025-04-02 by Nardol

Nerivec, 2025-04-02:
For future reference, this can help when dependencies are misbehaving (it clears the local cache completely):
```bash
rm -rf $(pnpm store path)
```

a2nt, 2025-05-15:
I had the same issue nothing above have fixed it.

So here's the solution:

pnpm store prune
pnpm i --frozen-lockfile
pnpm approve-builds

synap5e, 2025-05-19:
Ran into this while upgrading from 1.4.3 to 2.3.0/master.

Turns out it was something about my old logging settings in configuration.yaml (set to use console and syslog).
I removed all logging config and it worked. Have added back console + file with some different settings.

Also maybe I missed it listed as a breaking change, but I ran into my `pan_id: '!secret.yaml pan_id'` needing to be converted to directly included in configuration.yaml.

ikey-ru, 2025-07-18:
Zigbee2MQTT 2.5.1
ZStack30x - 20201010

I have the same problem. It works, but it constantly restarts.
Removing everything unnecessary from the configuration does not solve the problem.

These commands don't help either.
pnpm i --frozen-lockfile
pnpm rebuild unix-dgram

config:
```
homeassistant:
  enabled: true
advanced:
  log_level: error
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
  user: ik
  password: xxxxxxxx
serial:
  port: /dev/ttyACM0
frontend:
  enabled: true
  port: 8080
  host: 192.168.50.61
devices:
  '0x00124b0022e81116':
    friendly_name: '0x00124b0022e81116'
  '0x00124b0022e9f719':
    friendly_name: '0x00124b0022e9f719'
  '0x0c4314fffe5d955a':
    friendly_name: '0x0c4314fffe5d955a'
version: 4
```

problem:

[image]

therealbat, 2025-07-21:
in my case the issue was that contrary to what it seems, unix-dgram module is not installed correctly. no matter how many times you rebuild it, it will fail the same way - unless you have python3 installed beforehand. let's clean up unix-dgram first and try to reinstall it:

`[z2m@sarkad-hub node_modules]$ rm unix-dgram
[z2m@sarkad-hub node_modules]$ rm -r .pnpm/unix-dgram\@2.0.6
[z2m@sarkad-hub node_modules]$ pnpm i unix-dgram
Packages: +1
+
Progress: resolved 322, reused 273, downloaded 0, added 1, done
node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram: Running install script, failed in 348ms
.../node_modules/unix-dgram install$ node-gyp rebuild
│ gyp info it worked if it ends with ok
│ gyp info using node-gyp@11.1.0
│ gyp info using node@22.16.0 | linux | x64
│ gyp ERR! find Python 
│ gyp ERR! find Python Python is not set from command line or npm configuration
│ gyp ERR! find Python Python is not set from environment variable PYTHON
│ gyp ERR! find Python checking if "python3" can be used
│ gyp ERR! find Python - executable path is ""
│ gyp ERR! find Python - "" could not be run
│ gyp ERR! find Python checking if "python" can be used
│ gyp ERR! find Python - executable path is ""
│ gyp ERR! find Python - "" could not be run
│ gyp ERR! find Python 
│ gyp ERR! find Python **********************************************************
│ gyp ERR! find Python You need to install the latest version of Python.
│ gyp ERR! find Python Node-gyp should be able to find and use Python. If not,
│ gyp ERR! find Python you can try one of the following options:
│ gyp ERR! find Python - Use the switch --python="/path/to/pythonexecutable"
│ gyp ERR! find Python (accepted by both node-gyp and npm)
│ gyp ERR! find Python - Set the environment variable PYTHON
│ gyp ERR! find Python - Set the npm configuration variable python:
│ gyp ERR! find Python npm config set python "/path/to/pythonexecutable"
│ gyp ERR! find Python For more information consult the documentation at:
│ gyp ERR! find Python https://github.com/nodejs/node-gyp#installation
│ gyp ERR! find Python **********************************************************
│ gyp ERR! find Python 
│ gyp ERR! configure error 
│ gyp ERR! stack Error: Could not find any Python installation to use
│ gyp ERR! stack at PythonFinder.fail (/opt/zigbee2mqtt/.cache/node/corepack/v1/pnpm/10.13.1/dist/node_modules/node-gyp/lib/find-python.js:306:11)
│ gyp ERR! stack at PythonFinder.findPython (/opt/zigbee2mqtt/.cache/node/corepack/v1/pnpm/10.13.1/dist/node_modules/node-gyp/lib/find-python.js:164:17)
│ gyp ERR! stack at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
│ gyp ERR! stack at async configure (/opt/zigbee2mqtt/.cache/node/corepack/v1/pnpm/10.13.1/dist/node_modules/node-gyp/lib/configure.js:27:18)
│ gyp ERR! stack at async run (/opt/zigbee2mqtt/.cache/node/corepack/v1/pnpm/10.13.1/dist/node_modules/node-gyp/bin/node-gyp.js:81:18)
│ gyp ERR! System Linux 6.15.3-arch1-1
│ gyp ERR! command "/usr/bin/node" "/opt/zigbee2mqtt/.cache/node/corepack/v1/pnpm/10.13.1/dist/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
│ gyp ERR! cwd /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram
│ gyp ERR! node -v v22.16.0
│ gyp ERR! node-gyp -v v11.1.0
│ gyp ERR! not ok 
└─ Failed in 349ms at /opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram
 ELIFECYCLE  Command failed with exit code 1.
`

the funny thing is, that you actually need to remove the directory and the symlink from node_modules otherwise this error message never shows up.
but after installing python3, it all works out well:
`
sudo  pacman -S python3
resolving dependencies...
looking for conflicting packages...

Packages (2) mpdecimal-4.0.1-1  python-3.13.5-1

Total Download Size:   12.28 MiB
Total Installed Size:  67.86 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 python-3.13.5-1-x86_64                                                                                    12.3 MiB  9.48 MiB/s 00:01 [#################################################################################] 100%
(2/2) checking keys in keyring                                                                                                        [#################################################################################] 100%
(2/2) checking package integrity                                                                                                      [#################################################################################] 100%
[... 8 lines cut ...]
    python-pip: for installing Python packages using tooling that is usually bundled with Python
    python-pipx: for installing Python software not packaged on Arch Linux
    sqlite: for a default database integration [installed]
    xz: for lzma [installed]
    tk: for tkinter
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
[z2m@sarkad-hub node_modules]$
[z2m@sarkad-hub node_modules]$  rm -r .pnpm/unix-dgram\@2.0.6 unix-dgram
rm: cannot remove '.pnpm/unix-dgram@2.0.6': No such file or directory
rm: cannot remove 'unix-dgram': No such file or directory

[z2m@sarkad-hub node_modules]$ pnpm install unix-dgram
Packages: +1
+
Progress: resolved 322, reused 273, downloaded 0, added 1, done
node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram: Running install script, done in 15.2s <-- this here

dependencies:
+ unix-dgram 2.0.6

Done in 25.4s using pnpm v10.13.1
`

and afterwards the original sd_notify based systemd service script works as expected: it doesn't hang at start and you don't see the errors anymore about the missing bindings.

larrykluger, 2025-08-13:
# Solved!

I had the same problem (no bindings file **`unix_dgram.node`**).

Various re-install attempts for the `unix-dgram` NPM module listed above didn't work. 

**Diagnosis:** For unknown reasons, the unix-dgram module installation is not compiling/building the c code file into the `unix_dgram.node` file (and is not complaining about any build issues). 

## Solution: build unix_dgram.node yourself

````
## move to the installation dir
larryk@zigbee-gway:/home/larryk $ cd /opt/zigbee2mqtt
larryk@zigbee-gway:/opt/zigbee2mqtt $ node-gyp
-bash: node-gyp: command not found

## install node-gyp
larryk@zigbee-gway:/opt/zigbee2mqtt $ pnpm install -g node-gyp

## move to the unix_dgram NPM module directory. It includes the binding.gyp file
larryk@zigbee-gway:/opt/zigbee2mqtt $ cd node_modules/unix-dgram

## configure node-gyp
larryk@zigbee-gway:/opt/zigbee2mqtt/node_modules/unix-dgram $ node-gyp configure
## NOTE: this takes awhile since gyp downloads Node bindings files. 
## This step will also complain if Python, etc are not found

## build the unix_dgram.node file
larryk@zigbee-gway:/opt/zigbee2mqtt/node_modules/unix-dgram $ node-gyp build

## if it worked, then locate the new unix_dgram.node file.
## on my vanilla system, the file was added to the new build/Release/ directory
## in the unix-dgram directory:
larryk@zigbee-gway:/opt/zigbee2mqtt/node_modules/unix-dgram $ ls build/Release/
obj.target  unix_dgram.node
````

The [`bindings` NPM module](https://github.com/bnoordhuis/node-unix-dgram/blob/f52b70ad15328b9739543838fb61b2fc4fd01b90/lib/unix_dgram.js#L4) automatically looks in the project's `node_modules/unix-dgram/build/Release/` directory, so all is good.

HTH!

### node-gyp configure and build logs
````
larryk@zigbee-gway:/opt/zigbee2mqtt/node_modules/unix-dgram $ node-gyp configure
gyp info it worked if it ends with ok
gyp info using node-gyp@11.3.0
gyp info using node@20.19.2 | linux | arm64
gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"

gyp http GET https://nodejs.org/download/release/v20.19.2/node-v20.19.2-headers.tar.gz
gyp http 200 https://nodejs.org/download/release/v20.19.2/node-v20.19.2-headers.tar.gz
gyp http GET https://nodejs.org/download/release/v20.19.2/SHASUMS256.txt
gyp http 200 https://nodejs.org/download/release/v20.19.2/SHASUMS256.txt
gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args '/home/larryk/.local/share/pnpm/global/5/.pnpm/node-gyp@11.3.0/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args 'binding.gyp',
gyp info spawn args '-f',
gyp info spawn args 'make',
gyp info spawn args '-I',
gyp info spawn args '/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/config.gypi',
gyp info spawn args '-I',
gyp info spawn args '/home/larryk/.local/share/pnpm/global/5/.pnpm/node-gyp@11.3.0/node_modules/node-gyp/addon.gypi',
gyp info spawn args '-I',
gyp info spawn args '/home/larryk/.cache/node-gyp/20.19.2/include/node/common.gypi',
gyp info spawn args '-Dlibrary=shared_library',
gyp info spawn args '-Dvisibility=default',
gyp info spawn args '-Dnode_root_dir=/home/larryk/.cache/node-gyp/20.19.2',
gyp info spawn args '-Dnode_gyp_dir=/home/larryk/.local/share/pnpm/global/5/.pnpm/node-gyp@11.3.0/node_modules/node-gyp',
gyp info spawn args '-Dnode_lib_file=/home/larryk/.cache/node-gyp/20.19.2/<(target_arch)/node.lib',
gyp info spawn args '-Dmodule_root_dir=/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram',
gyp info spawn args '-Dnode_engine=v8',
gyp info spawn args '--depth=.',
gyp info spawn args '--no-parallel',
gyp info spawn args '--generator-output',
gyp info spawn args 'build',
gyp info spawn args '-Goutput_dir=.'
gyp info spawn args ]
gyp info ok 
larryk@zigbee-gway:/opt/zigbee2mqtt/node_modules/unix-dgram $ node-gyp build
gyp info it worked if it ends with ok
gyp info using node-gyp@11.3.0
gyp info using node@20.19.2 | linux | arm64
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build'
  CXX(target) Release/obj.target/unix_dgram/src/unix_dgram.o
In file included from ../../../../nan@2.22.2/node_modules/nan/nan.h:62,
                 from ../src/unix_dgram.cc:5:
/home/larryk/.cache/node-gyp/20.19.2/include/node/node.h:1226:7: warning: cast between incompatible function types from 'void (*)(v8::Local<v8::Object>)' to 'node::addon_register_func' {aka 'void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)'} [-Wcast-function-type]
 1226 |       (node::addon_register_func) (regfunc),                          \
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/larryk/.cache/node-gyp/20.19.2/include/node/node.h:1260:3: note: in expansion of macro 'NODE_MODULE_X'
 1260 |   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
      |   ^~~~~~~~~~~~~
../src/unix_dgram.cc:404:1: note: in expansion of macro 'NODE_MODULE'
  404 | NODE_MODULE(unix_dgram, Initialize)
      | ^~~~~~~~~~~
  SOLINK_MODULE(target) Release/obj.target/unix_dgram.node
  COPY Release/unix_dgram.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build'
gyp info ok 
larryk@zigbee-gway:/opt/zigbee2mqtt/node_modules/unix-dgram $ 
````

bigjsl, 2025-11-12:
Prior to having this problem I installed the dev version over the production version. Then I reverted to the production version. I did not have this problem doing a clean install on another device. Interesting learning experience but it would have been faster to reimage the SD card and start again. Thanks for the solution everyone, especially retikulumx.