Koenkk/zigbee2mqtt #26948: Not able to pair any device anymore

tvdhorst, 2025-04-02:
### What happened?

I bought an IKEA Tretakt plug and wasn't able to pair it. An aqara sensor on the network was functioning as expected at that time (the only active device on the network at the time). 

I ended up uninstalling Z2M and removing all configs to start over. But the problem remains. Whatever I try nothing happens: I enable "Permit join (all)" and push the pairing button on the Tretakt, but nothing happens. Not even a single line of code in the log (debug level). Additionally, the sensor that was previously working won't pair again after the uninstall and re-install and neither will a Hue bulb... Both without giving me a single line in the log at debug level... 

Tried uninstalling en re-installing several times, but nothing changes: nothing happens, not any logging with clues as to what might go wrong... I'm stuck.

I quickly installed ZHA to see if that worked and was able pair the Tretakt and control it. So the hardware functions.

I have a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P with the latest firmware.

### What did you expect to happen?

I expected that I Z2M would find the new device after enabling paring on both sides.

### How to reproduce it (minimal and precise)

Enable "Permit join (all)" and push the pairing button on Tretakt: nothing happens

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Home assistant add-on on RasPi

### Debug log

Info 2025-04-02 14:18:13z2m: Zigbee: allowing new devices to join.
Info 2025-04-02 14:18:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"80knx-1"}'

brtvts, 2025-04-02:
I have the same problem here ...

tvdhorst, 2025-04-02:
Here’s the log of Z2M starting, maybe that’s helpful?

`[16:39:30] INFO: Preparing to start...
[16:39:31] INFO: Socat not enabled
[16:39:31] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-04-02 16:39:34] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-02 16:39:34] info: 	z2m: Starting Zigbee2MQTT version 2.2.0 (commit #unknown)
[2025-04-02 16:39:34] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-02 16:39:34] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB2","manufacturer":"ITead","serialNumber":"cc0d5d3d8945ed11b582ca8f0a86e0b4","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_cc0d5d3d8945ed11b582ca8f0a86e0b4-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-04-02 16:39:34] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB2","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-04-02 16:39:34] info: 	zh:zstack:znp: Serialport opened
[2025-04-02 16:39:35] info: 	z2m: zigbee-herdsman started (resumed)
[2025-04-02 16:39:35] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20240710,"transportrev":2},"type":"zStack3x0"}'
[2025-04-02 16:39:35] info: 	z2m: Currently 0 devices are joined.
[2025-04-02 16:39:35] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-04-02 16:39:35] info: 	z2m: Connected to MQTT server
[2025-04-02 16:39:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-04-02 16:39:35] info: 	z2m: Started frontend on port 8099
[2025-04-02 16:39:35] info: 	z2m: Zigbee2MQTT started!
[2025-04-02 16:39:40] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b002a2f4d03/connection_state/config', payload '{"device":{"hw_version":"zStack3x0 20240710","identifiers":["zigbee2mqtt_bridge_0x00124b002a2f4d03"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.2.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.2.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b002a2f4d03_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'`

sjchickey, 2025-04-02:
Same issue for me. Using the Home Assistant add-on v 2.2.0-1.

I've had a couple of devices drop off the network and cannot get any of them to pair any not, not even an attempt to pair and a failure, just nothing and no evidence in the logs anything is trying.

Pr20100, 2025-04-03:
Same here, can't pair any new device

oliversweb, 2025-04-03:
I too are unable to enrolling a new tuya outlet ... unsure if Z2M v2.2.0 or my device

tvdhorst, 2025-04-08:
@Koenkk any suggestions on how to troubleshoot this or what might go wrong? 

I just checked the log again, z2m correctly finds the adapter (its details also visible in the gui under settings). But, after successfully enabling 'permit join' on the coordinator (which has no paired devices at all) not a single device triggers anything on the z2m side, no logs at all... These devices worked before, only the IKEA tretakt is new. I just do not understand why even the logs remain empty.

Again: same hardware works flawlessly if I test with ZHA.

Koenkk, 2025-04-08:
Try doing a coordinator check: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check

sjchickey, 2025-04-08:
> Try doing a coordinator check: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check

I have run this on my system. It shows missing routers but they are all my Hue bulbs and they are all working fine in the system. I think it shows them all, plus the one Osram bulb I have. None of my other routers, which are mainly Sonoff devices show in the missing list. This only seemed to be an issue after the 2.2.0 update and I noticed a couple of my end devices dropped off and have not been able to get anything to pair since, even if I specify pairing to a router that is not on the missing list.

tvdhorst, 2025-04-08:
> Try doing a coordinator check: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check

Thanks for your suggestions. I had already done this, I should've mentioned it, but just did again:

`[2025-04-08 21:58:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/coordinator_check', payload '{"data":{"missing_routers":[]},"status":"ok"}'`

Seems fine to me, correct?

Jaegerstefan, 2025-04-09:
I dont know what really was the issue here but i had the same problem until completly starting from scratch with Z2M. 
I am using the SLZB06, upgraded the firmware and Z2M to the newest version and the problems began. Tested several SLZB06 firmware versions and also Zigbee Coordinator version but was unable to get it back working. Only fully starting from scratch with Z2M including new/re- pairing all my devices solved it.
i had several devices as "missing_routers" but was also unable to get them back online by re pairing. With the new start now everything works smoothly. ....

tvdhorst, 2025-04-10:
Nice that you've got it to work again @Jaegerstefan, starting from scratch didn't solve anything for me unfortunately. I even manually searched for any Z2M related config and deleted it, before re-installing.

@Koenkk any other suggestions? I'm still on that fresh install that basically does nothing, without logging, something makes me think the error lies more on the hardware or network configuration side. But I'd think that a new install would fix such a problem.

Koenkk, 2025-04-10:
@tvdhorst to dive deeper here you would need to make a sniff when trying to pair the device: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

sdetweil, 2025-04-11:
I was away for a month, lost power sometime, UPS failed.
came back, worked around dead ups, powered up
lost docker compose file somehow..

anyhow,  got it back 
start fresh 
zigbee2mqtt starts, but coordiantor doesn't , no devices found
```text
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-04-11 16:36:20] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-11 16:36:22] info: 	z2m: Starting Zigbee2MQTT version 2.2.1 (commit #a14feef8c13f5ce49b0ac358d98d3234d21d7ff4
)
[2025-04-11 16:36:22] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-11 16:36:22] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20231007145920","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231007145920-if00","vendorId":"1a86","productId":"55d4"} => ember: path=/dev/ttyACM0, score=4
[2025-04-11 16:36:22] info: 	zh:ember: Using default stack config.
[2025-04-11 16:36:22] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-04-11 16:36:22] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-04-11 16:36:22] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-04-11 16:36:22] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-04-11 16:36:22] info: 	zh:ember:uart:ash: Serial port opened
[2025-04-11 16:36:22] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-04-11 16:36:23] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-04-11 16:36:23] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-04-11 16:36:23] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-04-11 16:36:23] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-04-11 16:36:23] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-04-11 16:36:24] info: 	zh:ember: [STACK STATUS] Network up.
[2025-04-11 16:36:24] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-04-11 16:36:24] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2025-04-11 16:36:24] info: 	z2m: zigbee-herdsman started (resumed)
[2025-04-11 16:36:24] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-04-11 16:36:24] info: 	z2m: Currently 0 devices are joined.
[2025-04-11 16:36:24] info: 	z2m: Connecting to MQTT server at mqtt://mosquitto
[2025-04-11 16:36:24] info: 	z2m: Connected to MQTT server
[2025-04-11 16:36:24] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-04-11 16:36:24] info: 	z2m: Started frontend on port 8080
[2025-04-11 16:36:24] info: 	z2m: Zigbee2MQTT started!
```
and thats it

nothing shows in the front end. 
dongle firmware has the latest , shows correctly
can't control any lights, as they are all zigbee thru HA.. ooops..

docker compose right out of doc here.. nothing special

Koenkk, 2025-04-12:
@sdetweil are the devices still in the `data/database.db` file? 

> lost docker compose file somehow..

Indicates there is some file loss, so maybe the `database.db` was removed as well

sdetweil, 2025-04-12:
i don't know what that means in practicality

database.db contains
```json
{"id":1,"type":"Coordinator","ieeeAddr":"0xe0798dfffebbe238","nwkAddr":0,"manufId":4169,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":101,"inClusterList":[0,3,6,8,10,25,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,1794,2817,2819,2820,4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{}}
```
updated this morning on restart

sdetweil, 2025-04-12:
i tried to reconnect a device, it goes to blinking , but timesout, never reported for interview

sdetweil, 2025-04-13:
i reflahed the adapter, in case HA did something.. 
i don't see any difference

I have MQtt Explorer,  but nothing exciting there different the the saved info .
I have 2 custom esp32/mqtt devices and they are publishing to MQTT correctly appearing in HA. 

this is the only data in mqtt for zigbee2mqtt
```
zigbee2mqtt
bridge
state = {"state":"online"}
converters = []
info = {"commit":"a14feef8c13f5ce49b0ac358d98d3234d21d7ff4\n","config":{"advanced":{"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[109,159,3,104,29,130,253,145],"last_seen":"disable","log_console_json":false,"log_debug_namespace_ignore":"","log_debug_to_mqtt_frontend":false,"log_directories_to_keep":10,"log_directory":"/app/d…
devices = [{"disabled":false,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genOnOff","genLevelCtrl","genTime","genOta","lightingColorCtrl"],"output":["genBasic","genIdentify","genGroups","genScenes","genOnOff","genLevelCtrl","genPollCtrl","lightingColorCtrl","msIlluminanceMeasurement","msTemperatureMeasurement","msRelativeHumidity","msOccupancySensing","ssIasZone","seMeterin…
groups = []
definitions = {"clusters":{"barrierControl":{"ID":259,"attributes":{"barrierPosition":{"ID":10,"type":32},"capabilities":{"ID":3,"type":24},"closeEvents":{"ID":5,"type":33},"closePeriod":{"ID":9,"type":33},"commandCloseEvents":{"ID":7,"type":33},"commandOpenEvents":{"ID":6,"type":33},"movingState":{"ID":1,"type":48},"openEvents":{"ID":4,"type":33},"openPeriod":{"ID":8,"type":33},"safetyStatus":{"ID":2,"type":25…
extensions = []
```

sdetweil, 2025-04-13:
i got out a new device, it was not reported at zigbee2mqtt

sdetweil, 2025-04-14:
i have ordered another adapter that is supposed to have the z-stack coordinator flash..
I cannot flash this one except to 7.4.4 whatever that is..  none of the tools work,  I don't have a windows native system , only a vm.

sdetweil, 2025-04-14:
interesting that the state says (state.json)
```txt
        "log_level": "debug",
        "network": {
            "channel": 11,
            "extended_pan_id": "0xf88d220b7d512c6d",
            "pan_id": 45487
        },
        "permit_join": false,          <-------- but the UI says yes
        "restart_required": false,
        "version": "2.2.1",
        "zigbee_herdsman": {
            "version": "3.4.11"
        },
        "zigbee_herdsman_converters": {
            "version": "23.20.1"
        }
```

sdetweil, 2025-04-14:
also see this in the log
```text
[2025-04-14 16:11:40] debug: 	zh:ember:ezsp: ===> [FRAME: ID=34:"PERMIT_JOINING" Seq=37 Len=6]
[2025-04-14 16:11:40] debug: 	zh:ember:ezsp: <=== [FRAME: ID=34:"PERMIT_JOINING" Seq=37 Len=6]
[2025-04-14 16:11:40] debug: 	zh:ember: ~~~> [ZDO PERMIT_JOINING_REQUEST BROADCAST to=65532 messageTag=1 payload=010001]
```

sdetweil, 2025-04-15:
so, i have found a good data folder

but the device id is different than currently being reported. 

i also think i have an mqtt server  db to match.

can i just replicate that data under the two services and start them back up again?

sdetweil, 2025-04-15:
SO, I stopeed z2m and copied the data folder back and started z2m again,
and all my devices are back... and back in HA.
NOW, I haven't tried adding a new yet..  I do see one error in the z2m logs
```
info 2025-04-15 07:11:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/master bathroom sensor', payload '{"battery":50.5,"comfort_humidity_max":0.6,"comfort_humidity_min":0.4,"comfort_temperature_max":0.25,"comfort_temperature_min":0.2,"enable_display":"OFF","humidity":49.92,"humidity_calibration":0,"linkquality":156,"show_smile":"SHOW","show_smiley":"HIDE","temperature":23.07,"temperature_calibration":0,"temperature_display_mode":"fahrenheit","update":{"installed_version":19083265,"latest_version":19083265,"state":"idle"},"voltage":3000}'

error 2025-04-15 07:11:00z2m: Failed to configure 'master bathroom sensor', attempt 2 (Error: Bind 0xa4c1380a35acccbd/1 msTemperatureMeasurement from '0xe0798dfffebbe238/1' failed (Status 'TABLE_FULL') at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:574:23) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:40:9) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/devices/custom_devices_diy.ts:871:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/index.ts:363:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```
TABLE_FULL

sdetweil, 2025-04-15:
seeing a few 
```
info 2025-04-15 08:10:19zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "44118".
info 2025-04-15 08:10:20zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "60503".
info 2025-04-15 08:10:20zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "44325".
```

sdetweil, 2025-04-15:
now the frontend won't start 
```
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-04-15 16:54:14] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-15 16:54:14] info: 	z2m: Starting Zigbee2MQTT version 2.2.1 (commit #a14feef8c13f5ce49b0ac358d98d3234d21d7ff4
)
[2025-04-15 16:54:14] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-15 16:54:14] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20231007145920","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231007145920-if00","vendorId":"1a86","productId":"55d4"} => ember: path=/dev/ttyACM0, score=4
[2025-04-15 16:54:14] info: 	zh:ember: Using default stack config.
[2025-04-15 16:54:14] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-04-15 16:54:14] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-04-15 16:54:14] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-04-15 16:54:14] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-04-15 16:54:14] info: 	zh:ember:uart:ash: Serial port opened
[2025-04-15 16:54:14] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-04-15 16:54:15] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-04-15 16:54:15] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-04-15 16:54:15] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-04-15 16:54:15] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-04-15 16:54:15] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-04-15 16:54:15] info: 	zh:ember: [STACK STATUS] Network up.
[2025-04-15 16:54:15] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-04-15 16:54:15] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2025-04-15 16:54:15] info: 	z2m: zigbee-herdsman started (resumed)
[2025-04-15 16:54:15] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-04-15 16:54:15] info: 	z2m: Ikea On/off (0x6c5cb1fffe839d57): E2201 - IKEA RODRET wireless dimmer/power switch (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: bedroom light 1 (0x040d84fffeca0f2e): LED2003G10 - IKEA TRADFRI bulb E26/27, white spectrum, globe, opal, 1055/1100/1160 lm (Router)
[2025-04-15 16:54:15] info: 	z2m: ikea light 2 (0x040d84fffec546be): LED2003G10 - IKEA TRADFRI bulb E26/27, white spectrum, globe, opal, 1055/1100/1160 lm (Router)
[2025-04-15 16:54:15] info: 	z2m: master bathroom sensor (0xa4c1380a35acccbd): LYWSD03MMC - Custom devices (DiY) Xiaomi temperature & humidity sensor with custom firmware (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: living room sensor (0xa4c138d08fb1e2e7): LYWSD03MMC - Custom devices (DiY) Xiaomi temperature & humidity sensor with custom firmware (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: bar light 1 (0x001788010b6d5445): 9290022166 - Philips Hue white and color ambiance E26/E27 (Router)
[2025-04-15 16:54:15] info: 	z2m: Motion sensor (0x001788010211a51d): 9290012607 - Philips Hue motion sensor (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: bar light 2 (0x0017880102736a71): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
[2025-04-15 16:54:15] info: 	z2m: bar light 3 (0x0017880102796cfd): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
[2025-04-15 16:54:15] info: 	z2m: fireplace (0x001788010321a5d5): 9290002579A - Philips Hue white and color ambiance BR30 (Router)
[2025-04-15 16:54:15] info: 	z2m: counter (0x00124b002fa1c4d8): S31ZB - SONOFF Zigbee smart plug (US version) (Router)
[2025-04-15 16:54:15] info: 	z2m: over cabinets (0x00124b002fa1c289): S31ZB - SONOFF Zigbee smart plug (US version) (Router)
[2025-04-15 16:54:15] info: 	z2m: overhead (0x38398ffffed73f88): TS0011 - Tuya Smart light switch - 1 gang (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: front porch (0x38398ffffed719de): TS0011 - Tuya Smart light switch - 1 gang (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: ikea motion sensor (0xf082c0fffe2e2c37): E1525/E1745 - IKEA TRADFRI motion sensor (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: water heater leak sensor (0xa4c13855d90a97f7): ZG-222ZA - HOBEIAN Water leak sensor (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: laundry water sensor (0xa4c138d8965904ef): ZG-222ZA - HOBEIAN Water leak sensor (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: back bedroom lamp (0x70b3d52b600b8448): TS011F_plug_3 - Tuya Smart plug (with power monitoring by polling) (Router)
[2025-04-15 16:54:15] info: 	z2m: garage outlet (0x70b3d52b600a23c4): TS011F_plug_3 - Tuya Smart plug (with power monitoring by polling) (Router)
[2025-04-15 16:54:15] info: 	z2m: table lamp (0x70b3d52b600bb944): TS011F_plug_3 - Tuya Smart plug (with power monitoring by polling) (Router)
[2025-04-15 16:54:15] info: 	z2m: floor lamp (0x70b3d52b600a8fcf): TS011F_plug_3 - Tuya Smart plug (with power monitoring by polling) (Router)
[2025-04-15 16:54:15] info: 	z2m: water valve (0x0ceff6fffe17d059): SWV - SONOFF Zigbee smart water valve (EndDevice)
[2025-04-15 16:54:15] info: 	z2m: Currently 22 devices are joined.
[2025-04-15 16:54:15] info: 	z2m: Connecting to MQTT server at mqtt://mosquitto
```
config
homeassistant:
  enabled: false
[... cut ...]
serial:
  port: /dev/ttyACM0
advanced:
  network_key:
    - 14
    - 47
    - 131
    - 167
    - 25
    - 105
    - 84
    - 167
    - 165
    - 253
    - 89
    - 1
    - 204
    - 247
    - 64
    - 38
  pan_id: 35573
  ext_pan_id:
```

worded fine, but had same data loss with nodered container.. stopped, copied data, restarted, got nothing
rebooted, still nothing

devices are in HA, and response

need to add some, but get no response from z2m. 
said enabled for joining:all

now can't get UI up

tvdhorst, 2025-04-16:
> [@tvdhorst](https://github.com/tvdhorst) to dive deeper here you would need to make a sniff when trying to pair the device: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

@Koenkk, ok thanks, I'll do this, but it will be after my holiday, so not before the beginning of may

closed 2025-06-25 by github-actions[bot]

FriedlNetDZ, 2025-07-19:
I have the same problem and use a Sonoff ZBDongle-P. The problem occurred from one day to the next. All devices are running normally, but I can no longer integrate a new device.
Has anyone found a solution to this problem?

sdetweil, 2025-07-19:
@FriedlNetDZ  did you go into the HA mqtt integration, find the hub and enable permit join?

FriedlNetDZ, 2025-07-19:
@sdetweil: Yes, I had already read your post above and tried it, but unfortunately it still doesn't work.

FriedlNetDZ, 2025-07-19:
But what surprises me is that I have two bridge entries?

[image]

sdetweil, 2025-07-19:
i dont have an entry for the mosquitto broker in my ha integration, i do see the server

[image]

sdetweil, 2025-07-19:
note that the permit join setting has a (hidden) timer and will turn off again. only a few minutes

i use the zigbeemqtt web page to watch the interview progress and status

bistvan12, 2025-07-20:
I have a same issue with LAN SLZB-06: newest firmware. I went back to a SONOFF ZigBee 3.0 USB Dongle. Same problem. Of course "permit_join: true".

sdetweil, 2025-07-21:
@bistvan12  where are you setting that 

it doesnt work in the zigbeemqtt server, or the xigbeemqtt ui. only in the ha mqtt integration for the hub

bistvan12, 2025-07-27:
> After I changed the coordinator (ZBDongle-P -> ZBDongle-E) without having to reinstall the zigbee-network (thanks to the excellent instructions from Simon42), my entire Zigbee network is running perfectly again and I can also learn new devices again.

I was not able to find the "excellent instructions from Simon42". I changed the adapter, restart Z2M and I got an error message: 
"Error: Adapter EZSP protocol version (8) is not supported by Host [13-16]"
I tried to downgrade the adapter firmware to version (7). I used "pip install silabs-flasher" and /dev/ttyUSB0, but the coda have not run on Loxberry.
Could you give somer specific advise, how to make Z2M 3.1 working again?

zyndata, 2025-08-01:
In my case, changing the Zigbee network channel solved the issue. My previous network was using channel 20, and even after reinstalling Zigbee2MQTT, nothing would connect. However, once I changed the channel (to 25 in my case), everything started working.