Koenkk/zigbee2mqtt #26950: MQTT failed to connect after updating to 2.2.0-1

ash286, 2025-04-02:
### What happened?

I'm using HAOS on RPi4b. I updated Z2M from previous version `2.1.3-1` to the latest verion `2.2.0-1` and it did revoke MQTT authorization. 

Log:
`z2m: MQTT failed to connect, exiting... (Connection refused: Not authorized)`

Restoring back to previous version `2.1.3-1` didn't work. RPi power cycle/full-reboot doesn't work so I needed to change MQTT user password, full-reboot my HA and update MQTT password upon MQTT error prompt.

I tried EZSP and ember both the driver and issue remains same.

### Zigbee2MQTT version

2.2.0-1

### Adapter firmware version

7.4.4.0

### Adapter

EZSP and ember both reproduces the same issue

### Setup

Home Assistant OS on RPi-4b

### Debug log

`z2m: MQTT failed to connect, exiting... (Connection refused: Not authorized)`

ash286, 2025-04-02:
I just found that it reqrites secret.yaml file as following

`zigbee-user: '!/config/secrets.yaml zigbee-user'`
`zigbee-password: '!/config/secrets.yaml zigbee-password'`

while that was something like below

`zigbee-user: mqtt-username`
`zigbee-password: mqtt-password`

Koenkk, 2025-04-02:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/26943

closed 2025-04-02 by Koenkk