Koenkk/zigbee2mqtt #26952: Exception while calling fromZigbee converter: kwhPrecision is not defined

technoman63, 2025-04-02:
### What happened?

Hi, 

I do the update of my z2m this morning and no i have a lot of  "Exception while calling fromZigbee converter: kwhPrecision is not defined" in my log 

### Zigbee2MQTT version

 2.2.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

Virtual machine on Qnap 

### Debug log

[log.log](https://github.com/user-attachments/files/19571109/log.log)

loic-perie, 2025-04-02:
I have the same problem with my Lixee ZLinky since the update in 2.2.0:
z2m: Exception while calling fromZigbee converter: kwhPrecision is not defined}

kwhPrecision is set to 2. I've changed the parameter, saved and rebooted the service, but the problem is still the same.

Koenkk, 2025-04-02:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/issues/9109

closed 2025-04-02 by Koenkk