Koenkk/zigbee2mqtt #26953: AVATTO ZDMS16-1 _TZE204_nqqylykc countdown not working correctly

mrmaximas, 2025-04-02:
### What happened?

I'm sending a MQTT command 
`zigbee2mqtt/Bathroom Mirror/set` with the payload `{ ‘countdown’: 5 }`
The device actually responds to the command (light switching on / off after 5s), but nothing changes in MQTT topic.

### What did you expect to happen?

When the device executes a command this should show up in the MQTT.

### How to reproduce it (minimal and precise)

Just send a MQTT command 
`zigbee2mqtt/Bathroom Mirror/set` with the payload `{ ‘countdown’: 5 }`

### Zigbee2MQTT version

2.2.0-dev commit: 5257c7ea

### Adapter firmware version

20240710

### Adapter

CC2652P7

### Setup

Mac OS latest on M2

### Debug log

[log.log.zip](https://github.com/user-attachments/files/19571181/log.log.zip)

mrmaximas, 2025-04-05:
@Koenkk please pay attention 🙏🏻

Koenkk, 2025-04-06:
It seems the device doesn't report it status, can you check if it works with the Tuya hub?

mrmaximas, 2025-04-06:
Tomorrow i will try to temporarily move this device to tuya hub and sniff all traffic including interview.

mrmaximas, 2025-04-07:
Here is 5, 10 and 15 second timers running sequentially from the tuya cloud side. 
The device is switched on and off, this is displayed in the app.

[awatto_dimmer_mirror_light_5-10-15s.pcapng.zip](https://github.com/user-attachments/files/19630275/awatto_dimmer_mirror_light_5-10-15s.pcapng.zip)

Here is interview and and a lot of timer starts from the cloud side.

[awatto_dimmer_mirror_light_interview_and_more.pcapng.zip](https://github.com/user-attachments/files/19630261/awatto_dimmer_mirror_light_interview_and_more.pcapng.zip)

Koenkk, 2025-04-08:
- When paired with z2m, does the state change when controlling the switch physically?
- When paired with the Tuya gateway, can you check if the switch also goes to the OFF state in the app when setting the countdown to 20 seconds, wait 5 seconds, remove the switch from power (such that it cannot communicate anymore)
- Can you sniff closer to the dimmer? There are some corrupted packages (e.g. 77 `Bad FCS`)

mrmaximas, 2025-04-09:
@Koenkk 
1. When paired with z2m, does the state change when controlling the switch physically? < YES
3. Today, I've put the sniffer, dimmer and hub right next to each other.
2. I first sniffed the inteview and started giving commands from my phone and from the cloud. It looks like the device is problematic and i really doesn't report that the dimmer has triggered and the state has changed. When I gave commands from the cloud the state on my phone screen did not change, it changes only if the countdown is started from the phone, from the phone it is possible only with a multiplicity of 1 minute. 
for examle the last packets
2555 10s set from the cloud
2559 10s accepted from the dimmer ??
2590 10s set from the cloud
2594 10s acceprted from the dimmer ??

Filter is `zbee_nwk.addr eq 0xf5ea and zbee_nwk.addr eq 0x0000 and _ws.col.protocol == "ZigBee HA"`
[2025-04-09-awatto-dimmer.pcapng.zip](https://github.com/user-attachments/files/19669089/2025-04-09-awatto-dimmer.pcapng.zip)

mrmaximas, 2025-04-10:
I requested a dimmer update today and they sent me a zigbee module update. The problem doesn't seem to be solved.
If I send command from the cloud, the device reports that it has accepted the command, but does not report its execution. in the tuya  log it looks like this.

[image]

mrmaximas, 2025-04-10:
@Koenkk tuya support says "you need to configure the reporting correctly in z2m" (c)