Koenkk/zigbee2mqtt #26956: Z2m restarts for no reason.

TomKassel, 2025-04-02:
### What happened?

Z2m restarts for no reason.

'Zigbee2MQTT server has exited unexpectedly'
'After 10s Z2m restarts!'

### What did you expect to happen?

No restart.

### How to reproduce it (minimal and precise)

You can't reproduce it. It happens every 4-24 hours.

### Zigbee2MQTT version

2.1.3

### Adapter firmware version

20250321

### Adapter

Zstack, SLZB-06P7

### Setup

PC, Windows 11

### Debug log

[log.log](https://github.com/user-attachments/files/19572418/log.log)

Nerivec, 2025-04-02:
I don't see any startup or shutdown sequence in that log file. Do you have the one with it?

Nerivec, 2025-04-02:
So, the first log should have the error that caused the restart, but it doesn't appear to. That would suggest a pretty hard crash of the nodejs process.
Can you check system logs, see if you can spot something around that time?
If you aren't already on it, I suggest you try updating to latest LTS version of nodejs too (v22): https://nodejs.org/en/download

TomKassel, 2025-04-03:
In my desperation, I updated the nodejs version a few days ago. You suspect a Windows or hardware problem?
I will check the system log files.
By the way, the last restart was a few minutes ago.

Nerivec, 2025-04-03:
The last few lines of logging from Z2M before restart are likely only printed to the console, so you may have to catch it there so we can get more details (unless some system log can point to a failure of some kind).

TomKassel, 2025-04-03:
In the Windows protocol there are only processor warnings.
I have just ported the system to new hardware. I'm curious.

I'll take a look at the console.
Thank you very.

TomKassel, 2025-04-04:
I installed my system on a new PC. It ran for 1 day and 7 hours, then there was a restart from Z2M and over 40 devices are offline for the time being.
The attached file is a copy directly from the console. I can't see any error.
How is this possible?

[Error Log.txt](https://github.com/user-attachments/files/19607655/Error.Log.txt)

Nerivec, 2025-04-04:
I don't see any restart at 17:47 in that log.

Your mentioned before `17:37:55` and now `17:47`, the two are close enough to make me wonder, can you put this crash around the same time of day every time? Or maybe it's about the runtime duration, 1d8h, 1d7h?

If a different machine is exhibiting the same behavior, you have a few avenues to explore:
- the two machines have similar-enough configurations that both are affected by the same bug
- problem with the adapter that's causing a crash
- problem with a device on the network that's causing a crash (some kind of bad behavior, spamming, corrupted data...)

If using USB mode:
- check the power saving settings of the COM port (in Device Manager), make sure the OS cannot turn it off to "save power"
- update the drivers: https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads

TomKassel, 2025-04-04:
I use the SLZB-06P7 in PoE network mode. Previously I had a Conbee2, which did not cause these problems.
The new hardware is quite different from the old one and has been completely reinstalled.

TomKassel, 2025-04-06:
I had a restart of Z2M today at 17:45.
This time, both before the restart (Log1) and after the restart (Log2) there are various error messages in the log, which I unfortunately can't do anything with.

[log1.log](https://github.com/user-attachments/files/19622457/log1.log)
[log2.log](https://github.com/user-attachments/files/19622456/log2.log)

Nerivec, 2025-04-06:
These errors should not crash Z2M. (_`0x3410f4fffe61e419` is causing most of them, it's failing poll check-in._)
We're still missing the one that does unfortunately.

What's the "core firmware" version on your adapter?

TomKassel, 2025-04-06:
First of all, thank you again for the support. 
I've had daily reboots for some time now, which is exhausting and annoying.
Core v2.8.2.dev3 is currently installed and Zigbee is 20240716.
But I've tried older versions, for example Core V 2.7.1.

Koenkk, 2025-04-06:
Could you try connecting the SLZB in USB mode instead of network? This will rule out any network related problems.

TomKassel, 2025-04-06:
I can implement this tomorrow with little effort.

Nerivec, 2025-04-06:
Make sure to follow the USB-related points [here](https://github.com/Koenkk/zigbee2mqtt/issues/26956#issuecomment-2779390140) beforehand.

TomKassel, 2025-04-09:
I have now connected the SLZB-06P7 via USB. I turned off the Ethernet connection. Installed the recommended Silabs driver and turned off all power-saving features on USB.
After 8 hours, at 1:59 a.m., Z2M made a restart again.

[log.log](https://github.com/user-attachments/files/19662176/log.log)

Nerivec, 2025-04-09:
Can you check in <kbd>right click</kbd> on `This PC` > `Manage` > `System Tools` > `Event Viewer` > `Windows Logs`:
- `System` logs
- `Application` logs

for each: `Filter Current Logs` > 1:54 AM to 2:04AM (5min before/after) > `Save Filtered Log File As`.

TomKassel, 2025-04-09:
In the time from 1:54 to 2:04 there are no (!) Entries in the logs. Neither in the application log nor in the system log.

Nerivec, 2025-04-09:
Can you use the console to start Z2M and disable the watchdog so that Z2M doesn't restart automatically on next crash? Maybe we can finally get the actual error in the console when it stops...

Nerivec, 2025-04-09:
I don't know Homeseer at all. Can't help there.
But you might want to make sure that Homeseer isn't the source of the restart... check the Homeseer logs if there are any...

TomKassel, 2025-04-09:
The Homeseer Log insists on the Homeseer messages and the original Z2M messages.
Of course, there is no error, except for the message:

'Zigbee2MQTT server has exited unexpectedly'
'After 10s Z2m restarts!'

Nerivec, 2025-04-09:
Any chance you have another "plugin" like Z2M that could mess with the adapter's connection?

TomKassel, 2025-04-09:
Yes, I use z-wave with an Aeotec Stick Gen5. But I do that as long as I use Z2M, there have never been any problems and have worked well so far. I haven't changed anything in the z-wave configuration either.

Nerivec, 2025-04-09:
https://forums.homeseer.com/forum/hs4-products/hs4-plugins/lighting-primary-technology-plug-ins-aa/zigbee-plus-homeseer/1656148-moved-hs4-to-new-computer-now-issues-zigbee2mqtt-server-has-exited-unexpectedly

Might want to do a more thorough search of their forums, see if you can find something that's already been reported like this one.

TomKassel, 2025-04-10:
Thanks for the constructive comment. I have read hundreds of comments in blogs and forums in the last few weeks.
The hint from the Homeseer forum is 14 months old and has not been confirmed by anyone.
Nevertheless, I swapped the starting order of Z-Wave and Zigbee. I even tried it without Z-wave.

closed 2025-06-10 by Nerivec