Koenkk/zigbee2mqtt #26962: TS0044_1 device action not added to mqtt

GSV3MiaC, 2025-04-02:
### What happened?

I just added a TS0044_1 device to zigbee2mqtt and the 'action' entity did not get picked up by mqtt, only batter is there (and rssi etc as disabled options). If I manually configure an mqtt entity as "zigbee2mqtt2/Zigbee 4 Buttons/action" if gets set correctly, but that entity is not there by default. when the device gets added (zigbee2mqtt startup) there is no action state, so maybe that is the issue?

### What did you expect to happen?

I expected an 'action' entity to be available, bnot just the battery state!

### How to reproduce it (minimal and precise)

Just start zigbee2mqtt (I am using edge version) and pair a TS044_1 4 button remote

### Zigbee2MQTT version

 2.1.3-dev commit: 6e6a9891

### Adapter firmware version

 8.0.2 [GA]

### Adapter

ember

### Setup

HAOS intel

syntesys87, 2025-04-03:
Hi,

try to enable in Z2M Settings -> Home Assistant Integration -> Home Assistant legacy action sensors; now it will be available.

GSV3MiaC, 2025-04-03:
My hero!! Yes that fixes it, but the chances of me figuring it out without your help would have been zero. I guess it us in the docs somewhere, somewhen?!

closed 2025-04-03 by GSV3MiaC

syntesys87, 2025-04-03:
Very well! It was in the documentation of migrating from 1.x to 2.x, it was one of things done when I migrated.