Koenkk/zigbee2mqtt #26965: Latest update error Starting Zigbee2MQTT without watchdog

jojkopeter, 2025-04-03:
### What happened?

Hello, after updating to 2.2.0-1, Zigbee2MQTT is starting without a watchdog.
Please help.

### Zigbee2MQTT version

2.2.0-1

### Adapter firmware version

20230507

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

 Pi

Nerivec, 2025-04-03:
This is not an error, it just means the Z2M watchdog is not active (this is not new to 2.2.0, it has been like this for several versions).
**The HA add-on watchdog is different from the [Z2M watchdog](https://www.zigbee2mqtt.io/guide/installation/15_watchdog.html).**
You can configure the Z2M watchdog by showing unused options on add-on configuration page, and checking the watchdog box, then follow the previous link to configure it.

jojkopeter, 2025-04-04:
Hello, I've already done that, and Zigbee2MQTT still starts a watchdog

Nerivec, 2025-04-04:
Can you post a screenshot of your add-on configuration page with "show unused options" checked?