Koenkk/zigbee2mqtt #26967: After upgrade to 2.2.0 docker container crashes when starting on Raspberry PI 3

alting, 2025-04-03:
### What happened?

After upgrade to 2.2.0 docker container crashes when starting on Raspberry PI 3:

/usr/bin/docker run --rm --name=zigbee2mqtt --net=host -p 8126:8080 -v /home/pi/zigbee2mqtt/data:/app/data  -v /run/udev:/run/udev:ro --device=/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E27422-if00:/dev/ttyACM0  --user 1000:1000  --group-add dialout -e TZ=Europe/Moscow ghcr.io/koenkk/zigbee2mqtt
WARNING: Published ports are discarded when using host network mode
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0x747d4744

### How to reproduce it (minimal and precise)

Just start latest container on RPI3

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20190425

### Adapter

usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E27422-if00

### Setup

Docker container. Raspberry PI3

Nerivec, 2025-04-03:
Can you post the output of the following commands?
`uname -a`
`uname -m`
`cat /etc/issue.net`
`docker --version`

alting, 2025-04-04:
uname -a
Linux raspberrypiha 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

 uname -m
armv7l

 cat /etc/issue.net
Raspbian GNU/Linux 10

 docker --version
Docker version 18.09.1, build 4c52b90

Nerivec, 2025-04-04:
Can you upgrade your OS to latest version?

alting, 2025-04-05:
After upgrade to Raspbian GNU/Linux 11 all OK!

closed 2025-04-05 by alting