Koenkk/zigbee2mqtt #26969: update 2.1 -> renders all devices offline

carlhye, 2025-04-03:
### What happened?

After updating to 2.2.0.1 all devices af unresponsive and the Z2M front end displays 'Availability: Offline'

Example from the add-log when turning on a group og light bulbs:
```
[2025-04-03 11:08:38] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 1' failed: 'Error: ZCL command 0x30fb10fffee0ae5d/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '23722'.)'
[2025-04-03 11:08:38] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 3' failed: 'Error: ZCL command 0x30fb10fffee0b9ec/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '59643'.)'
[2025-04-03 11:08:38] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 2' failed: 'Error: ZCL command 0x30fb10fffee06824/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '41881'.)'
[2025-04-03 11:08:43] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 1' failed: 'Error: ZCL command 0x30fb10fffee0ae5d/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '23722'.)'
[2025-04-03 11:08:43] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 3' failed: 'Error: ZCL command 0x30fb10fffee0b9ec/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '59643'.)'
[2025-04-03 11:08:43] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 2' failed: 'Error: ZCL command 0x30fb10fffee06824/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '41881'.)'
[2025-04-03 11:09:05] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 1' failed: 'Error: ZCL command 0x30fb10fffee0ae5d/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '23722'.)'
[2025-04-03 11:09:05] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 3' failed: 'Error: ZCL command 0x30fb10fffee0b9ec/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '59643'.)'
[2025-04-03 11:09:05] error: 	z2m: Publish 'set' 'state' to 'Saras Room Color Spot 2' failed: 'Error: ZCL command 0x30fb10fffee06824/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '41881'.)'
[2025-04-03 11:09:09] error: 	z2m: Publish 'set' 'brightness' to 'Saras Room Color Spot 1' failed: 'Error: ZCL command 0x30fb10fffee0ae5d/1 genLevelCtrl.moveToLevelWithOnOff({"level":122,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '23722'.)'
[2025-04-03 11:09:09] error: 	z2m: Publish 'set' 'brightness' to 'Saras Room Color Spot 3' failed: 'Error: ZCL command 0x30fb10fffee0b9ec/1 genLevelCtrl.moveToLevelWithOnOff({"level":122,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '59643'.)'
[2025-04-03 11:09:09] error: 	z2m: Publish 'set' 'brightness' to 'Saras Room Color Spot 2' failed: 'Error: ZCL command 0x30fb10fffee06824/1 genLevelCtrl.moveToLevelWithOnOff({"level":122,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '41881'.)'
```

### What did you expect to happen?

I expected everything to still work, now nothing works.

### How to reproduce it (minimal and precise)

Install 2.2.0.1

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

SMLIGHT SLZB-06M with ember

### Setup

HA as a VM in Proxmox, PoE Eth coordinator

### Debug log

[45df7312_zigbee2mqtt_2025-04-03T09-20-58.677Z.log](https://github.com/user-attachments/files/19583739/45df7312_zigbee2mqtt_2025-04-03T09-20-58.677Z.log)

Kryzek, 2025-04-03:
https://github.com/Koenkk/zigbee2mqtt/issues/26942

Maybe this sticky helps for now?

carlhye, 2025-04-03:
Apart from the coordinator and the coordinator driver, the issue, seems to be the exact same as what I experience.

Satrier, 2025-04-03:
https://github.com/Koenkk/zigbee2mqtt/issues/26942#issuecomment-2771382448

Nerivec, 2025-04-03:
The frontend "offline" is just a display bug (shows offline when availability is disabled, when it just should not show the column). It has been fixed in dev branch.

It seems you are having actual issues since deliveries are failing. I assume you tried rebooting the whole machine already?

dvfeinblum, 2025-04-03:
Just wanna add my experience to the issue in case it's helpful. I'm seeing this but it also seems just like a frontend UI bug as my devices are sending data to HA, my automations still work, etc

unluckyquote, 2025-04-03:
I'm having the same issue.  Rebooting doesn't fix it for me

Jacq144, 2025-04-03:
Same issue here

Krusenbaum, 2025-04-03:
For me automations that use Zigbee devices don't work anymore + devices cannot be operated via HA dashboards. Zigbee2MQTT dashboard is the only thing that still works. Rollback to last version 2.1.3-1 solved everything.

closed 2025-06-10 by github-actions[bot]