Koenkk/zigbee2mqtt #26970: Z2MQTT add-on not starting after 2.2.0

novygh, 2025-04-03:
### What happened?

[12:22:04] INFO: Preparing to start...
[12:22:05] INFO: Socat not enabled
[12:22:05] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-04-03 12:23:13] error: 	z2m: Error while starting zigbee-herdsman
[2025-04-03 12:23:13] error: 	z2m: Failed to start zigbee-herdsman
[2025-04-03 12:23:13] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-04-03 12:23:13] error: 	z2m: Exiting...
[2025-04-03 12:23:13] error: 	z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:372:31)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5))
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:374:23)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5)

SLZB-06, did update to 20250321 and using 2.7.1

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20250321

### Adapter

SLZB-06

### Setup

HA Pi5

novygh, 2025-04-03:
homeassistant:
  enabled: true
mqtt:
  server: mqtt://core-mosquitto
  user: ***
  password: ***
  keepalive: 60
  reject_unauthorized: true
  version: 5
serial:
  port: tcp://192.168.10.12:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
frontend:
  enabled: true
  port: 8099
devices:
.
.
.
availability:
  enabled: true
  active:
    timeout: 10
  passive:
    timeout: 1500
advanced:
  transmit_power: 20
  channel: 20
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  log_level: error
  cache_state_persistent: true
  cache_state_send_on_startup: true
groups: {}
version: 4
blocklist: []

Koenkk, 2025-04-03:
How did you update the SLZB?

novygh, 2025-04-03:
could be... i used its local web address. eventualy gave up and re-pair everything... 10h straight and i am exhausted. if somehow i will have to do it again i try zha

Koenkk, 2025-04-03:
> i used its local web address.

You updated via the web page of the SLZB06? There are some more SLZB06 users with this problem here: https://github.com/Koenkk/Z-Stack-firmware/discussions/545

This might not be a z2m bug but a result of reflashing the coordinator on the SLZB, so be aware that when you switch to ZHA and reflash the coordinator, you might run into the same issue.

closed 2025-06-10 by github-actions[bot]