Koenkk/zigbee2mqtt #26972: After latest update to V2.2.0 on Nabu Casa Home Assistant Yellow cannot connect to the coordinator

wojciechczyz, 2025-04-03:
### What happened?

After latest update to V2.2.0 on Nabu Casa Home Assistant Yellow cannot connect to the coordinator.

Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-04-03 14:08:36] error: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=RESET_WATCHDOG.

Rolling back update and restarting HA bring it back to working. 
Configuration:

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: /dev/ttyAMA1
  adapter: ember
watchdog: default

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

yellow_zigbee_ncp_7.4.4.0.gbl

### Adapter

Home Assistant Yellow ember

### Setup

Add On

Nerivec, 2025-04-03:
The `RESET_WATCHDOG` happens with pretty much any adapter that has hardware flow control, this should not prevent from starting (it retries after).
Do you have the rest of the startup `debug` logs?

wojciechczyz, 2025-04-03:
There is nothing after this line. After I roll back to previous version it starts well.

wojciechczyz, 2025-04-03:
My firmware is yellow_zigbee_ncp_7.4.4.0.gbl, is it the suggested one to use? It is the newest of the Home Asistant firmware github

Nerivec, 2025-04-03:
7.4.x and 8.0.2 are fine. _I don't think nabucasa has updated to 8 yet, so, it would be 7.4 in their github still._

Can you post the full startup [`debug` logs](https://www.zigbee2mqtt.io/guide/configuration/logging.html#debugging) so we can take a look at what's going on?

wojciechczyz, 2025-04-03:
Thank you,

I upgraded again, and now stayed with all devices Offline. I enabled debug log, attaching to this action.

[45df7312_zigbee2mqtt_2025-04-03T16-35-34.190Z.log](https://github.com/user-attachments/files/19591406/45df7312_zigbee2mqtt_2025-04-03T16-35-34.190Z.log)
[image]

Nerivec, 2025-04-03:
This is a display bug in frontend. It has already been fixed. Devices should be working fine, it's only cosmetic.
https://github.com/Koenkk/zigbee2mqtt/issues/26942#issuecomment-2773510637

wojciechczyz, 2025-04-03:
Thank you! I am remote from this location so could not try!

Krusenbaum, 2025-04-03:
Cosmetic? For me, automations that use Zigbee devices don't work anymore (e.g. plants are not getting watered), devices don't react when operated via HA dashboards, etc.

Nerivec, 2025-04-03:
@Krusenbaum it's a different problem then. Would need `debug` logs to see what's going on exactly.

Nerivec, 2025-04-04:
@wojciechczyz if you can confirm that with 2.2.1, everything is back in order 👍

closed 2025-06-11 by github-actions[bot]