Koenkk/zigbee2mqtt #26977: Zigbee network unstable: routes marked as unavailable

ChessSpider, 2025-04-03:
### What happened?

Ever since upgrade to 2.x.x branch of zigbee2mqtt, my zigbee network appears less stable. It is noticeable whenever I do a version upgrade of zigbee2mqtt. It usually stabilizes after ~24hours. Howeve I upgraded to 2.2.0, and it stays unstable.

Logs shows many fails to ping error messages:
```
kodi:~/z2mdeconz # docker-compose logs -f
WARN[0000] /storage/z2mdeconz/docker-compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion
zigbee2mqttdeconz  | Using '/app/data' as data directory
zigbee2mqttdeconz  | Starting Zigbee2MQTT without watchdog.

zigbee2mqttdeconz  | [2025-04-03 21:17:06] warning:     z2m: Failed to ping '0x001788010833c343' (attempt 1/1, ZCL command 0x001788010833c343/11 genBasic.read([                                                                                               "zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (6) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:17:18] warning:     z2m: Failed to ping '0xd0cf5efffe227377' (attempt 1/1, ZCL command 0xd0cf5efffe227377/1 genBasic.read(["                                                                                               zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (7) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:17:31] warning:     z2m: Failed to ping '0xd0cf5efffe29f041' (attempt 1/1, ZCL command 0xd0cf5efffe29f041/1 genBasic.read(["                                                                                               zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (8) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:17:43] warning:     z2m: Failed to ping '0x0017880106a1355d' (attempt 1/1, ZCL command 0x0017880106a1355d/11 genBasic.read([                                                                                               "zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (9) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:17:55] error:       z2m: Failed to read state of 'plug_tuya_electricblanket' after reconnect (ZCL command 0xa4c1383da06b526f                                                                                               /1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"write                                                                                               Undiv":false}) failed (no response received (10) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:17:56] warning:     z2m: Failed to ping 'light_ikea_eettafel1' (attempt 1/1, ZCL command 0x000b57fffee1b446/1 genBasic.read(                                                                                               ["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}                                                                                               ) failed (no response received (11) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:18:08] warning:     z2m: Failed to ping '0x00178801082ff074' (attempt 1/1, ZCL command 0x00178801082ff074/11 genBasic.read([                                                                                               "zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (12) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:18:20] warning:     z2m: Failed to ping '0x00178801061e6bbd' (attempt 1/1, ZCL command 0x00178801061e6bbd/11 genBasic.read([                                                                                               "zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (13) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:18:33] warning:     z2m: Failed to ping 'curtain_tuya_diningroom' (attempt 1/1, ZCL command 0xa4c138c9a42677cf/1 genBasic.re                                                                                               ad(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":fal                                                                                               se}) failed (no response received (14) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:18:45] warning:     z2m: Failed to ping '0xec1bbdfffead1dcf' (attempt 1/1, ZCL command 0xec1bbdfffead1dcf/1 genBasic.read(["                                                                                               zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (15) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:18:58] warning:     z2m: Failed to ping '0xa4c138b60de0ab83' (attempt 1/1, ZCL command 0xa4c138b60de0ab83/1 genBasic.read(["                                                                                               zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (16) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:19:10] warning:     z2m: Failed to ping '0xa4c13871b524231f' (attempt 1/1, ZCL command 0xa4c13871b524231f/1 genBasic.read(["                                                                                               zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})                                                                                                failed (no response received (17) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:19:23] warning:     z2m: Failed to ping 'light_ecodim_livingroom_ceiling' (attempt 1/1, ZCL command 0x0c4314fffefc15cb/1 gen                                                                                               Basic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUn                                                                                               div":false}) failed (no response received (18) Error: waiting for response TIMEOUT))
zigbee2mqttdeconz  | [2025-04-03 21:19:35] warning:     z2m: Failed to ping '0xa4c1385679408fff' (attempt 1/1, ZCL command 0xa4c1385679408fff/1 genBasic.read(["                                                                                               zclVersion"], {" [... cut ...] lse,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (20) Error: waiting for response TIMEOUT)'

```

### Zigbee2MQTT version

2.2.20

### Adapter firmware version

0x26780700

### Adapter

deconz (    ConBee2/RaspBee2)

### Setup

plain on docker container

### Debug log

[log.log](https://github.com/user-attachments/files/19593462/log.log)

NOTE: it seems the tuya plugs are spamming the network, sending multiple messages per second

Nerivec, 2025-04-03:
> Warning: Conbee 2 firmware versions newer than 0x26580700 will result in an unstable network with devices dropping randomly

Can you try downgrading the firmware to 0x26580700? See https://github.com/Koenkk/zigbee2mqtt/issues/9554

_Note that deconz is no longer recommended. There hasn't been a maintainer of the codebase for quite a while, and this lack is creating more problems as time passes (on top of the [missing features](https://www.zigbee2mqtt.io/guide/adapters/deconz.html#deconz-dresden-elektronik))._

ChessSpider, 2025-04-05:
hi @Nerivec 
thank you for the reply! The issue started happening when i upgraded to 2.x.x and it became very critical on 2.2.0.
I have now upgraded to 2.2.1 and it seems to be OK again, even though the changelog does seem to have changed something affecting this.
Inspecting logs now also shows the tuya devices have quietet down and are no longer spamming the zigbee network multiple times per scond.
Im closing this for now!

Nerivec, 2025-04-06:
Did you try changing the reporting settings to see if the device will allow lowering the update rate?

ChessSpider, 2025-04-06:
When a tuya plug is spamming, the zigbee network appears too overloaded to send commands. Those fail. 

I somehow managed to get the plugs to stop spamming again and my zigbee network is slowly recovering. This is the reporting tab of one of the plugs which was spamming earlier. Im not really sure what the settings mean or what the correct settings are

[image]

Nerivec, 2025-04-07:
https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-device-configure-reporting

These settings are explained in that section.
Note that if the behavior is intermittent, it probably won't make a difference, it's likely a device firmware bug, rather than a reporting config issue.

ChessSpider, 2025-04-07:
`currentSumDelivered` has a min rep change of `null`. Does that mean it could send every status update?

ChessSpider, 2025-04-07:
I thought it were only the tuya plugs, but its not. Many devices are spamming the network, e.g. 
```
zigbee2mqttdeconz  | [2025-04-07 22:54:46] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:46.510Z","linkquality":255,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:46] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:46.510Z","linkquality":255,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:47] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:47.700Z","linkquality":255,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:47] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:47.700Z","linkquality":255,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:48] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:48.908Z","linkquality":255,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:48] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:48.908Z","linkquality":255,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:51] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:51.316Z","linkquality":252,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:51] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor_tuya_bathroom', payload '{"battery":7,"humidity":35,"last_seen":"2025-04-07T20:54:51.316Z","linkquality":252,"temperature":18.1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2500}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.271Z","linkquality":232,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.271Z","linkquality":232,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.559Z","linkquality":152,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.559Z","linkquality":152,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.674Z","linkquality":236,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.674Z","linkquality":236,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.886Z","linkquality":200,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:54:59] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:54:59.886Z","linkquality":200,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:55:00] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:55:00.078Z","linkquality":180,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:55:00] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:55:00.078Z","linkquality":180,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:55:00] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:55:00.302Z","linkquality":164,"options":null,"position":0,"running":false,"state":"CLOSE"}'
[... cut ...]
zigbee2mqttdeconz  | [2025-04-07 22:55:02] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:55:02.075Z","linkquality":148,"options":null,"position":0,"running":false,"state":"CLOSE"}'
zigbee2mqttdeconz  | [2025-04-07 22:55:02] info:        z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/curtain_tuya_diningroom', payload '{"last_seen":"2025-04-07T20:55:02.075Z","linkquality":148,"options":null,"position":0,"running":false,"state":"CLOSE"}'
```

the curtain_tuya doesnt even have reporting settings according to the GUI.

:(

ChessSpider, 2025-04-10:
i just downgraded to version 0x26580700 .. otherwise ill start looking into getting a newer zigbee controller

ChessSpider, 2025-04-11:
Network still unstable. No difference :(

Nerivec, 2025-04-11:
Tuya devices are quite terrible at spamming indeed. Not much that can be done for this, except _not use Tuya_...

Any chance some interference could be affecting your network? Something could have changed in your environment recently (could be a neighbor setup a WiFi that's conflicting with your ZigBee channel for example...). Combined with spammy devices, it unfortunately does not take much interference to cause stability troubles...

ChessSpider, 2025-04-11:
hi @Nerivec , thank you so much for the quick responses and thinking along. I really appreciate it. 

I understand there are no easy fixes, and the tuya spamming may have been happening for a long time without causing an issue. 

Maybe ill try to re-pair all devices from scratch, or first get a new controller. Do you think a slzb-07p10 is a good future proof choice ? I have about 50 devices total currently :-) 

(unless there are further debugging steps we can investigate. System load is not an issue i think, cpu/mem usage is low and system feels snappy)

Nerivec, 2025-04-11:
@Koenkk any other ideas here?
And comment on the device choice since zStack is not my forte...
_For [EmberZNet-based](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#usb), the SLZB-07mg24 appears to be pretty good so far._

Koenkk, 2025-04-13:
> Ever since upgrade to 2.x.x branch of zigbee2mqt

One thing you can do is downgrade back to 1.x and see if that fixes things (you will also need to downgrade your `configuration.yaml`)

Regarding adapters, I would always pick a USB one, mg24 or any zstack 1352/2652 should be fine.

ChessSpider, 2025-04-14:
I downgraded to v1.42.0.

The good news is that it does not appear to be zigbee2mqtt related.

The bad news is that my zigbee network remains unstable. 

Ive decided that Ill probably need to rebuild my network from scratch, and I'd rather do that on a newer and supported adapter. So I bought a SLZB-07P7, which will arrive somewhere next week. 

Thank you for all help everyone!

closed 2025-04-14 by ChessSpider

ChessSpider, 2025-04-19:
I figured it out. 

My zigbee2mqtt instance is running on the RPI and is attached to the the TV. Its HDMI cable is causing the distortion. I noticed the zigbee network only had issues when the TV is on. Unplugging the HDMI cable fixed all connectivity issues.

Koenkk, 2025-04-21:
@ChessSpider added to the docs! I would recommend using an USB extension cable as described in the docs.