Koenkk/zigbee2mqtt #26982: [New device support]: Dresden Elektronik Phoscon FLS-M device joint as not supported

blastwavehosting, 2025-04-03:
### Link

https://phoscon.de/en/flsm/

### Database entry

{"id":62,"type":"Router","ieeeAddr":"0xf0f5bdfffe0e3958","nwkAddr":56433,"manufId":4405,"manufName":"dresden elektronik","powerSource":"DC Source","modelId":"FLS-M","epList":[1,242,2,3,4,5],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":0,"swBuildId":"1135_000D_0001","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1743717552281}

### Zigbee2MQTT version

2.2.0-1

### Comments

join the device

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['FLS-M'],
    model: 'FLS-M',
    vendor: 'dresden elektronik',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"5":5}}), m.light(), m.light(), m.light(), m.light(), m.light()],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

can not address different brightness, now is set to 5 individual led strips and only see one in z2m

Koenkk, 2025-04-04:
Can you try with:

```js
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['FLS-M'],
    model: 'FLS-M',
    vendor: 'dresden elektronik',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"5":5}}), m.light({endpointNames: ['1', '2', '3', '4', '5']})],
    meta: {"multiEndpoint":true},
};
```

blastwavehosting, 2025-04-05:
Hi, it works. Thank you Koenkk

closed 2025-04-05 by blastwavehosting

Koenkk, 2025-04-06:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/dresden_elektronik.ts)?

blastwavehosting, 2025-04-06:
Hi, i have added the new code via the url provided. This is my first interaction with GitHub hope is ok. Thank you.

Koenkk, 2025-04-06:
Thanks!

Could you also submit a picture for the docs? This can be done by:

1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)

2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `FLS-M.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

tetsuo-repo, 2025-05-03:
Not sure if this is a proper action, to comment on an existing issue or if I should create a new one for this topic. Since the PR to add FLS-M as supported device, I'm unable to change colors via the device.

I was struggling to register it as a new one, as the device allows so many configuration options that it was not clear to me how to achieve that with z2m.

The external definition that worked for changing colors (but just for the whole strip) is:
```yml
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['FLS-M'],
    model: 'FLS-M',
    vendor: 'dresden elektronik',
    description: 'Automatically generated definition',
    extend: [m.light({"colorTemp":{"range":[140,625]},"color":{"modes":["xy","hs"]}})],
    meta: {},
};
```
Any advice on how both requirements can coexist are welcome. Had to downgrade to an older version for now.