Koenkk/zigbee2mqtt #26983: After updating to 2.2.0 + zstack 20250321 Availability shows as Offline

JornDL, 2025-04-04:
### What happened?

After updating to 2.2.0 + zstack 20250321 Availability shows as Offline.

But everything is working ?

### What did you expect to happen?

Availability should show as Online

### How to reproduce it (minimal and precise)

After updating to 2.2.0 + zstack 20250321 Availability constantly shows as Offline.

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20250321

### Adapter

Sonoff ZBDongle-P

### Setup

Docker container - plain - Homebridge - x86

vsutra, 2025-04-04:
I have the same problem after upgrading to 2.2.0, but with a Conbee2 adapter.

gr3k, 2025-04-04:
I have this same for SLZB-06 and all my devices shows as offline. The good news is ,that it seems everything is working.
Restarting HA ,etc won't help.

[image]

PWnet, 2025-04-04:
Same problem here: Z2M versione 2.2.0, coordinator version: 20240710, but all the devices are working.

[image]

eriktack, 2025-04-04:
Seems fixed in dev already:
https://github.com/Koenkk/zigbee2mqtt/issues/26942#issuecomment-2773510637

tjneutral, 2025-04-04:
That i did is to enable availability to 30min. All devices are ok now.

closed 2025-04-04 by Nerivec