Koenkk/zigbee2mqtt #26987: Problem updating frontend settings when running as HA addon

sarthurdev, 2025-04-04:
### What happened?

Reported by @soif in https://github.com/Koenkk/zigbee2mqtt/pull/26805#issuecomment-2777726422

Seems that not only the filter value, but any changes to frontend settings are ignored.

Websocket log when updating frontend values from frontend, and resulting config:
```
{"topic":"bridge/request/options","payload":{"options":{"frontend":{"notification_filter":["z2m: Failed to ping.*"],"url":"http://asd","auth_token":"asdf"}},"transaction":"ugjw7-4"}}
{"payload":{"level":"info","message":"z2m: Successfully changed options"},"topic":"bridge/logging"}
{"payload":{"commit":"unknown","config":{ ... "frontend":{"base_url":"/","enabled":true,"port":8099}, ... }},"topic":"bridge/info"}
```

### What did you expect to happen?

Websocket log when updating frontend settings on bare-metal z2m 2.2.0 install:
```
{"topic":"bridge/request/options","payload":{"options":{"frontend":{"notification_filter":["z2m: Failed to ping.*","thisisatest"]}},"transaction":"yqsfr-1"}}
{"payload":{"level":"info","message":"z2m: Successfully changed options"},"topic":"bridge/logging"}
{"payload":{"commit":"c5c07e7d","config":{ ... "frontend":{"base_url":"/","enabled":true,"notification_filter":["z2m: Failed to ping.*","thisisatest"],"port":80}, ... },"topic":"bridge/info"}
```

### How to reproduce it (minimal and precise)

Running in HAOS addon, update frontend settings using UI.

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20240710

### Adapter

zstack

### Setup

HAOS 15.1, amd64 VM, Zigbee2mqtt 2.2.0 addon

sarthurdev, 2025-04-04:
Wondering if this an issue with the docker entrypoint setting an env var for the whole frontend config section.

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/master/common/rootfs/docker-entrypoint.sh#L58
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/settings.ts#L364

closed 2025-04-04 by sarthurdev

soif, 2025-04-04:
You are actually my hero. 😎 

Congrats for the ultra quick fix and happy that my report lead you to discover/fix a larger bug ! 🥂