Koenkk/zigbee2mqtt #26990: Try to overwrite unchanged yaml on read-only file system on 2.2.0

qraynaud, 2025-04-04:
### What happened?

I was using the following trick to use docker secrets for my mosquitto password:

`configuration.yaml`:
```yaml
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
  user: mosquitto
  password: '!/run/secrets/secrets.yaml mosquitto_password'
```

`/run/secrets/secrets.yaml`:
```yaml
mosquitto_password: strong-passwd
```

This worked like a charm on `2.1.x` but stopped working on `2.2.2` with the following logs:
```
zigbee2mqtt  | Error: EROFS: read-only file system, open '/run/secrets/secrets.yaml'
zigbee2mqtt  |     at Object.writeFileSync (node:fs:2426:20)
zigbee2mqtt  |     at writeIfChanged (/app/lib/util/yaml.ts:43:12)
zigbee2mqtt  |     at Object.updateIfChanged (/app/lib/util/yaml.ts:51:9)
zigbee2mqtt  |     at Object.write (/app/lib/util/settings.ts:200:22)
zigbee2mqtt  |     at onboard (/app/lib/util/onboarding.ts:538:18)
zigbee2mqtt  |     at start (/app/index.js:137:31)
```

Reading the codebase this seems to be related to dumping configuration changes. But I don't understand how this particular file could have changed in any way.

This is somewhat related to #25113 but not totally though.

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

N/A

### Adapter

N/A

### Setup

docker

qraynaud, 2025-04-04:
I tried to add some log to find more details so I can help a little. I compared `before` and `content` in the `writeIfChanged` method and here is the result:
```
BEFORE: { mosquitto_password: 'YNuvUGjDL6R3mXvi@3Xf3&oTXt83xH' }
 AFTER: { mosquitto_password: '!/run/secrets/secrets.yaml mosquitto_password' }
```

Seems like the value in mosquitto_password is now somewhat recursive and referencing itself. Strange…

qraynaud, 2025-04-04:
Note : seems like removing the `ZIGBEE2MQTT_CONFIG_MQTT_PASSWORD` env variable after the 1st start fixes the issue…

Koenkk, 2025-04-04:
Fixed in 2.2.1!

closed 2025-04-04 by Koenkk