Koenkk/zigbee2mqtt #26992: External extensions are started twice on startup

pyos, 2025-04-05:
### What happened?

When the `ExternalExtensions` extension starts, it calls `Controller`'s `addExtension` with all custom extensions, which adds them to `this.extensions` and calls `startExtension`. Because this is done inside the loop over `this.extensions` in `Controller.start`, that loop will eventually reach these new entries and call `startExtension` on them again.

### How to reproduce it (minimal and precise)

1. Create any external extension that logs in start() and stop()
2. Restart z2m
3. Observe 2 logs from start and 0 from stop

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

Irrelevant

### Adapter

Irrelevant

### Setup

Irrelevant

Koenkk, 2025-04-05:
Would you mind submitting a PR to fix this? Let me know if I can help somehow.

pyos, 2025-04-07:
> Would you mind submitting a PR to fix this? Let me know if I can help somehow.

ok, #27032

closed 2025-04-18 by Nerivec