Koenkk/zigbee2mqtt #26993: 2.2.1 After upgrading to version 1, the light sensor of Tuya ZY-M100-S_2 does not work properly

keidischoi, 2025-04-05:
### What happened?

After upgrading to version 2.2.1, the light sensor of Tuya ZY-M100-S_2 seems to have stopped working.

### What did you expect to happen?

There was no problem with the 2.2.0 version, but the problem occurred after the upgrade.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

ZBDongle-E

### Setup

proxmox,lxc

### Debug log

none

[log.log](https://github.com/user-attachments/files/19613451/log.log)
[log1.log](https://github.com/user-attachments/files/19613450/log1.log)
[log2.log](https://github.com/user-attachments/files/19613449/log2.log)

Koenkk, 2025-04-05:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-04-05 by Koenkk

Diegocampy, 2025-04-05:
I have 5 of these sensors Everything works except brightness. the value does not update I write it only to inform.  I hope for the next update