Koenkk/zigbee2mqtt #26998: Lots of devices offline with BUFFER FULL error....

pauloon, 2025-04-05:
### What happened?

I've upgraded to 2.2.1-1 and after a few hours I have a bunch of devices offline with BUFFER FULL error.

[image]

This is the second time this happens. Also had the same in one past update.
FIrst time I restored the backup and it solved.
Now I'm gonna restore again, because I need my automation working.

Thanks,
Paulo.

### What did you expect to happen?

To be working....

### How to reproduce it (minimal and precise)

Not sure....

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

20240710

### Adapter

Sonoff USB Dongle-P

### Setup

HASSIO latest version running on mini PC

### Debug log

[45df7312_zigbee2mqtt_2025-04-05T10-05-43.316Z.log](https://github.com/user-attachments/files/19614861/45df7312_zigbee2mqtt_2025-04-05T10-05-43.316Z.log)

pauloon, 2025-04-05:
I noticed all offline devices are lights, smart switches and zigbee routers. The was showing ok....

Koenkk, 2025-04-05:
Please try the fw from https://github.com/Koenkk/Z-Stack-firmware/discussions/545 and provide feedback there.

closed 2025-04-05 by Koenkk

pauloon, 2025-04-05:
Great! That fixed. Thanks @Koenkk