Koenkk/zigbee2mqtt #26999: Niko 552-721X2 & 552-721X1 stuck in thinking it's disconnected

desjoerd, 2025-04-05:
### What happened?

After a random amount of hours the Niko 552-721X2 & 552-721X1 start flashing red (which means they think they are disconnected). All the functions still work though.

### What did you expect to happen?

That the switch doesn't get into the disconnected state.

### How to reproduce it (minimal and precise)

Join the switch into the network.
Wait for a day.
Both switches will be flashing red.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Z2M with Mosquitto and HA Core via docker compose. Raspberry Pi 5, 256GB SSD, Ubuntu 24.04 LTS

Bwko, 2025-04-05:
@desjoerd 

I’m seeing similar kinds of issues. I have around 30 Niko switches, and a few of them (about 2 out of 15) of type 552-721X seem to think they’re no longer connected on a weekly(not daily) basis. The other switches of type 552-72201 don’t appear to have these problems. I just checked my logs and I can see that switches 552-721X regularly show the following error message:
Error: ZCL command 0x70ac08fffef3cb68/1 manuSpecificNikoConfig.read(["ledSyncMode"], .... failed (Status 'UNSUPPORTED_ATTRIBUTE')....

desjoerd, 2025-04-05:
This afternoon I've updated zigbee2mqtt and have set the logs to debug. Now I am waiting again till one of them goes red and check the logs ^^. Till now the support of Niko was, zigbee2mqtt is open source and it's best effort. But maybe they will change their stance on it as it's one of the few/only EU switches which look "normal" and are available in the Netherlands.

desjoerd, 2025-04-06:
I've had the issue again an hour ago on the double switch, after removing and adding again the single switch also started blinking. Did not find anything in the logs though. I disabled avilability because that was causing quite a bit of noise in the logging. I think I dont see the error `manuSpecificNikoConfig.read(["ledSyncMode"], .... failed (Status 'UNSUPPORTED_ATTRIBUTE')` but it could be because I've updated Zigbee2mqtt to 2.2.1 which acording the change log has updates in the converters for the Niko switches.

I've also noticed that both switches have a binding by default to the coordinator, on gen OnOff. I removed those and they still seem to work (decoupled), so maybe that also helps..... But we will have to see and wait again.... :P.

desjoerd, 2025-04-06:
And again, blinking red switches, so removing the bindings does not help 😅.

Tomorrow I am going to do a new dive into the logs.

desjoerd, 2025-04-07:
When looking at the logs, the only thing I can find between it was correctly working and blink red was the following:

```
[2025-04-06 19:20:46] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: <-- [254,32,68,129,0,0,25,0,239,225,1,13,0,76,0,251,45,141,0,0,12,1,28,1,0,95,18,4,170,6,0,59,17,239,225,29,37]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,25,0,239,225,1,13,0,76,0,251,45,141,0,0,12,1,28,1,0,95,18,4,170,6,0,59,17,239,225,29,37]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,239,225,1,13,0,76,0,251,45,141,0,0,12,1,28,1,0,95,18,4,170,6,0,59,17,239,225,29] - 37
[2025-04-06 19:20:46] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":57839,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":76,"securityuse":0,"timestamp":9252347,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,28,1,0,95,18,4,170,6,0,59,17]}}
[2025-04-06 19:20:46] debug: 	zh:controller: Received payload: clusterID=25, address=57839, groupID=0, endpoint=1, destinationEndpoint=13, wasBroadcast=false, linkQuality=76, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":28,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4703,"imageType":43524,"fileVersion":289079302},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-04-06 19:20:46] debug: 	z2m: Received Zigbee message from 'woonkamer_switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":289079302,"imageType":43524,"manufacturerCode":4703}' from endpoint 1 with groupID 0
[2025-04-06 19:20:46] debug: 	z2m: Device 'woonkamer_switch' requested OTA
[2025-04-06 19:20:46] debug: 	zh:controller:endpoint: CommandResponse 0x38398ffffe8fb4e2/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false})
[2025-04-06 19:20:46] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x38398ffffe8fb4e2:57839/1 (0,0,1)
[2025-04-06 19:20:46] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":57839,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":247,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,28,2,152]}}
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:writer: --> frame [254,14,36,1,239,225,1,1,25,0,247,0,30,4,25,28,2,152,78]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-04-06 19:20:46] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,247,49]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,247,49]
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,247] - 49
[2025-04-06 19:20:46] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":247}
[2025-04-06 19:20:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-04-06 19:20:46] debug: 	z2m: Responded to OTA request of 'woonkamer_switch' with 'NO_IMAGE_AVAILABLE'
[2025-04-06 19:20:47] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,224,210,0,176]
[2025-04-06 19:20:47] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,224,210,0,176]
[2025-04-06 19:20:47] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [224,210,0] - 176
```

So it looks like it's maybe caused by an Ota request of the switch. Going for a new round.

desjoerd, 2025-08-11:
Yes, both the 552-721X1 & 552-721X2 still have this problem. Sometimes when I turn off and in the power in the house, they will stop blinking, only for at max a day.

Interesting that it could be a regression between 1.x and 2.x. That gives more room for investigation.

GeertB69, 2025-08-20:
@desjoerd , after some days of trial and error i managed to get a stable network.  No more red lights since a couple of days. 
I tried various combinations of new HAOS installations in combination with ZHA and Z2MQTT (1.4.x ; 2.6 etc )  I only added the NIKO switches and motor controllers, outcome had mixed results. Stable for few hours, then suddenly red lights on few switches (even on switched very close to the zigbee radio) and thus also not functioning/reachable anymore.

My current setup which seems to be stable now for over 24 hours is using the SONOFF Dongle E with the custom ember firmware 7.4.5.0 and Z2MQTT 2.6.0.    No more red lights.

When pairing the NIKO single connected switches, i got the below error (also with other Z2MQTT versions ; not in ZHA). All seems to work fine though.  Maybe @Koenkk  can have a look at this?

[image]

When turning on/off a switch, i noticed the below in the log viewer:

2025-08-20 08:27:30.244 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.lichtschakelaar_overloop_operation_mode: 'unknown' (valid options: ['control_relay', 'decoupled'])
2025-08-20 08:27:30.255 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.lichtschakelaar_overloop_operation_mode: 'unknown' (valid options: ['control_relay', 'decoupled'])
2025-08-20 08:27:31.478 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.lichtschakelaar_overloop_operation_mode: 'unknown' (valid options: ['control_relay', 'decoupled'])
2025-08-20 08:27:31.505 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.lichtschakelaar_overloop_operation_mode: 'unknown' (valid options: ['control_relay', 'decoupled'])

the Z2MQTT log shows below after turning on/off the switch:

[2025-08-20 08:27:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschakelaar_Overloop', payload '{"action_reporting":null,"led_enable":true,"led_state":"OFF","led_sync_mode":null,"linkquality":172,"operation_mode":"unknown","state":"ON"}'
[2025-08-20 08:27:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschakelaar_Overloop', payload '{"action_reporting":null,"led_enable":true,"led_state":"OFF","led_sync_mode":null,"linkquality":172,"operation_mode":"unknown","state":"ON"}'
[2025-08-20 08:27:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschakelaar_Overloop', payload '{"action_reporting":null,"led_enable":true,"led_state":"OFF","led_sync_mode":null,"linkquality":172,"operation_mode":"unknown","state":"OFF"}'
[2025-08-20 08:27:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lichtschakelaar_Overloop', payload '{"action_reporting":null,"led_enable":true,"led_state":"OFF","led_sync_mode":null,"linkquality":172,"operation_mode":"unknown","state":"OFF"}'

As said, light turns off/on fine despite these errors.  Also when using the physical switches (most can be switched from 2 - 3 places , what we call a 'wissel/hotel schakeling' or 'kruisschakeling') ;)

the switch in Z2MQTT:

[image]

Cannot really draw conclusions on which combination of dongle/Z2MQTT version to use.  My current combination works fine and hope it remains like that.

GeertB69, 2025-08-21:
@Koenkk , sure, i will do today.

And bad news, the red lights are back since last night.  The only thing which has changed is adding my other zigbee devices.  Luckily i took a number of backups , before adding devices , NIKO only and final setup including all devices.

Going to try to narrow it down.

GeertB69, 2025-08-21:
@Koenkk ,
[45df7312_zigbee2mqtt_2025-08-21T06-39-36.607Z.log](https://github.com/user-attachments/files/21911864/45df7312_zigbee2mqtt_2025-08-21T06-39-36.607Z.log)

I removed one of the NIKO switched and paired again with DEBUG.  Device is called 'Lichtschakelaar_Overloop'

thanks in advance.

GeertB69, 2025-08-21:
@Koenkk , i restored the backup with only the NIKO devices but it seems since i paired other devices after that yesterday, some of them are joining even Permit join is disabled.  (only the Philips HUE motion sensors and some SONOFF temperature sensors)  
Not directly related to my issue but is that normal behaviour?

I forced off the NIKO switches and motor controls and paired them again.  I left DEBUG on to see if later tonight or tomorrow morning they went red blinking again. 

Note sure if this is important but I also noticed a difference in operation mode between identical switches (both in a 'kruisschakeling' /  3-point physical switch setup)

[image]

[image]

both are 

[image]

I am still on 

[image]

[image]

Antigen-sibella, 2025-08-26:
With regards to action reporting: my niko devices, which have had numerous issues similar to to the ones discussed here "forget" action reporting between 1 to a random number of home assistant restarts. When it happens they report action reporting: null. I wrote a nodered automation which automaticly corrects this, but I don't think this should be normal behaviour. 

Furthermore, certain functions like "Led Sync Mode" always give errors when configuring these devices or when sending a command to change the mode. It seems that this function is not supported.

"z2m: Publish 'set' 'led_sync_mode' to 'Badkamer - Ventilatie Schakelaar' failed: 'Error: ZCL command 0xcc86ecfffee624d8/1 manuSpecificNikoConfig.write({"ledSyncMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4703,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'"

GeertB69, 2025-08-30:
I gave up on this. Spend too much time testing various combinations of Z2MQTT and adapters.
Bought me a second-hand niko wireless smart hub, added all devices and then installed the Niko Home Control II integration (https://github.com/joleys/niko-home-control-II ) , all working perfect now.

Hinchukardt, 2025-10-22:
I had this same issue. Hue hub and Z2M were interfering both on channel 11. Switched to channel 25 for Z2M and haven't had this issue anymore.

I think the niko loses connection and goes into error mode instead of trying to reconnect because a reboot would fix the problem for a few hours or days.

closed 2025-12-30 by github-actions[bot]