Koenkk/zigbee2mqtt #27002: External converters not loading in 2.2.1

jesse-kaufman, 2025-04-05:
### What happened?

When starting, my unsupported water level sensor still shows as unsupported even though I have a working external converter for it in data/external_converters/TZE200.js (I'm in a Docker environment)

The log entry seems to imply it's looking inside the dist folder for external_converters, which is different from before. I don't recall seeing anything in 2.2.0 or 2.2.1 release notes saying it had moved.

### What did you expect to happen?

I expected the sensor to show as supported like in 2.0.x

### How to reproduce it (minimal and precise)

Upgrade to 2.2.1 and restart Z2M

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240710

### Adapter

SMLIGHT SLZB-06 (non-M)

### Setup

Docker

### Debug log

[2025-04-05 07:08:40] error: 	z2m: Failed to start 'ExternalConverters' (Error: ENOENT: no such file or directory, mkdir '/app/dist/external_converters'

jesse-kaufman, 2025-04-05:
I was successfully able to map my external_converters dir to /app/dist/external_converters in my Docker compose and it works now, but I assume that isn't the intended way for it to work.

skvarel, 2025-04-05:
> I would check the documentation ?
> 
> https://www.zigbee2mqtt.io/advanced/more/external_converters.html
> 
> It seems you should use *.mjs extension

That's not necessary.

*.js works on v2.1.3

[image]

[image]

@jesse-kaufman ... when updating to v2.1.3, I had to delete and recreate the directory. Then the converters worked again.

jesse-kaufman, 2025-04-05:
The .mjs extension isn't necessary. It's looking in /app/dist/external_converters instead of /app/data/external_converters and tries to create the directory if missing (hence the ENOENT error).

I was able to map my external_converters directory to /app/dist/external_converters and it works fine with no other changes (still named .js)

> [@jesse-kaufman](https://github.com/jesse-kaufman) ... when updating to v2.1.3, I had to delete and recreate the directory. Then the converters worked again.

I'm coming from 2.1.3 upgrading to 2.2.1

chris-1243, 2025-04-05:
@jesse-kaufman 

```
[2025-04-05 19:57:31.097] info: 	z2m: Loaded external converter 'SR-ZG9101CS.mjs'.
```

This is from Z2M 2.2.1... if *.mjs is now the prefered way to go, I would consider to move your converters to this extension.
I do agree *.js file should be also supported

jesse-kaufman, 2025-04-05:
> [@jesse-kaufman](https://github.com/jesse-kaufman)
>
> ```
> [2025-04-05 19:57:31.097] info: 	z2m: Loaded external converter 'SR-ZG9101CS.mjs'.
> ```
>
> This is from Z2M 2.2.1... if *.mjs is now the prefered [...]

ok that's fine and all, but that doesn't address the fact that it's looking in /app/dist/external_converters instead of /app/data/external_converters .. renaming the file to .mjs won't fix the error: Error: ENOENT: no such file or directory, mkdir '/app/dist/external_converters'

chris-1243, 2025-04-05:
I am also on Docker and Z2M 2.2.1 and I am not able to reproduce your error

Nerivec, 2025-04-05:
The "current version" of converters are copied to the `/app/dist` folder as part of the dynamic importing & MQTT-enabled edition. Z2M handles that on its own. You should _not_ be putting converters in there yourself (or mapping), it's a temporary runtime-only mechanism.

> Error: ENOENT: no such file or directory, mkdir '/app/dist/external_converters'

It seems for some reason, your container is unable to create the temporary directory to then allow Z2M to copy into. The error is a bit strange though.
Are you overriding anything in your compose file that could potentially affect this?
Do you have the full error in the logs (stack trace that points to a specific line in Z2M)?

closed 2025-06-13 by github-actions[bot]