Koenkk/zigbee2mqtt #27004: External converter don't work correctly.

gargarse, 2025-04-05:
### What happened?

Hello
I'm using z2m v1.42.0, and I'm having no problems using an external converter. I worked with this converter since version 1.28.0. 
However, when I upgrade to v2 and move the converter to the external_converters directory to load it, I get the following error when I start z2m:

**converter "legacy.firstDpValue" is not a function**

Could you help me get the converter to read correctly in z2m v2?

I'm attaching the converter.

`const legacy = require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const x5h_set_time_request = {
    cluster: 'manuSpecificTuya',
    type: ['commandMcuSyncTime'],
    convert: async (model, msg, publish, options, meta) => {
        const hour = new Date().getHours();
        if (hour < 9) {
            meta.logger.info('NOT Sync Time message to TGR85-ZB, to solve gap problem');
        } else {
            meta.logger.info('Send Sync Time message to TGR85-ZB');
            const currentTime = new Date().getTime();
            const tzo = new Date().getTimezoneOffset();  // Hours in minute with sign.
        
            // Problem Beok  
            const gap = 8 * 3600; 
         
            const utcTime = Math.round((currentTime) / 1000 ) - (tzo * 60) - gap;
            //const localTime = Math.round(currentTime / 1000) - (new Date()).getTimezoneOffset() * 60;
            const localTime = 0;
            const endpoint = msg.endpoint;
            const payload = {
                payloadSize: 8,
                payload: [
                    legacy.convertDecimalValueTo4ByteHexArray(utcTime),
                    legacy.convertDecimalValueTo4ByteHexArray(localTime),
                ],
            };
            await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
        };
    },
};

const tzDataPoints = {
    ...tuya.tzDataPoints,
    keys: ['system_mode', 'preset', 'running_state', 'current_heating_setpoint', 'upper_temperature', 'local_temperature', 'deadzone_temperature', 'heating_temperature_limit'],
}
const valueConverter = {
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),
    runningState: tuya.valueConverterBasic.lookup({'heat': true, 'idle': false}),
    preset: tuya.valueConverterBasic.lookup({'manual': 0, 'program': 1}),
    divideBy10: tuya.valueConverterBasic.divideBy(10),
};
const fzLocal = {
    x5h_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
[... 291 lines cut ...]
        exposes.numeric('deadzone_temperature', ea.STATE_SET).withUnit('Â°C').withValueMax(9.5)
            .withValueMin(0.5).withValueStep(0.5).withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        exposes.numeric('upper_temp', ea.STATE_SET).withUnit('Â°C').withValueMax(60)
            .withValueMin(35).withValueStep(1).withPreset('default', 35, 'Default value'),
    ],
};

module.exports = definition;`

### Zigbee2MQTT version

1.42.0 commit: 861cba6

### Adapter firmware version

20221226

### Adapter

zStack3x0 - Sonoff 

### Setup

Z2m on docker container

Nerivec, 2025-04-08:
This device should be supported by Z2M.
https://github.com/Nerivec/zigbee-herdsman-converters/blob/240b4fb6c3b010e7dcb2fd234bde792ae0f56c98/src/devices/tuya.ts#L7832-L7893

gargarse, 2025-04-08:
Yes, but this device had a serious problem with time synchronization. See this link: [#14037]
That's the reason for using an external converter. It doesn't respond to the "commandMcuSyncTime" request between midnight and 9:00 a.m.

My question is, how can I get the external converter that worked in z2m version 1.42 to continue working in version 2.x?

It seems the error I'm getting is due to the use of the "legacy.firstDpValue" function.

Nerivec, 2025-04-08:
That converter is seriously outdated, some of that stuff isn't even valid in 1.42. _You might want to try the built-in converter again if you have not recently._

Otherwise, best would be to go from the existing built-in converter and re-do the time-related modifications, something like this:

```mjs
import {ignore_basic_report, ignore_tuya_set_time} from 'zigbee-herdsman-converters/converters/fromZigbee';
import * as exposes from 'zigbee-herdsman-converters/lib/exposes';
import * as legacy from 'zigbee-herdsman-converters/lib/legacy';
import * as tuya from 'zigbee-herdsman-converters/lib/tuya';

const e = exposes.presets;
const ea = exposes.access;

/** @type{import('zigbee-herdsman-converters/lib/types').DefinitionWithExtend | import('zigbee-herdsman-converters/lib/types').DefinitionWithExtend[]} */
export default {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_2ekuz3dz']),
    model: 'X5H-GB-B',
    vendor: 'Tuya',
    description: 'Wall-mount thermostat',
    fromZigbee: [ignore_basic_report, ignore_tuya_set_time, legacy.fromZigbee.x5h_thermostat],
    toZigbee: [legacy.toZigbee.x5h_thermostat],
    whiteLabel: [
        {vendor: 'Beok', model: 'TGR85-ZB'},
        {vendor: 'AVATTO', model: 'ZWT-100-16A'},
    ],
    exposes: [
        e
            .climate()
            .withSetpoint('current_heating_setpoint', 5, 60, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET)
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withPreset(['manual', 'program']),
        e.temperature_sensor_select(['internal', 'external', 'both']),
        e
            .text('schedule', ea.STATE_SET)
            .withDescription(
                'There are 8 periods in the schedule in total. ' +
                    '6 for workdays and 2 for holidays. It should be set in the following format for each of the periods: ' +
                    '`hours:minutes/temperature`. All periods should be set at once and delimited by the space symbol. ' +
                    'For example: `06:00/20.5 08:00/15 11:30/15 13:30/15 17:00/22 22:00/15 06:00/20 22:00/15`. ' +
                    "The thermostat doesn't report the schedule by itself even if you change it manually from device",
            ),
        e.child_lock(),
        e.week(),
        e.enum('brightness_state', ea.STATE_SET, ['off', 'low', 'medium', 'high']).withDescription('Screen brightness'),
        e.binary('sound', ea.STATE_SET, 'ON', 'OFF').withDescription('Switches beep sound when interacting with thermostat'),
        e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('Antifreeze function'),
        e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF').withDescription("Resets all settings to default. Doesn't unpair device."),
        e
            .numeric('heating_temp_limit', ea.STATE_SET)
            .withUnit('°C')
            .withValueMax(60)
            .withValueMin(5)
            .withValueStep(1)
            .withPreset('default', 35, 'Default value')
            .withDescription('Heating temperature limit'),
        e
            .numeric('deadzone_temperature', ea.STATE_SET)
            .withUnit('°C')
            .withValueMax(9.5)
            .withValueMin(0.5)
            .withValueStep(0.5)
            .withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        e
            .numeric('upper_temp', ea.STATE_SET)
            .withUnit('°C')
            .withValueMax(95)
            .withValueMin(35)
            .withValueStep(1)
            .withPreset('default', 60, 'Default value'),
    ],
    onEvent: tuya.onEventSetLocalTime,
};
```

Make sure to save this in a file with `.mjs` extension, I used the [new format](https://www.zigbee2mqtt.io/advanced/more/external_converters.html) available from 2.2.0.

I used the alternative onEvent for tuya, if that doesn't work, you can copy the appropriate function in the converter directly, and make the changes you need:
https://github.com/Nerivec/zigbee-herdsman-converters/blob/240b4fb6c3b010e7dcb2fd234bde792ae0f56c98/src/lib/tuya.ts#L193-L246

gargarse, 2025-04-09:
Thanks @Nerivec 
I'll try to create a new converter for my thermostats using the new APIs in version 2. Worst case, I'll be stuck on version 1.42, where the converter developed in version 1.28 worked fine.

closed 2025-04-09 by gargarse