Koenkk/zigbee2mqtt #27005: Tuya ZY-M100-S_2 illuminance not change

Diegocampy, 2025-04-05:
### What happened?

After the latest update 2.2.1.1  All 5 of my presence sensors continue to work, but then do not update the lighting value.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

7.4.4.0

### Adapter

EZSP v13

### Setup

Add-on

blackst225, 2025-04-05:
Same happend to me on Tuya Presence. They still availiable but no triggers work at all

Scialla, 2025-04-06:
I've two presence sensors, one at 5.8 GHz and one at 24 GHz. Only 5.8 sensor have same problem with `illuminance`

Biero1, 2025-04-06:
I have the same problem.
After the update the illumination sensor is frozen. Presence sensor Tuya ZY-M100-S_2, _TZE204_qasjif9e

Koenkk, 2025-04-06:
Should be fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-04-06 by Koenkk

timmo65, 2025-04-08:
I can confirm this is fixed in the edge version -  I had  3x ZY-M100-S_1s which were always ok and 3x ZY-M100-S_2s with illuminance stuck  on 2.2.1.1 (or unknown when repaired) - back working now - thanks

saunus00, 2025-04-22:
Zigbee2MQTT version: 2.2.1-1 two ZY-M100-S_2 devices.
on one I have Illuminance 0lx and the other Illuminance 1lx so it still doesn't work.

kyrgorsh, 2025-04-22:
same problem. However, I have not installed edge version. waiting for release.

naevtamarkus, 2025-04-28:
I've got the same problem.
Apparently, someone tried the dev branch and didn't work either, see here:
https://github.com/Koenkk/zigbee2mqtt/issues/27062

lukaszzyla, 2025-04-29:
For me illuminance works on both stable and edge HA addons.
check my problem here - for me, the presense is frozen.
https://github.com/Koenkk/zigbee2mqtt/issues/27212
I did install 2.2.0 in docker as recommended but then presence is always true (frozen in true opposed to frozen in null)
Any idea if this is the same issue?
I only bought this sensor and was wondering if it works as motion sensor at all?

naevtamarkus, 2025-05-05:
Hi all! Just wanted to say that 2.3.0 fixed it for me. Thanks!

gedekai, 2025-05-05:
It's true, 2.3.0 fixed it. Everything works fine now!

Diegocampy, 2025-05-05:
I Confirm, all work fine, now!

lukaszzyla, 2025-05-06:
I am in the unlucky team then, because for me it is still the same.
The funny thing is that illuminance was always working for me together with distance.
but the presence is either true or none
I have removed and readded the device so many times an many different versions of z2m.
I was hoping that 2.3.0 fixes it for me like for rest of you but unfortunatelly - no difference.
mqtt broker:

[image]

z2m:
When I add the device it is being interviewed and reports success.
But then presence is frozen on null or true.
nobody's at home for an hour or so and:

[image]

[image]

[image]

[image]

Illuminance levels are far away from my knx sensors that report several thousands lux compared to 2-3 lx from tuya.
Any ideas what to do next?
It never worked for me as I bought it 2 or 3 weeks ago.
Do you think this could be faulty device?
I do not have tuya gateway to check with native app.

Bogdis, 2025-07-14:
Same sensor here bur for me the Presence and range doesn't work or kinda not working. The Presence works sometimes if I set the range to maximim. The range sensor dosen't work at all for me it says unknown.