Koenkk/zigbee2mqtt #27007: STYRBAR recmote control not working

AndreKR, 2025-04-05:
### What happened?

Continuation

### Zigbee2MQTT version

2.2.0

### Adapter firmware version

20220219

### Adapter

CC2652P stick

### Setup

Docker on Raspbarry Pi OS

AndreKR, 2025-04-05:
When I paired it, it showed this:

[image]

Does that look like a successful pairing?

A minute or so later, I saw this:

[image]

Is that an OTA update? Or just a firmware version check? I didn't request an OTA update, at least not since I re-paired the device today.

This is with a display filter `zbee_nwk.addr==0x49c2`.

k1lln1n3, 2025-04-12:
I'm experiencing this as well. Both of my remotes are not working as they're not communicating long press and double press events.

closed 2025-06-20 by github-actions[bot]