Koenkk/zigbee2mqtt #27010: Manhot BX82-TYZ1 curtains motor - remote control works oppositely to zigbee control

dekamilv, 2025-04-06:
### What happened?

Buttons on physical remote gives opposite actions then control via zigbee. Means that if state "open" i have to press "open" button on remote to close the curtains and opposite. "Direction" settings change gives same opposite result.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

2024.03.01

### Adapter

SLZB-06p7

### Setup

add-on on Home Assistant OS

MrCrrispy, 2025-04-14:
I just want to confirm I have the same issue with this motor.
But I have also found that this seem to be common in several other Tuya based covers.

dekamilv, 2025-06-20:
Still no any ideas how to amend it ? :(