Koenkk/zigbee2mqtt #27011: [Feature request]: Add router device property no-join

bongosmurf, 2025-04-06:
### Is your feature request related to a problem? Please describe

I'm always frustrated when I want to add a new device to the network and have to power off my lab/testing devices to prevent them from parenting the new device. Rationale is that I want the new device to join through a device that has an expectation of permanently being part of the network, so that the new device will not have to time out to search for a new parent. Also, there are to my knowledge devices that will not look for a new parent.

### Describe the solution you'd like

I would like to be able to mark devices as exempt from "Permit join (All)".

### Describe alternatives you've considered

I've considered making the join-dropdown a multi-select, but in practise I would expect there to be far fewer non-production routers as opposed to production routers in the network, making selection tedious and error-prone.
I've considered creating a favorites-list of router-devices to include in a "Permit join (Favorites)" selection, but that would also be tedious and error-prone. It is far better to have to take special actions with test devices and not with regular production devices. 

### Additional context

I frequently have new devices/device types that I'm experimenting with in my lab. They have no expectation of being powered while in the lab. Also, they will be located elsewhere, when placed into service.

Nerivec, 2025-04-06:
This is not "feasible", in that it would require sending a permit join request to each device "not filtered out" (as opposed to a single device -1 unicast message-, or all -1 broadcast message-). In large networks, that could result in hundreds of requests, even smaller networks would have a hard time handling this. Any attempt to alleviate this problem would result in mismatching timings, unreliable feedback...

closed 2025-04-06 by Nerivec

bongosmurf, 2025-04-06:
@Nerivec, thanks for elaborating. I understand your reasoning.