Koenkk/zigbee2mqtt #27013: Update device firmware SONOFF SNZB-02D

mrochette01, 2025-04-06:
### What happened?

Hello,
In the latest version of Z2M, I received a message via the OTA tab inviting me to update my two SONOFF SNZB-02D devices. I selected the red "Update device firmware" button. I got an error message telling me that the SONOFF object is not accessible. The state of SONOFF is:
{
    "battery": 60,
    "humidity": 37.4,
    "last_seen": "2025-04-05T21:43:14.474Z",
    "linkquality": 138,
    "temperature": 20.3,
    "update": {
        "installed_version": 4105,
        "latest_version": 8960,
        "state": "available"
    }
}

I get the error message: Z2M: Update of 'Temp: salon_2' failed (Device didn't respond to OTA requets) and Update of 'Temp: salon_2' failed (Device didn't respond to OTA requets)

Thanks for the help.

### What did you expect to happen?

To be able to update the firmware via Z2M.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

2.7.1

### Adapter

SLZB-06

### Setup

HA, Pi4

Nerivec, 2025-04-06:
See https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#troubleshooting

mrochette01, 2025-04-06:
I removed the battery and did a reset. The messages in the log are:

info 2025-04-06 08:19:55z2m: Updating 'Temp: salon_2' to latest firmware
info 2025-04-06 08:21:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Plante_1', payload '{"battery":57,"battery_state":"high","last_seen":"2025-04-06T12:21:16.694Z","linkquality":109,"soil_moisture":66,"temperature":19.2,"temperature_unit":"fahrenheit"}'
info 2025-04-06 08:21:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Temp: salon_2', payload '{"battery":100,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"humidity":46.9,"humidity_calibration":null,"last_seen":"2025-04-06T12:06:45.823Z","linkquality":112,"temperature":21.5,"temperature_calibration":null,"temperature_units":null,"update":{"installed_version":4105,"latest_version":8960,"state":"available"}}'
info 2025-04-06 08:21:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Temp: salon_2' failed (Device didn't respond to OTA request)","status":"error","transaction":"s2c21-2"}'
error 2025-04-06 08:21:18z2m: Update of 'Temp: salon_2' failed (Device didn't respond to OTA request)

chris-1243, 2025-04-06:
I have just updated mine without any issues. As this an end device, you must awake the device by shortly pressing the pairing button at the back of the device until the OTA starts.

Be aware, OTA on this device may kill your battery. According to Sonoff, you should have minimum 70% battery remaining.

mrochette01, 2025-04-06:
The battery was at 100%. I replaced it anyway.
I then proceeded as you suggested, and I still get the same error.

2025-04-06 15:39:16z2m: Updating 'Temp: salon_2' to latest firmware

2025-04-06 15:40:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Temp: salon_2', payload '{"battery":100,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"humidity":46.9,"humidity_calibration":null,"last_seen":"2025-04-06T12:06:45.823Z","linkquality":112,"temperature":21.5,"temperature_calibration":null,"temperature_units":null,"update":{"installed_version":4105,"latest_version":8960,"state":"available"}}'

2025-04-06 15:40:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Temp: salon_2' failed (Device didn't respond to OTA request)","status":"error","transaction":"xq3cu-1"}'

chris-1243, 2025-04-06:
Here is what I did while updating.

I pressed the pairing button at the same time as I clicked `update` in the OTA tab. Then and until the update started, I kept the device awake by pressing every 2-3 seconds the pairing button quickly.

It went smoothly at the first try

mrochette01, 2025-04-06:
I have two devices. I followed your advice with the second one, and the update is in progress.

I'll try it with the first one next.

mrochette01, 2025-04-06:
Everything worked fine for about 25 minutes and then I got a "timeout error".

I repeated the operation twice with the same error. Part of the download worked before the error.

error 2025-04-06 17:06:27z2m: Update of 'Temp: 2e étage' failed ([0x44e2f8fffe17a200 | SNZB-02D] Timeout. Device did not start/finish firmware download after being notified. (Timeout - 9522 - 1 - undefined - 25 - 3 after 150000ms))

mrochette01, 2025-04-06:
I managed to update the firmware on the second device.

I'll try again with the first.

mrochette01, 2025-04-07:
Thanks for your recommendations and comments, I have successfully updated my two SONOFF SNZB-02Ds.

closed 2025-04-07 by mrochette01

MauroMP1, 2025-07-07:
> I have just updated mine without any issues. As this an end device, you must awake the device by shortly pressing the pairing button at the back of the device until the OTA starts.
>
> Be aware, OTA on [...]

Works, thanks

darki79, 2026-02-06:
> Everything worked fine for about 25 minutes and then I got a "timeout error".
>
> I repeated the operation twice with the same error. Part of the download worked before the error.
>
> error 2025-04-06 [...]

Did you solve this problem? I had the same issue, now Z2M says there is no image available alltough the last update was not installed