Koenkk/zigbee2mqtt #27019: [Feature request]: Add router device property favorite

bongosmurf, 2025-04-06:
### Is your feature request related to a problem? Please describe

I'm always frustrated when I want to add a new device to the network and have to power off my lab/testing devices to prevent them from parenting the new device. Rationale is that I want the new device to join through a device that has an expectation of permanently being part of the network, so that the new device will not have to time out to search for a new parent. Also, there are to my knowledge devices that will not look for a new parent.

### Describe the solution you'd like

I would like to be able to mark a limited number of strategically located router devices as favorites and be able to use a "Permit join (Favorites)" selection rather than "Permit join (All)". In order not to overburden the network with individual permit join request messages, the number of favorites should be limited to a dozen or so.

### Describe alternatives you've considered

See 
[Feature request]: Add router device property no-join #27011 

### Additional context

I frequently have new devices/device types that I'm experimenting with in my lab. They have no expectation of being powered while in the lab. Also, they will be located elsewhere, when placed into service.

PayneRae, 2025-04-26:
Hello, have you tried pairing via a specific device: https://www.zigbee2mqtt.io/guide/usage/pairing_devices.html
or binding: https://www.zigbee2mqtt.io/guide/usage/binding.html?

PayneRae, 2025-05-02:
Hi again, after reviewing the documentation and alternative fix more closely, I would recommend pairing via a specific device during onboarding. Suggesting binding was a mistake. Pairing via a specific router or the coordinator would be more tedious and case-by-case than your suggestion, but perhaps it could be a temporary fix. 

I understand @Nerivec described the problem with your previous suggestion #27011 as not feasible due to the many join requests that would be sent out to individual devices overloading the network, and any attempt to alleviate this would cause timing and feedback problems. 

For your favorites idea above, instead of just the single Permit join(all) request being sent, you would also have a Permit join(favorites) request. I wonder, would the network be constrained with just these two requests? Perhaps I have misunderstood and individual device requests would be sent instead.

I am interested in following this issue and seeing the comments. I hope pairing via a specific router or coordinator can be a temporary solution for you.

bongosmurf, 2025-06-03:
Hi,
in the interim, my network has grown quite a bit and it does get tedious selecting a specific device to join to from the dropdown.

@PayneRae, to my understanding, the network is generally not constrained by manually selecting a specific node to for the join, as it is a mesh and most devices are able to reconfigure themselves to use the nearest router. However, there seem to be exceptions that require new pairing.

I'm no longer sure, whether this feature request is worth the effort, so I'm leaning towards closing it.

closed 2025-07-31 by Nerivec