Koenkk/zigbee2mqtt #27021: Failed to start 'Bridge' and 'HomeAssistant' (TypeError: Invalid version. Must be a string. Got type "undefined".)

Frix-x, 2025-04-06:
### What happened?

Hello,

After updating the Home Assistant addon from 2.1.1-1 to 2.2.1-1 I got an error when starting Zigbee2MQTT.
Rolling back to 2.1.1-1 fix the issue.

Here is an extract of the log with the error, related to zigbee-herdsman-converters and Nodon devices (that I'm using):

```
[2025-04-06 20:18:52] error: 	z2m: Failed to start 'Bridge' (TypeError: Invalid version. Must be a string. Got type "undefined".
    at new SemVer (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/classes/semver.js:19:13)
    at compare (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/functions/compare.js:3:3)
    at Object.gt (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/functions/gt.js:2:29)
    at result.exposes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.21.0/node_modules/zigbee-herdsman-converters/src/devices/nodon.ts:112:38)
    at Object.exposes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.21.0/node_modules/zigbee-herdsman-converters/src/index.ts:389:40)
    at Device.exposes (/app/lib/model/device.ts:53:45)
    at Bridge.getDefinitionPayload (/app/lib/extension/bridge.ts:826:29)
    at Bridge.publishDevices (/app/lib/extension/bridge.ts:756:34)
    at Bridge.start (/app/lib/extension/bridge.ts:180:20)
    at Controller.startExtension (/app/lib/controller.ts:268:13))
[2025-04-06 20:18:52] info: 	z2m: Started frontend on port 8099
[2025-04-06 20:18:52] error: 	z2m: Failed to start 'HomeAssistant' (TypeError: Invalid version. Must be a string. Got type "undefined".
    at new SemVer (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/classes/semver.js:19:13)
    at compare (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/functions/compare.js:3:3)
    at Object.gt (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/functions/gt.js:2:29)
    at result.exposes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.21.0/node_modules/zigbee-herdsman-converters/src/devices/nodon.ts:112:38)
    at Object.exposes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.21.0/node_modules/zigbee-herdsman-converters/src/index.ts:389:40)
    at Device.exposes (/app/lib/model/device.ts:53:45)
    at HomeAssistant.getConfigs (/app/lib/extension/homeassistant.ts:1344:36)
    at HomeAssistant.discover (/app/lib/extension/homeassistant.ts:1480:35)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:438:24)
    at processTicksAndRejections (node:internal/process/task_queues:105:5))
```

The log is then normal after this: devices are correctly reporting their states, etc... But I can't control them in Home Assistant as they are all unavailable. Additionally, when opening Zigbee2MQTT frontend, it's completely empty like if all devices are gone. This also happens with the "EDGE" addon version so I'm not sure if it's linked with [the issue that was reported 5 days ago](https://github.com/Koenkk/zigbee2mqtt/issues/26942) or due to the reported Nodon TypeError...

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZnet on a Sonoff Zigbee dongle E

### Setup

Add-on in Home Assistant on a small x86-64 Lenovo computer (HaOS)

Frix-x, 2025-04-06:
Oh! apparently the issue was just fixed in the commit : https://github.com/Koenkk/zigbee-herdsman-converters/commit/138099a1660d1a704af9e20bc70db82326cccac1

But the latest version of the Home Assistant EDGE addon doesn't look to have this fix yet. So probably this issue is already solved. I'll try tomorrow again and see :)

Koenkk, 2025-04-06:
Edge will be updated in 1 hour from now!

closed 2025-04-06 by Koenkk