Koenkk/zigbee2mqtt #27035: MQTT Addon not starting after core update

Illyze, 2025-04-07:
### What happened?

After updating the core and rebooting everything multiple times, the addon still wont start up.

[10:59:41] INFO: Preparing to start...
[10:59:41] INFO: Socat not enabled
[10:59:41] INFO: Starting Zigbee2MQTT...
node:internal/modules/run_main:91
  const type = getNearestParentPackageJSONType(mainPath);
               ^
Error: Invalid package config /app/package.json.
    at shouldUseESMLoader (node:internal/modules/run_main:91:16)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:163:20)
    at node:internal/main/run_main_module:36:49 {
  code: 'ERR_INVALID_PACKAGE_CONFIG'
}
Node.js v22.13.1~~
 

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20250404.0

### Adapter

sonoff zigbee stick

### Setup

I run the add on on a raspberry pi.

Nerivec, 2025-04-08:
Have you tried uninstalling the add-on then installing it again? (_Make sure you don't delete the add-on data!_)

Koenkk, 2025-04-08:
`1.22.1` is a very old z2m version, can you try updating to the latst?

Illyze, 2025-04-08:
sorry it wasnt 1.22.1 but it was 2.2.1, my mistake sorry.
Anyway i put a backup and now i am back to 2.1.3 and it works, i also tried to reinstall it and it didnt work.

Koenkk, 2025-04-09:
Could it be that your SD card is corrupted?

Illyze, 2025-04-09:
really? i am not sure, how can i find out if its corrupted?

Nerivec, 2025-04-09:
It's a common-enough pi problem, there are a lot of guides online, for example:
https://www.howto-do.it/raspberry-pi-sd-card-health-check/

Illyze, 2025-04-10:
Sure, but I never really used this SD Card much, and also all the other addons work correctly and MQTT too when with the older version. Storage is only 10% used.

Nerivec, 2025-04-10:
Thing is, corruption like this is entirely unpredictable, you have to hit exactly the spot to encounter troubles, so, plenty of things could work well, and one just not at all, create weird errors like this, etc.. It would be good if you could at least run a check, so we can eliminate that possibility (if you haven't already).

Illyze, 2025-05-05:
sorry if i come back to this a little later, but i was very buys.
I checked everything, i tried to get some help from chatgpt to help me guide on how to check, and everything seems fine and good on it. Yestarday i also tried to update zigbee to 2.3 and after a bit it got stucked and stopped working again.
Now i tried one more time to update it.
what could i do?

closed 2025-07-12 by github-actions[bot]