Koenkk/zigbee2mqtt #27041: No device list on V2.2.1 after Nodon SIN-4-2-0 paring

Tonio16, 2025-04-08:
### What happened?

Hello

I am using zigbee2mqtt on a proxmox LXC, v2.2.1.
```
Version Zigbee2MQTT
[2.1.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.1.0) commit: [83ff4a76](https://github.com/Koenkk/zigbee2mqtt/commit/83ff4a76)
Type de coordinateur
zStack3x0
Révision du coordinateur
20221226
Coordinator IEEE Address
0x00124b00258d7e3c
Frontend version
0.9.4
Version de zigbee-herdsman-converters
21.27.1
Version de zigbee-herdsman
3.2.5
```
I added to my network two Nodon SIN-4-2-20 from the same order (so I expect that they are similar), I added already a third one in my zigbee2mqtt, also from the same order, in December 2024, without issues.

One of the two new nodon is somehow buggy with zigbee2mqtt V2.2.1:
1) reported  as unsupported though the one already in is ok. Reinterview didn't help.
2) 2nd nodon was difficult to add but it was ok after some trials.
3) I tried to delete the buggy  nodon but it didn't work, with forced delete
4) I restart zigbee2mqtt but I have an issue similar to https://github.com/Koenkk/zigbee2mqtt/issues/26576

I switched back to an old instance of zigbee2mqtt, V2.1.0, and did my pairing with V2.1.0. It worked fine for the pairing and both are regonized in zigbee2mqtt.

I was able to update the firmware of the good nodon, the buggy one is giving me some error messages:
`image CRC-32 is invalid`
I updated it by using the Nodon App and it looks to be ok now.

If I try to switch again to V2.2.1, the device list is again empty.

### What did you expect to happen?

No issue to add new Nodon SIN-4-2-20 since it worked on elder version.

### How to reproduce it (minimal and precise)

I expect with some Nodon SIN-4-2-20 pairing with V2.2.1

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20221226

### Adapter

Sonoff CC2652

### Setup

LXC on Proxmox 8.3

Tonio16, 2025-04-08:
Ok, I will test in one month so. I don't know how to change of version for my zigbee2mqtt LXC. So to change of version, no chance.

Antoine

BenDavidson90, 2025-04-08:
> Some Nodon firmware are apparently not following the versioning from others, it caused some issues with parsing said versions. It should be fixed in latest-dev/edge (make sure to update to latest). [Koenkk/zigbee-herdsman-converters#9130](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9130)
> 
> CC: [@BenDavidson90](https://github.com/BenDavidson90)

Updated to latest dev, it works now for me ! Thanks !

closed 2025-04-18 by Nerivec