Koenkk/zigbee2mqtt #27049: [Feature request]: Get timeout message when zigbee device is ooo

smuteau, 2025-04-08:
### Is your feature request related to a problem? Please describe

I have a SMLIGHT SLZB-06M powered with POE.
Sometimes when it's offline, I don't see any information about that.
I can see that all,my devices in Z2M are not receiving updates since minutes and most of time in hours.

### Describe the solution you'd like

That could be very nice to know if the zigbee controller (internal or external) is down.
Z2M could give/send information about the status of the zigbee controller through mail, telegram, mqtt status sending to GladysAssistant, HA, Jeedom or others, to get a warning.
Then we could be able to check very quickly why controller is down.

In my mind, if Z2M checks online devices and sees that ALL devices are not sending updates since a certain time (time parameter in Z2M ? 5mn timeout for all, or more ?), then something is going wrong and message is sent to mqtt broker.

### Describe alternatives you've considered

Maybe that could be a ping timeout for external zigbee controller, but it has to be considered for all kind of devices (attached like usb or external).

### Additional context

-

Nerivec, 2025-04-08:
Z2M shuts down when the coordinator becomes unavailable (since it breaks everything). So you should get the `offline` [publish over MQTT](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-state).

Note: If you enable the Z2M watchdog, it will try to restart at specific intervals instead of just shutting down: https://www.zigbee2mqtt.io/guide/installation/15_watchdog.html

smuteau, 2025-04-09:
Thx @Nerivec for this info, I saw that Z2M docker was trying to restart and restart but didn't know about the MQTT offline status. 
I will have a look but I know that Gladys assistant is not getting string value yet, just numerical value. 

I guess we can close this issue then.

closed 2025-04-09 by smuteau