Koenkk/zigbee2mqtt #27051: Sonoff Zbmini R2 - Event types "on"

raffoul, 2025-04-08:
### What happened?

For every zbminir2 devices, the following event type is sent periodically to MQTT : 
{"event_type":"on"}

The z2m documentation about this device explains only toggle event is supported.

Please have a look at this issue : 
https://github.com/home-assistant/core/issues/141309

### What did you expect to happen?

Send only toggle events to MQTT

### How to reproduce it (minimal and precise)

Enable experimental events
Add sonoff zbminir2 device
Listen Mqtt events

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

0x26780700

### Adapter

Combee2

### Setup

HA ADDON / HomeAssistant yellow

Nerivec, 2025-04-18:
@mundschenk-at can you take a look at this?

mundschenk-at, 2025-04-18:
The converter only exposes `toggle` for the `action` property (via `modernExtend.commandsOnOff`).

raffoul, 2025-08-20:
Reopen

raffoul, 2025-08-24:
Hi,
I’m experiencing this issue only with the Sonoff Zbmini R2: Zigbee2MQTT is sending periodic “on” events to MQTT, even when there’s no user interaction or physical switch press. Home Assistant throws errors because it doesn’t support this specific event type sent by Z2M.
According to the Zigbee2MQTT documentation, only “toggle” events should be sent for this device, so the periodic “on” event seems unexpected and I think this is the reason why HA throw an error.

Debug logs :
```
[2025-08-24 09:59:23] debug: 	z2m: Received Zigbee message from 'Switch couloir', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
[2025-08-24 09:59:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Switch couloir', payload '{"action":"on","delayed_power_on_state":false,"delayed_power_on_time":33.5,"detach_relay_mode":true,"external_trigger_mode":"edge","last_seen":"2025-08-24T07:59:23.281Z","linkquality":255,"network_indicator":true,"power_on_behavior":"on","state":"ON","turbo_mode":false,"update":{"installed_version":4100,"latest_version":4100,"state":"idle"}}'
[2025-08-24 09:59:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Switch couloir/action', payload 'on'
```

Home assistant logs : 
```
2025-08-24 09:59:23.287 WARNING (MainThread) [homeassistant.components.mqtt.event] Invalid event type on for event.switch_couloir_action received on topic zigbee2mqtt/Switch couloir, payload {"event_type":"on"}
```

This happens for all my Zbmini R2 devices, always after the config payload.
I agree that only “toggle” events should be sent. Could this be related to how experimental events are handled?

Let me know if you need more logs or details from my setup and I hope this is clear.

Thanks!

Koenkk, 2025-08-24:
Do you have the [state_action](https://www.zigbee2mqtt.io/devices/ZBMINIR2.html#options) option enabled? If yes, can you try disabling it?

raffoul, 2025-08-24:
Yes. It seems to be the root cause. I will close the issue tomorrow if everything works.
Thanks a lot.

closed 2025-08-25 by raffoul