Koenkk/zigbee2mqtt #27052: Impossible to pair new devices with SLZB-06M

surfingbytes, 2025-04-08:
### What happened?

I have SLZB-06M in Zigbee Coordinator mode, connected via UTP cable. I replaced my older Sonoff ZBDongle-E with it, as it supposedly handles more devices, buut mainly, I can put it anywhere where I have UTP, not just at the server location (USB). I cloned the IEEE address so I didn't have to re-pair all devices and everythin worked. Everything, but... Whenever I try to pair new device, the device is discovered, paired, configured, and then "device left network". And in a second, all over again - discovered, paired, configured, left.... So basically it is impossible to add any new device to my zigbee network.

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

SLZB-06M

### Setup

Docker containerm zigbee managed by eclipse-mosquitto (also docker)

Nerivec, 2025-04-08:
Have you tried pairing to specific routers or to the coordinator instead of "Permit join (all)"? If yes, does it show the same issue?
Have you tried pairing another device (not same brand/model)?

Note: the SLZB06M has exactly the same chip as the Dongle-E, the only difference is the connection mode. For better chips, you want the `MG24` variants. https://www.zigbee2mqtt.io/guide/adapters/emberznet.html

surfingbytes, 2025-04-08:
At the time I was checking, MG24 seemed more experimental, but maybe you're right, and I can upgrade again. Anyway, I never noticed I can permit join only on some device. Not sure if it's any different to Permit join (all), when there is just 1 coordinator, but Itried it. The result is th same. Log is attached, and you can see there is: accepting join, interview, joined, successfully interviewed, device supported, device left network. This log is for Tuya plug, but the same issue happens with all other devices, and also, the same plug (same brand ant type, not the exact piece) was paired just fine with previous coordinator device. Not too sure why there is 2x "interview started" in logs though - once for 18zh and once for 18z2m. Same with "Succesfully interviewed", also 2x.

Nerivec, 2025-04-08:
Can you set the `log_level` to `debug` so we can get a better picture of what's going on when you pair a device?

surfingbytes, 2025-04-08:
It was set to debug - in UI. But I think I saw somewhere that setting it in UI is not enough?

[image]

surfingbytes, 2025-04-09:
I've got the debug log - [log.log](https://github.com/user-attachments/files/19672468/log.log). The device I'm trying to pair is 0xa4c138cea2454620

Nerivec, 2025-04-09:
Can you describe the steps you took to migrate coordinator?

surfingbytes, 2025-04-09:
I plugged the SLZB via USB to my laptop, configured it as zigbee coordinator and updated firmware. Then I changed settings to connect via ethernet, and connected it via UTP. After that, turned off Z2M, cloned the IEEE of my old coordinator, took the old coordinator out from server and changed Z2M settings to use coordinator over tcp. When it was ready, I turned off the server, trned off main breaker in apartment, waited 2 mins, turned the breaker back on, turned server on, and everything worked. Except pairing new devices, as I found out now...

Nerivec, 2025-04-09:
The device paired fine, the key is verified, so it has no reason to leave. Normally, I'd say the device is having troubles, but you mentioned it happened with other models/brands...
Could you try [backing up the NVM3 tokens](https://github.com/Nerivec/ember-zli/wiki/Stack#backup-nvm3-tokens) of both adapters and sending both files to me (here, in private repo, on Discord in DM, whatever you prefer)? _Hopefully you haven't yet reset the old one._

surfingbytes, 2025-04-10:
Yes, I tried pairing Eglo Crosslink.2 bulb, couple of Tuya smart plugs, and now light sensor and it's all the same. I have troubles running ember-zli. For some reason it won't install on server (npm ERR! request to https://registry.npmjs.org/ember-zli failed, reason: connect ENETUNREACH 2606:4700::6810:1d22:443), and if I install it on my laptop under WSL, it wont find the adapter - neither the UTP one, nor the old USB..

surfingbytes, 2025-04-10:
If there is nothing else we can do, I'll order the MG24, clone the IEEE into it again and try with that. Will update this when it arrives.

surfingbytes, 2025-04-29:
As much as I'd like to help with the traffic sniffing, that guide seems abit too advanced for me. I linked the issue because of this [comment](https://github.com/Koenkk/zigbee2mqtt/issues/26948#issuecomment-2790651666), which also mentions SLZB06, and all versions.
Anyway, I've got the MG24, which came on version v2.8.1.dev9, and pairing works. I didn't even clone the IEEE, just followed the steps when replacing adapters, all zigbee devices work without needing to re-pairing them, and pairing new devices is also fine. So I'm disabling internet access to this adapter and it will never see any firmware update, as I'm not sure if it's caused by that.

phillipo6, 2025-05-11:
I have the exact same issue! I figured out that if I unplug the device that got paired, it stops disconnecting and stays connected. So it appears that the devices do not leave pairing mode and keep disconnecting and then connect again

Ankolana, 2025-06-04:
Hello
I have also exact the same issue!

I have the SLZB-06Mg24. I also flashed the IEEE address from the Sonoff Dongle-E in to the SLZB-06Mg24, and the network worked without any problems from the beginning. I also connected the coordinator via TCP:
port: tcp\://192.168.2.156:6638
baudrate: 115200
adapter: ember

When I try to pair a new device, it connects to Zigbee2MQTT, and I can control it for about 10 seconds. Then the following happens: *device XXXXXXX left the network*. I also think that the device remains in pairing mode and keeps resetting itself.

But when I try to pair a 3-year-old IKEA Zigbee smart bulb, it works without any problems!
It seems to me that the issues mainly occur with newer devices.
I experienced problems with the Gledopto GL-C-201P and GL-C-310P.

thogens, 2025-06-19:
FYI: Re-enabling sonoff stick with same IEEE resulted in proper pairing and the device remained functional after going back to SLZB-06!

bistvan12, 2025-07-20:
I have a same issue with SLZB-06: newest firmware. I went back to a SONOFF ZigBee 3.0 Dongle. Same problem. Of course "permit_join: true".

bistvan12, 2025-07-25:
> [@bistvan12](https://github.com/bistvan12) any update on your side?

> _I got this from another forum, though I haven't tried it yet:_  
> 
> “After I changed the coordinator (**ZBDongle-P → ZBDongle-E**) without having to reinstall the Zigbee network  
> (thanks to the excellent instructions from **Simon42**),  
> my entire Zigbee network is running perfectly again,  
> and I can also learn new devices again.”

thogens, 2025-08-05:
… curious to know what happens when you want to pair a new battery powered device from TUYA…

Hupeleto, 2025-08-16:
Hello,

Exactly the same issue: pairing, configuring, device continuosly blinking, the device XXX left the network...
Coordinator: SLZB-06M - ethernet - Z2M
Device: Aqara MCCGQ14LM, Tuya TS0207_water_leak_detector. Tuya ZG-2002-RF

Pairing without problem: Sonoff ZBMINIR2.

Thanks for any help

JaimePalazuelosB, 2025-08-16:
Hi,

I have been trying and I have managed to join one problematic device. What I have done:

  *
Permit join on the Coordinator
  *
Pair the device. The moment it showed up on Z2M, I disabled the join. All of a sudden, the device stopped the pairing and started to expose data.

Can somebody with the same problem to try it?

Cheers.

Jaime

knobblysnail, 2025-08-17:
Pulling out a battery from an Ikea Styrbar remote immediately after pairing worked for me, but I understant that is no use for any device with a built in battery.

JaimePalazuelosB, 2025-08-17:
I tried that, and it is true that it was not being deleted, but then the device was not sending the actions or modifying the state, so, it was there, but not actually working.

Hupeleto, 2025-08-17:
I tried these tricks, remove the battery or disable pairing as soon as the device is recognized. With the same results: device present but without data or incomplete data.
I finally found the root of the problem: the 15-year-old Cat 5 Ethernet cable that was slowing down data transmission between the SLZB-06M and Z2M.
**A new Cat 6 cable and everything is back to normal.**
I hope that my experience can be used by others

Hupeleto, 2025-08-17:
Maybe a new faulty cable?
Maybe a saturated switch?
Maybe an overheating router?
Maybe the weak POE injector?
Test everything between Z2M and the coordinator

AlexMKX, 2025-08-17:
Seems have the same with 2652P

https://github.com/Koenkk/zigbee2mqtt/issues/28282#issuecomment-3194496676
For me I don't believe anything like overheating router/weak POE/faulty cable is a case. The 150 devices network is steady apart from joining. 

[image]

thogens, 2025-09-07:
I had similar issues and at the end I realized that I had too many devices paired. (200). If you have about tue same, your network will become unstable… Note: after I migrated a lot of devices to a new stick, I was able to pair again without issues!
How many devices do you have?

Yves911, 2025-09-09:
> I had similar issues and at the end I realized that I had too many devices paired. (200). If you have about the same, your network will become unstable… Note: after I migrated a lot of devices to a new stick, I was able to pair again without issues! How many devices do you have?

"Only" 60 devices here, and i have no issue to pair devices with the old sonoff usb key.

Karol-007, 2025-09-13:
I have same Problem with SLZB-06MG24 after pairing few seconds and it is leaving zigbee network :-(

Karol-007, 2025-09-14:
Today I tried to pair another new device OXT presence Radar TS0601 and it is the same situation is paired then reviewed for few seconds it is okay but device itself is not closig pair mode and after few seconds is not anymore in network then is again paired and succesfully interviewed and again it is few seconds in network I hope we will get some new firmware update :-(

Yves911, 2025-09-14:
> Have you tried this on the configuration?
> 
> Enable Zigbee Socket packet processing Allow multi-threaded socket connection

I did it and there was no change on the pairing process

Tarik2142, 2025-10-01:
Well, it seems that replacing IEEE causes some conflict in Z2M or zigbee network but it's not clear what.

ninux51, 2025-10-03:
Go to SLZB-06M web interface
Go to menu Z2M and ZHA scroll down to Flash custom IEEE address and put 0xFFFFFFFFFFFFFFFF click on Write IEEE address
0xFFFFFFFFFFFFFFFF will restore original IEEE address of SLZB-06M.
After write SLZB-06M will reboot.
Restart Z2M.
For information I only need to repair 2 devices on 42 devices.

Yves911, 2025-10-03:
@ninux51 thanks for the info, i did exactly what you said and now everything is OK, i can pair new devices without any issue. I also had to repair 2 devices and nothing more.
Thanks for having sharing this solution !

thogens, 2025-10-03:
Could it be that the 2 devices were the culprit? I’m asking because in my case the deletion of ~100 devices was also sufficient to make the stick working again… I did not change the IEEE! Maybe in my case there were also some „faulty“ devices, doing something with the network?

Other question: which IEEE are you using now? The original IEEE from SLZB or the „old Sonoff“?

thogens, 2025-10-03:
And in this case you did not need to re-pair all devices? I can’t believe that, as all devices must have stored the old sonoff IEEE as coordinator.
I am confused.

ninux51, 2025-10-03:
I didn't re-pair all my device, on 42 devices only 2 was ko and need a paring.

Quenbo, 2025-12-07:
Hello, 
I had exactly the same symptoms after switching from a Sonoff Dongle E to a SLZB-06M on which I copied the IEEE adress of the previous Sonoff Dongle E.

After resetting the IEEE using 0xFFFFFFFFFFFFFFFF on my SLZB-06M, I did not had any problem adding new devices that "left the network" instantly before.
I was afraid of needing to repair all my devices but only 3 or 4 temperature sensors were lost.

But, now, I'm getting tons of "Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0"." in the Z2M logs.
This happens always at the same second that a device send a MQTT publish.
But HA seems to receive their datas...

Did you notice the same behaviour? Any idea on how to solve it?

Quenbo, 2025-12-07:
> Yes I also have this issue, you need to find witch router send this message and repairing it. For find it power off a maximum zigbee router to see if message stop and find the routeur who always have the old IEEE address by deduction. In my case it was a the Lixee router. Since I never add this message and my Zigbee network works very fine.

Unfortunately I see your message now but my problem is resolved.
I turned OFF the maximum of devices I could but I did not notice any improvement.
Then it appeared that my Lixee was also reporting error and indeed it was the guity device, like you.

The problem is this Lixee is a damn shit to pair again and to reconfigure.
For now I don't have any update of my total electric consumption (but that is a different problem from this topic)
I'm going to wait for it to stabilize for one hour before digging...

ninux51, 2025-12-07:
For resolve the issue in Lixee I juste delete it on Z2M and open pairing, Lixee automaticaly paire with no action on it.
Yes Lixee v1 buton is a shame ...

wolfgangkt1, 2026-03-26:
Hallo,
auch ich habe das Problem das sich neue Geräte nicht koppeln lassen. Meistens sind es Geräte die eine Routerfunktion haben.
Habe alles probiert aber keinen Erfolg. Wirklich frustrierend.

blair287, 2026-04-25:
I am using ZHA but i had migrated from a conbee 2 to a mr10w and also tried a o6m both had the same behaviour all migrated over bar 1 ikea blind but completely unable to pair anything new or add back the ikea blind i tried everything powered off everything all router for hour but no difference. I logged my issue here but no one had any info.

https://github.com/home-assistant/core/issues/150734

im going to guess as you are all describing the exact same behaviour as i have but on Z2M there is a bug or defect in smlights products and have boxed them back up and gone back to my old hardware which is a pain as i really need a poe zigbee and z wave solution that actually works.

Yves911, 2026-09-25:
### Issue Summary & Resolution: Pairing Failures on SLZB-06M with Custom IEEE Address

**Problem:**
When pairing new Zigbee end-devices, they initially connect, then immediately drop off and remain stuck in pairing mode. Temporarily changing the adapter's IEEE address, pairing the device, and reverting back to the custom IEEE address worked as a workaround, but was required every time.

**Root Cause:**
When migrating from an older coordinator to the SLZB-06M using a **Custom IEEE address**, leftover network states or table corruptions in the Silicon Labs EFR32MG21 **NVM (NVRAM) memory** prevent new security key handshakes from completing properly under that custom address.

---

### Solution: Clear Radio NVM & Reflash Firmware

Follow these steps to clean the NVM memory and restore stable pairing under your custom IEEE address:

1. **Stop Zigbee2MQTT:**  
   Stop the Zigbee2MQTT add-on/container in Home Assistant to close all active serial connections to the coordinator.

2. **Flash EFR32MG21 & Clear NVM:**  
   - Open the SLZB-06M Web UI (`http://<your-adapter-ip>`).
   - Go to **Firmware Update** → **EFR32MG21 over-the-air (OTA) update** and click **Check for updates** (or *Flash custom firmware*).
   - In the pop-up modal, **toggle ON** the `Clear NVM memory` option.
   - Click **Next** to flash the firmware. This purges the corrupted radio NVM table.

3. **Re-apply Custom IEEE Address:**  
   - Once the flash finishes, reboot the SLZB-06M via **System** → **Reboot**.
   - Re-enter your **Custom IEEE address** (from your previous coordinator) in the dedicated SMLIGHT settings page and save.

4. **Restart Zigbee2MQTT:**  
   - Start the Zigbee2MQTT add-on.  
   - Z2M will automatically restore the network backup into the freshly cleared NVM memory.

After completing this procedure, new Zigbee devices will pair smoothly without requiring temporary IEEE address swaps.

tunatoksoz, 2026-09-29:
I am having a similar issue with MRW10U. It doesn't have clearing nvm memory as an option however.

Also noticed something:
1. Read factory IEEE address - always return an address. ([CC2674P10] Factory IEEE address of the radio module)
2. Read current IEEE address - sometimes return an address. Other times it's "unknown" ([CC2674P10] Read current IEEE).

I decided to reapply factory IEEE as the current IEEE assuming it'd fix my problems, but it didn't quite work.

BooWseR, 2026-10-15:
If your flasher or web UI doesn't offer a "Clear NVM / Erase NVRAM" option after cloning a coordinator IEEE address (which leaves the internal Trust Center EUI64 out of sync and causes new Zigbee rev. 21 devices to drop after 20 seconds due to `VERIFY_KEY_TIMEOUT`), you can force the `ember` driver in Z2M to format and cleanly restore the stick's NVM3 storage using the **PAN-ID +1 trick**:

1. **Stop Zigbee2MQTT.**
2. **Increment PAN-ID by +1:**
   * In `configuration.yaml`, add/change `pan_id` under `advanced:` to your current decimal PAN-ID + 1 (e.g., if hex is `0x1a62` / dec `6754`, set it to `6755`).
   * In `coordinator_backup.json`, increment the hex `"pan_id"` by 1 as well (e.g., `"1a62"` -> `"1a63"`).
3. **Start Zigbee2MQTT:** Because the PAN-ID no longer matches the stick's NVRAM, the `ember` adapter is forced to wipe the NVM3 storage and write the backup from scratch.
4. **Stop Zigbee2MQTT again** once it has fully booted.
5. **Revert to original PAN-ID:** Change the `pan_id` in both `configuration.yaml` and `coordinator_backup.json` back to their original values (`6754` / `"1a62"`).
6. **Start Zigbee2MQTT:** It will perform one more clean restore with your original network parameters and a properly synced Trust Center EUI64.

After doing this, my existing network worked right away and new Tuya devices pair and stay connected without dropping after 20 seconds. *(Troubleshooting credit goes to Gemini Argon for pointing out the NVM3 mismatch and this workaround!)*