Koenkk/zigbee2mqtt #27054: S60ZBTPF

MiAutomations, 2025-04-09:
### What happened?

Hello This device S60ZBTPF [https://www.zigbee2mqtt.io/devices/S60ZBTPF.html](url)

That is in the list of Z2M as a supported device is not reporting the data related to the energy 

Can you please check? 

Thank you 

### What did you expect to happen?

[image]

### How to reproduce it (minimal and precise)

I paired with success but the device doesn't reporte the energy data 

### Zigbee2MQTT version

 2.2.1

### Adapter firmware version

20240710

### Adapter

SLZB

### Setup

Add-on home assistant running on a nuc

### Debug log

N/A

mgerald21, 2025-04-11:
As described in https://www.zigbee2mqtt.io/devices/S60ZBTPF.html the plug only exposes switch (state), power_on_behavior but no energy related data.

Nothing wrong with Z2M

Koenkk, 2025-04-12:
Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

MiAutomations, 2025-04-14:
> Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

`import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['S60ZBTPF'],
    model: 'S60ZBTPF',
    vendor: 'SONOFF',
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false}), m.electricityMeter()],
    meta: {},
};
`

ccorderor, 2025-04-23:
> As described in https://www.zigbee2mqtt.io/devices/S60ZBTPF.html the plug only exposes switch (state), power_on_behavior but no energy related data.
> 
> Nothing wrong with Z2M

The device reports energy related data using ZHA, so i think the energymeter cluster has not been added in the converter

Sverto, 2025-05-20:
I'm not familiar with the inner workings of Zigbee, but doing this seems to report the correct voltage atleast (231V):

Device -> Dev Console -> generated_external_definition
Then below:
Endpoint 1 -> Cluster haElectricalMeasurement -> Attribute rmsVoltage -> Read
->zhc:tz: Read result of 'haElectricalMeasurement': {"rmsVoltage":23115}

I haven't looked into it further (yet), but hopefuly this can already help.

Koenkk, 2025-05-21:
Support for electricity measurements has been added in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-21 by Koenkk

vturekhanov, 2025-07-08:
It seems that energy-related data is only updated when the switch is turned on. This might be a device firmware-related issue.

Petika86, 2025-07-19:
> It seems that energy-related data is only updated when the switch is turned on. This might be a device firmware-related issue.

Me too,
When i turned off the state, the Current, Voltage, and the Power dont refresh, and the device show the last values

I think it is a issue

I use: S60ZBTPF,  FW date: 20250411 FW version: 1.0.2
Zigmee2MQTT version: 2.5.1

lluisd, 2025-08-25:
I have also the same problem. After socket is turned off I still see the last value in Power, Voltage and current. Values are not updated anymore.