Koenkk/zigbee2mqtt #27055: [New device support]: SNZB-02LD

MiAutomations, 2025-04-09:
### Link

Not available yet

### Database entry

{"id":224,"type":"EndDevice","ieeeAddr":"0xa4c13800dec8ffff","nwkAddr":4436,"manufId":4742,"manufName":"SONOFF","powerSource":"Battery","modelId":"SNZB-02LD","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,32,64599,64529],"outClusterList":[25],"clusters":{"64529":{"attributes":{"7":0}},"genBasic":{"attributes":{"modelId":"SNZB-02LD","manufacturerName":"SONOFF","powerSource":3,"zclVersion":3,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"20250221","swBuildId":"1.0.7"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2030}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"genPollCtrl":{"attributes":{"checkinInterval":6960}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d129fc","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d129fc","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0026d129fc","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"20250221","swBuildId":"1.0.7","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1744191419960,"checkinInterval":1740}

### Zigbee2MQTT version

 2.2.1

### Comments

I try to pair the device but the device appears as unsupported 

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['SNZB-02LD'],
    model: 'SNZB-02LD',
    vendor: 'SONOFF',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.temperature()],
    meta: {},
};
```

### What does/doesn't work with the external definition?

All the things are working correctly, It just to be appear as a supported device

Koenkk, 2025-04-10:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/sonoff.ts)?

chris-1243, 2025-04-10:
Does this device as a screen like the SNZB-02D?

If yes, would you mind try this converter before submitting a PR? I guess this new device might be quite similar to the SNZB-02D in terms of functionalities.

Create an *.mjs file in `data/external_converters` folder and copy/paste the code below

```javascript
import {Zcl} from "zigbee-herdsman";
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
    zigbeeModel: ["SNZB-02LD"],
    model: "SNZB-02LD",
    vendor: "SONOFF",
    description: "Temperature sensor with screen",
    //ota: true,
    extend: [
        m.deviceAddCustomCluster("customSonoffSnzb02ld", {
            ID: 0xfc11,
            attributes: {
                comfortTemperatureMax: {ID: 0x0003, type: Zcl.DataType.INT16}, // here is the problem and withe the 3 lines below....
                comfortTemperatureMin: {ID: 0x0004, type: Zcl.DataType.INT16},
                temperatureUnits: {ID: 0x0007, type: Zcl.DataType.UINT16},
                temperatureCalibration: {ID: 0x2003, type: Zcl.DataType.INT16},
            },
            commands: {},
            commandsResponse: {},
        }),
        m.battery(),
        m.temperature(),
        m.bindCluster({cluster: "genPollCtrl", clusterType: "input"}),
        m.numeric({
            name: "comfort_temperature_min",
            cluster: "customSonoffSnzb02ld",
            attribute: "comfortTemperatureMin",
            description:
                "Minimum temperature that is considered comfortable. The device will display ❄️ when the temperature is lower than this value. Note: wake up the device by pressing the button on the back before changing this value.",
            valueMin: -10,
            valueMax: 60,
            scale: 100,
            valueStep: 0.1,
            unit: "°C",
        }),
        m.numeric({
            name: "comfort_temperature_max",
            cluster: "customSonoffSnzb02ld",
            attribute: "comfortTemperatureMax",
            description:
                "Maximum temperature that is considered comfortable. The device will display 🔥 when the temperature is higher than this value. Note: wake up the device by pressing the button on the back before changing this value.",
            valueMin: -10,
            valueMax: 60,
            scale: 100,
            valueStep: 0.1,
            unit: "°C",
        }),
        m.enumLookup({
            name: "temperature_units",
            lookup: {Celsius: 0, Fahrenheit: 1},
            cluster: "customSonoffSnzb02ld",
            attribute: "temperatureUnits",
            description:
                "The unit of the temperature displayed on the device screen. Note: wake up the device by pressing the button on the back before changing this value.",
        }),
        m.numeric({
            name: "temperature_calibration",
            cluster: "customSonoffSnzb02ld",
            attribute: "temperatureCalibration",
            description: "Offset to add/subtract to the reported temperature",
            valueMin: -50,
            valueMax: 50,
            scale: 100,
            valueStep: 0.1,
            unit: "°C",
        }),
    ],
};
```

MiAutomations, 2025-04-11:
> Does this device as a screen like the SNZB-02D?
> 
> If yes, would you mind try this converter before submitting a PR? I guess this new device might be quite similar to the SNZB-02D in terms of functionalities.
> 
> Create an *.mjs file in `data/external_converters` folder and copy/paste the code below
> 
> import {Zcl} from "zigbee-herdsman";
> import * as m from "zigbee-herdsman-converters/lib/modernExtend";
> 
> export default {
>     zigbeeModel: ["SNZB-02LD"],
>     model: "SNZB-02LD",
>     vendor: "SONOFF",
>     description: "Temperature sensor with screen",
>     //ota: true,
>     extend: [
>         m.deviceAddCustomCluster("customSonoffSnzb02ld", {
>             ID: 0xfc11,
>             attributes: {
>                 comfortTemperatureMax: {ID: 0x0003, type: Zcl.DataType.INT16}, // here is the problem and withe the 3 lines below....
>                 comfortTemperatureMin: {ID: 0x0004, type: Zcl.DataType.INT16},
>                 temperatureUnits: {ID: 0x0007, type: Zcl.DataType.UINT16},
>                 temperatureCalibration: {ID: 0x2003, type: Zcl.DataType.INT16},
>             },
>             commands: {},
>             commandsResponse: {},
>         }),
>         m.battery(),
>         m.temperature(),
>         m.bindCluster({cluster: "genPollCtrl", clusterType: "input"}),
>         m.numeric({
>             name: "comfort_temperature_min",
>             cluster: "customSonoffSnzb02ld",
>             attribute: "comfortTemperatureMin",
>             description:
>                 "Minimum temperature that is considered comfortable. The device will display ❄️ when the temperature is lower than this value. Note: wake up the device by pressing the button on the back before changing this value.",
>             valueMin: -10,
>             valueMax: 60,
>             scale: 100,
>             valueStep: 0.1,
>             unit: "°C",
>         }),
>         m.numeric({
>             name: "comfort_temperature_max",
>             cluster: "customSonoffSnzb02ld",
>             attribute: "comfortTemperatureMax",
>             description:
>                 "Maximum temperature that is considered comfortable. The device will display 🔥 when the temperature is higher than this value. Note: wake up the device by pressing the button on the back before changing this value.",
>             valueMin: -10,
>             valueMax: 60,
>             scale: 100,
>             valueStep: 0.1,
>             unit: "°C",
>         }),
>         m.enumLookup({
>             name: "temperature_units",
>             lookup: {Celsius: 0, Fahrenheit: 1},
>             cluster: "customSonoffSnzb02ld",
>             attribute: "temperatureUnits",
>             description:
>                 "The unit of the temperature displayed on the device screen. Note: wake up the device by pressing the button on the back before changing this value.",
>         }),
>         m.numeric({
>             name: "temperature_calibration",
>             cluster: "customSonoffSnzb02ld",
>             attribute: "temperatureCalibration",
>             description: "Offset to add/subtract to the reported temperature",
>             valueMin: -50,
>             valueMax: 50,
>             scale: 100,
>             valueStep: 0.1,
>             unit: "°C",
>         }),
>     ],
> };

@chris-1243 Yes the device have a screen 

I followed your instructions and I Successfully have the device integrated into Z2M. Tks for you support !!Cheers

chris-1243, 2025-04-11:
Are all the "extra functions" working ?

Wake up the device and then apply a parameter. Do this for all settings available in the expose tab.
I have the hope Sonoff did a copy/paste of the SNZB-02D without the humidity measurement

chris-1243, 2025-04-13:
@MiAutomations Would you mind tell us if the converter I provided is fully working (all the basic + extra functions).

I may then add this device to zigbee-herdsmann-converter but I need a confirmation

MiAutomations, 2025-04-14:
> I may then add this device to zigbee-herdsmann-converter but I need a confirmation

Hello and tks you for your support and work ! 

At my side all are working well with this device and with this external_converter provided

Best regards

chris-1243, 2025-04-14:
@MiAutomations 

Could you submit a picture for the docs? This can be done by:

Create a fork by clicking here
Go to the public/images/devices directory, Add file -> Upload files
Upload the files and press Commit changes
Press Contribute -> Open pull request -> update title/description -> Create pull request
Make sure that:

The filename is SNZB-02LD.png
The size is 512x512
The background is transparent (use e.g. Adobe remove background)

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-add-device-picture-to-zigbee2mqtt-io-documentation