Koenkk/zigbee2mqtt #27058: ZY-M100-S_2 - since update of z2mqtt 2.2.1 - illuminance is steady at 126lx and does not change over days

rcmcronny, 2025-04-09:
### What happened?

Since the update of z2mqtt to 2.2.1 the  ZY-M100-S_2  illuminance value does not change anymore, it reports 126lx, i only catched this, as my automation did not work anymore.
The same value is also reported in the z2mqtt status view. 
SLZB-06 with 20240710 firmware in lan /poe modus.

### What did you expect to happen?

correct value for light level :)

### How to reproduce it (minimal and precise)

always there:

Illuminance
Measured illuminance
126lx

### Zigbee2MQTT version

2.2.1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Addon on Homeassistant on a libvirt VM

### Debug log

info 2025-04-09 15:41:54z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1388438f3be7c', payload '{"detection_delay":0.1,"fading_time":3,"illuminance":126,"last_seen":"2025-04-09T15:41:54+02:00","linkquality":91,"maximum_range":6.5,"minimum_range":0,"presence":false,"radar_sensitivity":5,"target_distance":0}'
info 2025-04-09 15:41:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1388438f3be7c', payload '{"detection_delay":0.1,"fading_time":3,"illuminance":126,"last_seen":"2025-04-09T15:41:56+02:00","linkquality":83,"maximum_range":6.5,"minimum_range":0,"presence":false,"radar_sensitivity":5,"target_distance":0}'
info 2025-04-09 15:41:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rauchmelder_Flur_OG1', payload '{"alarm":null,"battery":100,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b00258fe1d9","last_seen":"2025-04-09T15:41:57+02:00","linkquality":102,"max_duration":

Koenkk, 2025-04-10:
This should be fixed in the dev branch already.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

rcmcronny, 2025-05-02:
New version fixes this, many thanks

closed 2025-05-02 by rcmcronny