Koenkk/zigbee2mqtt #27059: No device list or network map after HA backup.

Kriss1670, 2025-04-09:
### What happened?

After a hardware crahs, I got some troubles to backup HA. I managed to make everything work but Z2M page doesn't anymore show the list of devices or the network page even if all devices seem to work properly (not tested one by one).

HA Core 2025.4.1
HA Supervisor 2025.03.4
HA Operating System 15.1
Z2M 2.2.1-1

Here is a boot log :

[2025-04-09 16:30:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-09 16:30:49] info: 	z2m: Starting Zigbee2MQTT version 2.2.1 (commit #unknown)
[2025-04-09 16:30:49] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-09 16:30:49] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20231008171311","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231008171311-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-04-09 16:30:49] info: 	zh:ember: Using default stack config.
[2025-04-09 16:30:49] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-04-09 16:30:49] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-04-09 16:30:49] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-04-09 16:30:49] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-04-09 16:30:49] info: 	zh:ember:uart:ash: Serial port opened
[2025-04-09 16:30:49] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-04-09 16:30:50] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-04-09 16:30:50] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-04-09 16:30:50] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-04-09 16:30:50] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-04-09 16:30:50] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.3 [GA]","build":0,"major":7,"minor":4,"patch":3,"special":0,"type":170}
[2025-04-09 16:30:50] info: 	zh:ember: [STACK STATUS] Network up.
[2025-04-09 16:30:50] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-04-09 16:30:50] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2025-04-09 16:30:51] info: 	z2m: zigbee-herdsman started (resumed)
[2025-04-09 16:30:51] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":3,"revision":"7.4.3 [GA]","special":0,"type":170},"type":"EmberZNet"}'

followed by a list of devices found, then...

[2025-04-09 16:30:51] info: 	z2m: Currently 46 devices are joined.
[2025-04-09 16:30:51] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-04-09 16:30:51] info: 	z2m: Connected to MQTT server
[2025-04-09 16:30:51] info: 	z2m:mqtt: MQTT publish: topic 'Z2M/bridge/state', payload '{"state":"online"}'
[2025-04-09 16:30:51] error: 	z2m: Failed to start 'Bridge' (TypeError: Invalid version. Must be a string. Got type "undefined".
    at new SemVer (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/classes/semver.js:19:13)
    at compare (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/functions/compare.js:3:3)
    at Object.gt (/app/node_modules/.pnpm/semver@7.7.1/node_modules/semver/functions/gt.js:2:29)
    at result.exposes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/devices/nodon.ts:112:38)
    at Object.exposes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/index.ts:389:40)
    at Device.exposes (/app/lib/model/device.ts:53:45)
    at Bridge.getDefinitionPayload (/app/lib/extension/bridge.ts:826:29)
    at Bridge.publishDevices (/app/lib/extension/bridge.ts:756:34)
    at Bridge.start (/app/lib/extension/bridge.ts:180:20)
    at Controller.startExtension (/app/lib/controller.ts:268:13))
[2025-04-09 16:30:51] info: 	z2m: Started frontend on port 8099

Under the Z2M device list I discovered 2 bridges, one concerning an old configuration (with a Conbee 2) and the other with the newest one (EmberZNet 7.4.3 [GA]). I specify that everything was working fine with the Ember key before the HA restoration due to hardware crash.

I also find all duplicate entities : Permit join, Restart, Log level, Connection (one is disconnected, the other is connected), Version (Unavailable), Permit join timeout (Unavailable), Coordinator version (both deactivated), Network map (both deactivated) and Restart required (both deactivated).

How can I solve this problem ?

### Zigbee2MQTT version

2.2.1-1

### Adapter firmware version

0442

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Z2M run as a Home Assistant Add-on. HA running under a Proxmox server on a N100 mini PC.

Nerivec, 2025-04-10:
Some nodon devices apparently aren't following others in firmware versioning. Problem isn't related to your backup.
Should already be fixed in dev.

closed 2025-04-10 by Nerivec

sdetweil, 2025-04-10:
I see the same thing.. was away for a month, had power failure, UPS died.. lost configs everywhere.. SO much fun!!!
new HA version, everything lost, 

now back, docker added zigbee2mqtt again v2.1.1, now have no devices.. 

don't get any errors on startup 
```
	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-04-10 08:41:09] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-04-10 08:41:09] info: 	zh:ember: [STACK STATUS] Network up.
[2025-04-10 08:41:09] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-04-10 08:41:09] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1249ms until next broadcast.
[2025-04-10 08:41:10] info: 	z2m: zigbee-herdsman started (resumed)
[2025-04-10 08:41:10] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-04-10 08:41:10] info: 	z2m: Currently 0 devices are joined.
[2025-04-10 08:41:10] info: 	z2m: Connecting to MQTT server at mqtt://mosquitto
[2025-04-10 08:41:10] info: 	z2m: Connected to MQTT server
[2025-04-10 08:41:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-04-10 08:41:10] info: 	z2m: Started frontend on port 8080
[2025-04-10 08:41:10] info: 	z2m: Zigbee2MQTT started!
```

don't see a working solution..

Kriss1670, 2025-04-10:
I really don’t think NodOn device is not the cause. I use them since many months without any problems and didn’t have any issues just before my system hardware crash and the GA restoration. My complete backup was old and some partial backups came to an error. This is why I now have 2 bridges in the MQTT Home Assistant device list, one with the old adapter. It’s not possible to delete its entities (option is grayed). Moreover, all the NodOn devices are listed in the log as successfully connected and work properly (not sure id it is a crucial information but wanted to share it). I know the problems you speak about because I saw this on forums. But there was significant differences also…I say it again : everything was working fine before the restoration (complete + unsuccessful and successful partials). And anyway, I need a solution to remove this duplicate entry.I don’t think it was a good solution to close the entry. Please could you consider another option ?Thank you.

Should already be fixed in dev/edge.

fyrstyk, 2025-04-11:
I have exactly the same issue, but I didn't restore from a backup.

Device list empty, but z2m working normally.

Same error 'z2m: Failed to start 'Bridge' (TypeError: Invalid version. Must be a string. Got type "undefined".'
And 'z2m: Failed to start 'HomeAssistant' (TypeError: Invalid version. Must be a string. Got type "undefined".'

Nerivec, 2025-04-11:
Nodon devices are the cause for `TypeError: Invalid version. Must be a string. Got type "undefined".` which prevents proper startup of Z2M (even though devices are fine, Z2M is not in a proper state, causing weird problems like missing devices). This was due to a [missing check in the converters](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9130).
As mentioned, the problem should be fixed in dev/edge. You can try switching to confirm: https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

@Kriss1670 about your duplicate entries in HA, you should be able to remove the stale entries manually as long as they are no longer in use (it appears HA didn't remove them automatically for some reason, or your backup forced their re-creation...). _This is all on HA's side though._

@sdetweil your issue appears to be different from this. Can you open a new issue and post `debug` logs of the full startup sequence?

Kriss1670, 2025-04-12:
@Nerivec 
As I explained earlier, I sincerely don’t think that NodOn devices are in cause. They were working properly even after the Z2M update and the only change in my system just before getting the issue is a restoration of an old complete backup followed by some partial restoration some of which didn’t end well. Morever, NodOn devices a properly set up in the logs… error appear after with mention of « bridge ».
As said also, some duplicate entities have a grayed « Remove » option. I tried also to delete the duplicate bridge device but without any success.
HA works fine. It is Z2M which is buggy. So I’m not sure they would consider a new issue due to unremovable duplicate entities but will try.

I still have the same question : how to proceed ? Will I loose all my devices if I uninstall and re-install Z2M ? What is the solution ?

Nerivec, 2025-04-12:
Yes, it is Nodon devices creating that error, there is no doubt about that... 
```
devices/nodon.ts:112:38
```
It's not a problem with the devices themselves, it was a problem in the converters, as mentioned/linked. You likely triggered it when you restored your backup with a slightly different set of data stored in the database.

The duplicates would not even be tied to Z2M anymore, just entities that were in the backup and got restored (and somehow are not detected as stale by HA...). The solution for this one, I don't know. I've never dealt with HA backup system much. You'd probably have better chances asking in the HA Discord (or HA community), if someone ever encountered this.
Note that it likely does not affect Z2M since the duplicates aren't relevant to Z2M anymore (only one set should work once you run a Z2M version with the fixed Nodon issue -only in "edge" at the moment-). Thing is, if your backup made a mess in the HA system, you might have a lot more "hidden" problems that you may find later on...

You can uninstall/re-install the HA add-on at will, as long as you don't tick the box to delete the data, it's fine.
That's how you can also switch to dev branch (stop regular add-on, and install "edge" add-on, and just copy any settings from the add-on configuration page on regular to "edge"). I suggest you try that first, which should solve your Z2M not starting properly, then re-assess.

Kriss1670, 2025-04-12:
Thank you for offering a solution. I will try and see if HA Discord may help eventually.
I just disagree on 'it does not affect Z2M' as I can't get to the device properties, to the network map, to the reconfigure option and so on...
Be sure that my idea is to improve the system. I am a poor developer but enough to perfectly understand that it is a real, big, respectfull job.

Will post if I find a solution.
Thanks.

Nerivec, 2025-04-12:
I think you're confusing the two problems regarding which one is preventing Z2M from working:
- The duplicate entities in HA should have no bearing on Z2M not working properly (should strictly be on the HA side).
- The Z2M startup problem is because of the nodon bug that should be fixed in dev branch (_the error is tiny, but it is unfortunately happening at a very crucial moment that really messes with the Z2M startup_).

Try switching to the `edge` add-on first, as I described above, this should fix that Nodon problem (_has been confirmed by other users, [e.g.](https://github.com/Koenkk/zigbee2mqtt/issues/27079#issuecomment-2798761333)_) and Z2M should work properly after that (assuming everything else from your backup worked).
Then, you can see about cleaning up those duplicate entities.

_Note: you can always switch back to "regular" add-on after next release on May 1st._

Kriss1670, 2025-04-12:
I will check that. Thank you for helping.