Koenkk/zigbee2mqtt #27060: Missing illuminance entity for Philips motion sensor since v2.0 (including v2.2)

mdp08, 2025-04-09:
### What happened?

Hi,
since version 2.0 (and also in 2.2.0), the illuminance entity for the Philips motion sensor is missing. In previous versions, this entity was available and worked correctly.

Please see the attached screenshot, which shows the illuminance entity in a version prior to 2.0.

This entity is essential for automations that rely on ambient light levels. Could you please check whether this is a bug or an intentional change?

Thanks a lot for your work!

### What did you expect to happen?

I expected the Philips Hue motion sensor to expose an illuminance entity (light level), just like it did in versions prior to 2.0.
This entity is important for creating automations based on ambient light conditions (e.g., turning on lights only when it’s dark).

### How to reproduce it (minimal and precise)

Set up Zigbee2MQTT with a deCONZ (ConBee) stick.

Pair a Philips Hue motion sensor (e.g., Hue Motion Sensor model SML001).

Check the list of exposed entities in Zigbee2MQTT (via MQTT or frontend).

Notice that the illuminance (light level) entity is missing since version 2.0.

In versions before 2.0, the illuminance entity (e.g., sensor.hue_motion_illuminance) was present and updated correctly.

### Zigbee2MQTT version

2.2.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

Zigbee2MQTT running in Docker
Host OS: Debian 12 (Bookworm)
VM on x86-64 hardware
Zigbee2MQTT version: 2.2.2

mdp08, 2025-05-28:
**Update:**

The missing entity was due to a change in Zigbee2MQTT since v2.0 — `illuminance_lux` was replaced with `illuminance`.

In my case, the new entity was named `sensor.og_flur_sensor_illuminance`, but it was disabled by default in Home Assistant. After enabling it under:  
**Settings → Devices & Services → MQTT - [Sensor] → Show disabled entities → Enable `sensor.og_flur_sensor_illuminance`**,  
everything worked as expected again.

So this is not a bug, just a structural change. Keeping this open for visibility to help others.

closed 2025-05-28 by mdp08

sumitkm, 2025-09-10:
@mdp08 thank you so much for posting this! I was going nutty with all the broken automations that suddenly lost illuminance settings on all my motion sensors!